TheImagingSource/tiscamera #271: Auto expose based on region of interest

rbrown-acertaralabs, 2019-04-30:
I am using multiple TIS GigE cameras, model DFM 25GP031-ML.  My software application is written in Python running on Windows using the TIS module `tisgrabber.py` (supplied by TIS) which in turn uses `tisgrabber_x64.dll`.  It appears that the cameras perform auto-exposure based on the entire image captured.  Thus, with a light subject object and a dark background, the camera tends to over-expose the subject object because it sees mostly the dark background (and conversely, with a dark subject and a light background it will tend to underexpose the subject object).  We can improve the situation by re-composing the image so that the subject object fills most of the image area and by closely matching the background brightness to the subject object's brightness.  However, to get further improvement, is it possible to apply a region of interest that is used for auto exposure adjustment?  For example, if I could define a small rectangle within the center of the image and have the camera auto-expose based on that smaller rectangular region, then I believe the auto-exposure would be less affected by background and it would give us another degree of freedom in how we compose our shots.

Thank you.

rbrown-acertaralabs, 2019-05-01:
Updated information...

For example, in the TIS application "IC Capture" (v2.4) I have a sample image below which shows my subject on a dark blue background and IC Capture allows me to select an ROI (picture below is after selecting the ROI but before activating the ROI):

[image]

...notice in the above image that the subject is washed out presumably because the camera autoexposure is looking at the entire field of view and arriving at an ideal exposure based on the bright subject and the dark background.  In this case the dark background causes the camera to expose longer which then washes out the subject.  Now in IC Capture, after activating the ROI, it looks like the following:

[image]

...and we see that the autoexposure now appears to be calculated based only on the ROI and thus the bright subject is not washed out because the camera is excluding the dark background.  This behavior is promising, but:

1. Can I achieve this behavior programmatically using `tisgrabber.py`/`tisgrabber_x64.dll` ?  That is, make autoexposure work only within the ROI?  I would need a way to define ROI boundaries, of course.
2. Assuming I can define an ROI, I still want the entire image, not just the ROI.  I want only the autoexposure to work on the ROI but then capture the entire image/frame/field of view.

TIS-Stefan, 2019-05-02:
There are no Auto Function ROI properties available for the 25G camera series. Therefore, will encounter the the small over exposed region on a big dark background. 
You could use the Exposure Auto Reference Value for making the image somewhat darker. 

You see the auto exposure working as wanted on the small ROI, because the auto exposure simple receives the small ROI. The dark background is missing in there and therefore does not influence the average brightness of the whole image

An ROI is set by the video format.
Simply set it is 
"RGB32 (256x256)"

This is usually centered. It can be moved with the Partial Scan Offset properties.

I am very sorry, but I have no good recommendation, except using a gray background.

Stefan

rbrown-acertaralabs, 2019-05-02:
Hello Stefan,

Thank you for your response.

So I do have a few options to explore according to your comments:

1--
Use "Exposure Auto Reference Value" ... The only documentation that I have for "tisgrabber" is the built-in ".chm" help file that came with the DLL.  Sometimes it can be difficult to find items in there, but is "Exposure Auto Reference Value" the same as "IC_SetExpAbsVal" or "IC_SetExpRegVal"?  Or, do I use "IC_SetPropertyValue" using the proper property name for "Exposure Auto Reference Value"?  Regarding "IC_SetPropertyValue"...do you have a list of supported properties?  I can query the camera using "IC_GetPropertyValue" or check if a property exists by using "IC_IsPropertyAvailable", but when I first look for a property I am somewhat guessing about the name...thus, a master list of supported properties for all properties would be great.  At any rate, if I can get the correct property or function call, I can experiment with "Exposure Auto Reference Value".

2--
When you say "ROI is set by video format"...could I simply use "IC_SetVideoFormat" to set it to a small area like "RGB(30x10)" and the camera will grab only a 30x10 area and apply autoexpose only to that area?  If so, I could let it auto-expose on the 30x10 area, read the autoexposure final value that it settles on (call it AEV), then turn off autoexposure and manually set exposure to AEV, then use "IC_SetVideoFormat" to set it to the full-size width and height.  What do you think about that approach?

3--
You mention "Partial Scan Offset" properties...do you know the specific names of these properties?  Are they set with "IC_SetPropertyValue" or some other call in the DLL?

Thank you!
Rob

--
Robert Brown
Lead Software Engineer
Acertara Acoustic Laboratories, LLC
1900 South Sunset Street, Unit F
Longmont, CO 80501
(303)834-8413 x310
rbrown@acertaralabs.com<mailto:rbrown@acertaralabs.com>
www.acertaralabs.com<http://www.acertaralabs.com>
An ISO 13485:2016 certified and ISO 17025:2005 accredited company

rbrown-acertaralabs, 2019-05-02:
Thank you for your response.  

So I do have a few options to explore according to your comments:

1--
Use "Exposure Auto Reference Value" ... The only documentation that I have for "tisgrabber" is the built-in ".chm" help file that came with the DLL.  Sometimes it can be difficult to find items in there, but is "Exposure Auto Reference Value" the same as "IC_SetExpAbsVal" or "IC_SetExpRegVal"?  Or, do I use "IC_SetPropertyValue" using the proper property name for "Exposure Auto Reference Value"?  Regarding "IC_SetPropertyValue"...do you have a list of supported properties?  I can query the camera using "IC_GetPropertyValue" or check if a property exists by using "IC_IsPropertyAvailable", but when I first look for a property I am somewhat guessing about the name...thus, a master list of supported properties for all properties would be great.  At any rate, if I can get the correct property or function call, I can experiment with "Exposure Auto Reference Value".

2--
When you say "ROI is set by video format"...could I simply use "IC_SetVideoFormat" to set it to a small area like "RGB(30x10)" and the camera will grab only a 30x10 area and apply autoexpose only to that area?  If so, I could let it auto-expose on the 30x10 area, read the autoexposure final value that it settles on (call it AEV), then turn off autoexposure and manually set exposure to AEV, then use "IC_SetVideoFormat" to set it to the full-size width and height.  What do you think about that approach?

3--
You mention "Partial Scan Offset" properties...do you know the specific names of these properties?  Are they set with "IC_SetPropertyValue" or some other call in the DLL?

TIS-Stefan, 2019-05-02:
I am working on this. A new tisgrabber.dll is already created, with some more functions. One of them prints

IC_SetPropertyValue

Exposure

        Value

                Interface : Range

                Interface : AbsoluteValue

        Auto

                Interface : Switch

        Auto Reference

                Interface : Range

        Auto Max Value

                Interface : Range

                Interface : AbsoluteValue

        Auto Max Auto

                Interface : Switch

The correct parameters for IC_SetPropertyValue are

IC_SetPropertyValue( hGrabber, "Exposure","Auto Reference", 80);

where 80 is to be exchanged against a value of your choice between 0 and 255.

The second approach is correct, but the area is too small. The smallest video format is (96x4) . I am not sure about the stepsizes to the next format, but I suppose 8 is a good value, so you get 104x12 as next size (or 96x12, or 104x4)

Partial Scan is

Partial scan

        Auto-center

                Interface : Switch

        X Offset

                Interface : Range

        Y Offset

                Interface : Range

IC_SetPropertySwitch( hGrabber, "Partial scan","Auto-center", 80);

IC_SetPropertyValue( hGrabber, "Partial scan","X Offset", 80);

IC_SetPropertyValue( hGrabber, "Partial scan","Y Offset", 80);

If you install the normal IC Imaging Control SDK 3.4, then you will get the VCD Property Inspector as sample. This shows all property names and their elements and interfaces.

closed 2019-05-02 by TIS-Stefan

rbrown-acertaralabs, 2019-05-02:
Thank you for your very quick response!  I can experiment with the following as you suggest:

`IC_SetPropertyValue( hGrabber, "Exposure","Auto Reference", [0-255])`

That is promising about setting the video size...my 30x10 was just a fabricated example...in reality I would likely need something near 500x300, so maybe this would be a viable approach...I will also experiment with that.

Regarding Partial Scan, I interpret "80" as the pixel offset for "X Offset" and "Y Offset" (so the center of the scan would be 80 pixels to the right and 80 pixels lower than center in your example).  But what does "80" represent in the "Auto-center" Switch-type variable?  I thought a Switch-type variable was just True (non-zero) or False (zero)...so when you set "Auto-center" to non-zero (80) are you simply telling the partial offset do an auto-center as a starting point and then apply X Offset and Y Offset from the auto-center point?  Maybe "80" was just a copy-and-paste leftover?
 
OK on VCD Property Inspector...I've used that in the past but thought I would ask if there was another way to get a list of available properties...I will continue to use VCD Property Inspector, no problem.

rbrown-acertaralabs, 2019-05-02:
Can you also describe in detail what the following items are doing?
    `Exposure:Auto Reference`
    `Exposure:Auto Max Value`
    `Exposure:Auto Max Auto`

(I will experiment with `Exposure:Auto Reference` as you indicate, but I am curious about what these parameters are actually doing, just to help me understand the process during my experimentation).

rbrown-acertaralabs, 2019-05-02:
Also, you indicate "a new tisgrabber.dll is already created"   Is this already posted on the TIS Web site available for download (IC Imaging Control SDK)?  Do I need this new version for the experiments discussed on this ticket?

TIS-Stefan, 2019-05-02:
Hi
Property descriptions are in the technical reference manuals. You may look at this: https://s1-dl.theimagingsource.com/api/2.5/packages/documentation/manuals-trm/trmdmk33ux287/5e1552b6-0d03-58e0-a202-ada591c8d537/trmdmk33ux287.en_US.pdf

This is a different camera model, but the property description is for all cameras.

IC_SetPropertyValue( hGrabber, "Partial scan","X Offset", 80);
IC_SetPropertyValue( hGrabber, "Partial scan","Y Offset", 80);

Means, the ROI is at pos 80/80 of the sensor. The Auto Center must be disabled, before you can set the offsets.

The DLL is still in work, I just finished the new callback function. That is for C. Now I have to translate all this to Pyhton.  Unfortunately I have some more tasks to do, so there is no date line.

rbrown-acertaralabs, 2019-05-02:
OK, thank you for the link to the technical manual...that helps a lot.
So when you originally indicated:
    `IC_SetPropertySwitch( hGrabber, "Partial scan","Auto-center", 80)` 
...did you actually mean to set it to 0 rather than 80 (to turn off Auto Center)?:
    `IC_SetPropertySwitch( hGrabber, "Partial scan","Auto-center", 0`)

The partial scan approach looks promising.  I will experiment.

OK on the DLL delivery date ... could you please notify me when it becomes available?  So to conclude, it sounds like I can proceed with experiments (1), (2), and (3) above with the current DLL and with the 25G, is that right?  That is, can I do the following with the current DLL (without waiting for the new DLL):
1. Experiment with "Exposure Auto Reference Value"
2. Experiment with "IC_SetVideoFormat"
3. Experiment with "Partial scan"