TheImagingSource/tiscamera #272: Limited properties on Jetson TX2

eloibove, 2019-05-01:
Hi, I am trying to modify the properties of my DFK33ux252 on a NVIDIA Jetson TX2 board. I am having issues retrieving the properties using tcam-ctrl (using tiscamera last version driver). It only shows very few properties, comparing with my desktop computer (see properties below).
Thank you!

**On Jetson TX2**
```
nvidia@tegra-ubuntu:~$ tcam-ctrl -p 49810042
Found 10 propert(y/ies)
Brightness                              (int) min=0 max=4095 step=1 default=240 value=240 category=Exposure
Saturation                              (int) min=0 max=255 step=1 default=64 value=64 category=Color
Hue                                     (int) min=-180 max=180 step=1 default=0 value=0 category=Color
Whitebalance Auto                       (bool) default=true value=true category=Color
Whitebalance Red                        (int) min=0 max=255 step=1 default=64 value=80 category=Color
Whitebalance Blue                       (int) min=0 max=255 step=1 default=64 value=229 category=Color
Gamma                                   (int) min=1 max=500 step=1 default=100 value=100 category=Image
Gain                                    (int) min=0 max=480 step=1 default=0 value=0 category=Exposure
Exposure Auto                           (bool) default=true value=true category=Exposure
Exposure                                (int) min=100 max=4000000 step=100 default=300 value=300 category=Exposure
```

**On another PC**
```
Found 47 propert(y/ies)
Brightness                              (int) min=0 max=4095 step=1 default=240 value=240 category=Exposure
Saturation                              (int) min=0 max=255 step=1 default=64 value=64 category=Color
Hue                                     (int) min=-180 max=180 step=1 default=0 value=0 category=Color
Whitebalance Auto                       (bool) default=true value=true category=Color
Whitebalance Red                        (int) min=0 max=255 step=1 default=64 value=64 category=Color
Whitebalance Blue                       (int) min=0 max=255 step=1 default=64 value=64 category=Color
Gamma                                   (int) min=1 max=500 step=1 default=100 value=100 category=Image
Gain                                    (int) min=0 max=480 step=1 default=0 value=0 category=Exposure
Exposure                                (int) min=20 max=4000000 step=1 default=333 value=333 category=Exposure
Exposure Auto                           (bool) default=true value=true category=Exposure
Exposure Auto Reference                 (int) min=0 max=255 step=1 default=128 value=128 category=Exposure
Gain Auto                               (bool) default=true value=true category=Exposure
Trigger Mode                            (bool) default=false value=false category=Special
Software Trigger                        (button) category=Special
Trigger Delay (us)                      (int) min=0 max=1000000 step=1 default=0 value=31 category=Special
Strobe Enable                           (bool) default=false value=false category=Special
Strobe Polarity                         (bool) default=false value=false category=Special
Strobe Exposure                         (bool) default=true value=true category=Special
Strobe Duration                         (int) min=0 max=32767 step=1 default=100 value=100 category=Special
Strobe Delay                            (int) min=0 max=32767 step=1 default=0 value=0 category=Special
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special
GPIO                                    (int) min=0 max=1 step=1 default=0 value=0 category=Special
Offset X                                (int) min=0 max=1792 step=2 default=0 value=0 category=Partial Scan
Offset Y                                (int) min=0 max=1532 step=2 default=0 value=0 category=Partial Scan
Offset Auto Center                      (bool) default=true value=true category=Partial Scan
Trigger Polarity                        (enum)  default=RisingEdge category=Special
						value=FallingEdge
							FallingEdge
							RisingEdge
Trigger Exposure Mode                   (enum)  default=FrameStart category=Special
						value=FrameStart
							ExposureActive
							FrameStart
Trigger Burst Count                     (int) min=1 max=1000 step=1 default=1 value=1 category=Special
Trigger Debounce Time (us)              (int) min=0 max=1000000 step=1 default=0 value=0 category=Special
Trigger Mask Time (us)                  (int) min=0 max=1000000 step=1 default=0 value=0 category=Special
Trigger Noise Surpression Time (us)     (int) min=0 max=1000000 step=1 default=0 value=0 category=Special
Auto Functions ROI Control              (bool) default=true value=true category=Special
Auto Functions ROI Left                 (int) min=0 max=2032 step=1 default=0 value=512 category=Special
Auto Functions ROI Top                  (int) min=0 max=1520 step=1 default=0 value=384 category=Special
Auto Functions ROI Width                (int) min=16 max=2048 step=1 default=0 value=1024 category=Special
Auto Functions ROI Height               (int) min=16 max=1536 step=1 default=0 value=768 category=Special
Whitebalance                            (bool) default=false value=false category=Color
Whitebalance Auto Preset                (enum)  default=Any    category=Color
						value=Any
							Any
							Cool White
							Daylight
							Deep Shade
							Warm White
Whitebalance Green                      (int) min=0 max=255 step=1 default=64 value=64 category=Color
White Balance Temperature Preset        (enum)  default=       category=Color
						value=Cool White Fluorescent Light
							Blue Sky
							Cool White Fluorescent Light
							Cool White LED
							Cool White/Daylight (CFL)
							Daylight
							Halogen Light
							Neutral White
							Sodium-Vapur Lamp
							Warm Light
Whitebalance Temperature                (int) min=2500 max=10000 step=100 default=6500 value=4000 category=Color
Reverse X                               (bool) default=false value=false category=Unknown
Reverse Y                               (bool) default=false value=false category=Unknown
Exposure Auto Upper Limit Auto          (bool) default=true value=true category=Exposure
Exposure Auto Upper Limit               (int) min=20 max=1000000 step=1 default=333 value=33333 category=Exposure
Override Scanning Mode                  (int) min=1 max=16 step=1 default=1 value=0 category=Unknown
Auto Functions ROI Preset               (enum)  default=Center 50% category=Special
						value=Center 50%
							Bottom Half
							Center 25%
							Center 50%
							Custom Rectangle
							Full Sensor
							Top Half
```

RogerAylagas, 2019-05-01:
Hi,
I am having the same issue. Does anyone have a solution?
Thank you!

TIS-Stefan, 2019-05-02:
Hello

This is a known issue of V4L2 on TX2 made by the NVidia engineers.
There are two solutions:
1) Look at https://github.com/TheImagingSource/tiscamera/issues/268
That may helps
2) Uses Aravis instead of v4l2. 
 Download file 
[aravis_usbvision_0_10.zip](https://github.com/TheImagingSource/tiscamera/files/3137407/aravis_usbvision_0_10.zip)
 and copy it into your tiscamera  directory of your TX2. Then perform following steps:
```
cd tiscamera
patch -p1 aravis_usbvision_0_10.patch
mkdir build
cd build
cmake -DBUILD_ARAVIS=ON -DBUILD_TOOLS=ON ..
make
sudo make install
```
tcam-ctrl should list the USB camera as "aravis" now. 
`DFK 37UX265	ARAVIS	34819936`

== known problems
indexing takes long -> aravis issue. gige-daemon only lists gige devices not UsbVision
high framerates do not work -> aravis issue. 

The second method is safer than patching the Kernel. But Aravis is somewhat less performant, than v4l2.

I hope, this helps

eloibove, 2019-05-11:
Hi Stefan,

We first tried the aravis solution but we couldn't make it work. In the end, with the solution in #268 it finally works! Thank you very much for your help!

You can close this issue for my part.

TIS-Stefan, 2019-05-13:
Thank you very much for your feedback!

closed 2019-05-13 by TIS-Stefan