TheImagingSource/tiscamera #274: Debian buster: tcam-gigetool install to /usr/local misses python site packages

GertRozing, 2019-05-15:
I am running a up to date Debian Buster machine and installed the tiscamera project. The installed tcam-gigetool however failed to start and quits with the following error message.

``` Python3
Traceback (most recent call last):
  File "/usr/local/bin/tcam-gigetool", line 34, in <module>
    from .version import TCAM_VERSION, TCAM_GIGETOOL_VERSION, TCAM_GIGETOOL_GIT_REVISION
ModuleNotFoundError: No module named '__main__.version'; '__main__' is not a package
```

I traced the install information and found out, that the python packages were installed in the wrong directory. I found them in **/usr/local/usr/lib/python3/dist-packages/tcamgigetool** which is wrong. A symlink from the correct path **/usr/lib/python3/dist-packages** semi-solved the issue.

The root of the issue was found in the CMakeLists.txt of tcam-gigetool. There is a python call, which tries to find the correct dist-packages path. However it fails when using a custom CMAKE_PREFIX_PATH. I fixed it in the PR PLACEHOLDER_LINK_THE_PR

However python still refused to find the required modules. By adding the module name to the import also fixed this problem. The fix for this is also in the PR.

CMake configuration call.
``` bash
cmake -D CMAKE_INSTALL_PREFIX=/usr/local -D BUILD_TOOLS=ON -D BUILD_ARAVIS=ON -S $HOME/Workspace/tiscamera -B $HOME/Workspace/tiscamera/build
```

CMake build call
```
cmake --build $HOME/Workspace/tiscamera/build --config Release
```

CMake install call
```
cmake --build $HOME/Workspace/tiscamera/build --config Release --target install
```

TIS-Stefan, 2019-05-15:
Hello

Please use the camera-ip-conf tool. The tcam-gigetool is not complete, as it seems. 
I am sorry for that.

Stefah

GertRozing, 2019-05-15:
Ah I see there is an alternative. I am pretty sure that I have always used the tcam-* programs for our TIS cameras. Maybe I am wrong about that. However camera-ip-conf also seems to meet my requirements. Thank you very much.

GertRozing, 2019-05-16:
Closed, as the connected PullRequest was merged.

closed 2019-05-16 by GertRozing