TheImagingSource/tiscamera #277: Missing property in new release

jassib, 2019-05-16:
Hi,
When using the library at the previous release (0.10), there was the property Trigger Polarity that I cannot find anymore on this release (0.11). Any ideas why this may be? I have followed the instructions at https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units but gotten 
`<pre>[libwebcam] Unknown V4L2 control ID encountered: 0x0199E201
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E201, name = &apos;Exposure Time (us)&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E202
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E202, name = &apos;Auto Shutter&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E203
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E203, name = &apos;Auto Exposure Reference&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E205
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E205, name = &apos;Gain Auto&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E208
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E208, name = &apos;Trigger Mode&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E209
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E209, name = &apos;Software Trigger&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E210
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E210, name = &apos;Trigger Delay&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E211
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E211, name = &apos;Strobe Enable&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E212
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E212, name = &apos;Strobe Polarity&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E213
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E213, name = &apos;Strobe Exposure&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E214
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E214, name = &apos;Strobe Duration&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E215
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E215, name = &apos;Strobe Delay&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E216
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E216, name = &apos;GPOUT&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E217
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E217, name = &apos;GPIN&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E218
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E218, name = &apos;ROI Offset X&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E219
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E219, name = &apos;ROI Offset Y&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E220
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E220, name = &apos;ROI Auto Center&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E234
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E234, name = &apos;Trigger Polarity&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E236
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E236, name = &apos;Trigger Expposure Mode&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E237
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E237, name = &apos;Trigger Burst Count&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E238
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E238, name = &apos;Trigger Debounce Time (us)&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E239
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E239, name = &apos;Trigger Mask Time (us)&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E240
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E240, name = &apos;Trigger Noise Suppression Time&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E241
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E241, name = &apos;Auto Functions ROI Control&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E242
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E242, name = &apos;Auto Functions ROI Left&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E243
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E243, name = &apos;Auto Functions ROI Top&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E244
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E244, name = &apos;Auto Functions ROI Width&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E245
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E245, name = &apos;Auto Functions ROI Height&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E251
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E251, name = &apos;Flip Horizontal&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E252
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E252, name = &apos;Flip Vertical&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E254
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E254, name = &apos;Exposure Auto Upper Limit Auto&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E256
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E256, name = &apos;Exposure Auto Upper Limit (us)&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E257
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E257, name = &apos;Override Scanning Mode&apos;
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E258
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E258, name = &apos;Auto Functions ROI Preset&apos;
[... 114 lines cut ...]
usb3.xml: error: video1: unable to map &apos;Highlight Reduction&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
usb3.xml: error: video1: unable to map &apos;Trigger Polarity&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Trigger Operation&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
usb3.xml: error: video1: unable to map &apos;Trigger Exposure Mode&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Trigger Burst Count&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Trigger Debounce Time (us)&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Trigger Mask Time (us)&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Trigger Noise Suppression Time&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Auto Functions ROI Preset&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Auto Functions ROI Control&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Auto Functions ROI Left&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
usb3.xml: error: video1: unable to map &apos;Auto Functions ROI Top&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
[... cut ...]
usb3.xml: error: video1: unable to map &apos;Gain Auto Lower Limit&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
usb3.xml: error: video1: unable to map &apos;Gain Auto Upper Limit&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
usb3.xml: error: video1: unable to map &apos;Override Scanning Mode&apos; control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)</pre>`

TIS-Stefan, 2019-05-21:
Hello

Sorry for the delay. We confirm the missing property and figure out, why it is missing currently.
I suppose, there will be an update in the near future.
Stefan

TIS-Stefan, 2019-05-22:
Hello

The bug is fixed.
Please "git pull" and rebuild, install. Result after reconnecting the camera is;

```
$ tcam-ctrl -p 45610198 | grep Polarity
Strobe Polarity                         (bool) default=false value=false category=Special
Trigger Polarity                        (enum)  default=Rising Edge category=Special
```

Stefan

jassib, 2019-05-22:
Hi. I have just pulled, rebuilt and tested, and the bug is fixed. Thanks for the help!
You can close this issue on my end!

closed 2019-05-22 by TIS-Stefan