TheImagingSource/tiscamera #281: What are the default values for camera properties

yoelk, 2019-05-22:
Hello,
I'm using a `DMK 39GX265-Z20-TI` with a serial of 16910171
When doing `tcam-ctrl -p 16910171` and I look at the default values, I don't always get them the same.
For example, for `Gain`, I got once `default=47.9` and another time `default=0`

So what are the rules for default values? are they supposed to change or stay the same?
And if they change, why are they there? What do they mean?

I'd also like to mention that while using the `tiscamera` python driver to read this default value, I got 47.900000000000006, which isn't coherent with the defined step of the `Gain` property (0.001). This seems like a bug to me.

Thanks,
Joel

yoelk, 2019-05-22:
And adding to this issue, when I'm already looking at other default values:
Are these numbers correct? They seem too high / too low (e.g. 4.66632e-43).
Also for them, the default value isn't coherent with the step size and the min/max values
```
MultiFrameSetOutputModeExposureTime0    (double) min=20 max=3e+07 step=0.001 default=4.66632e-43 value=4.66632e-43 category=Unknown group=0
MultiFrameSetOutputModeExposureTime1    (double) min=20 max=3e+07 step=0.001 default=4.66632e-43 value=4.66632e-43 category=Unknown group=0
MultiFrameSetOutputModeExposureTime2    (double) min=20 max=3e+07 step=0.001 default=4.66632e-43 value=4.66632e-43 category=Unknown group=0
MultiFrameSetOutputModeExposureTime3    (double) min=20 max=3e+07 step=0.001 default=4.66632e-43 value=4.66632e-43 category=Unknown group=0
SensorPixelWidth                        (double) min=-1.79769e+308 max=1.79769e+308 step=0.001 default=3.45 value=3.45 category=Unknown group=0
```

TIS-Stefan, 2019-05-23:
Hello Yoek

you are right, 47,9 is not the default value for gain, the default is 0.0,
What do you mean by "tiscamera python driver"?
Also I am not sure, whether the stepping is really considered, since the camera does not handle dB. It is converted into a matching register values internally. 

The properties are documented at https://s1-dl.theimagingsource.com/api/2.5/packages/documentation/manuals-trm/trmdmk39gx265z20/3825943c-ab8a-5757-ae6d-d611e97afc55/trmdmk39gx265z20.en_US.pdf

There is no MultiFrameSetOutputModeExposureTime0  in, therefore, I wonder where it comes from.

TIS-Stefan, 2019-05-23:
Hello Yoelk

My programmer just told me, that Genicam has no default values. We have an extra propreitary extension in the camera's xml file with the default values, but Aravis does not support that. 
Thus, the default values are set with the current set values.

However, that does not explain the MultiFrameSetOutputModeExposureTime*. We investigate.

Stefan

yoelk, 2019-05-23:
Thanks Stefan.
So where should I be able to read the default values?
Are they the ones that I read the first time I connect the camera?
Here is the full output of `tcam-ctrl -l` for this camera:
```
    Found 83 propert(y/ies)
    Focus                                   
    (int) min=0 max=10000 step=1 default=10000 value=10000 category=Lens

    AutoFocusROIEnable                      
    (bool) default=false value=false category=Unknown

    AutoFocusROILeft                        
    (int) min=0 max=1920 step=1 default=0 value=0 category=Unknown

    AutoFocusROITop                         
    (int) min=0 max=1520 step=1 default=0 value=0 category=Unknown

    AutoFocusROIWidth                       
    (int) min=128 max=2048 step=1 default=640 value=640 category=Unknown

    AutoFocusROIHeight                      
    (int) min=16 max=1536 step=1 default=480 value=480 category=Unknown

    Focus Auto                              
    (button) category=Lens

    Zoom                                    
    (int) min=0 max=1000 step=1 default=0 value=0 category=Lens

    Iris                                    
    (int) min=0 max=255 step=1 default=255 value=255 category=Lens

    IrisAuto                                
    (bool) default=false value=false category=Unknown

    IRCutFilterEnable                       
    (bool) default=false value=false category=Unknown

    GPIO                                    
    (int) min=0 max=3 step=1 default=2 value=2 category=Special

    Strobe Delay                            
    (int) min=0 max=32767 step=1 default=0 value=0 category=Special

    Strobe Duration                         
    (int) min=0 max=32767 step=1 default=100 value=100 category=Special

    Strobe Operation                        
    (enum)  default=Exposure category=Special
    value=Exposure
    Exposure
    FixedDuration

    Strobe Polarity                         
    (enum)  default=ActiveLow category=Special
    value=ActiveLow
    ActiveHigh
    ActiveLow

    Strobe Enable                           
    (enum)  default=Off    category=Special
    value=Off
    Off
    On

    GPIn                                    
    (int) min=0 max=1 step=1 default=1 value=1 category=Special

    GPOut                                   
[... 226 lines cut ...]
    (bool) default=false value=false category=Exposure

    Exposure Auto Upper Limit Auto          
    (bool) default=false value=false category=Exposure

    Exposure Auto Upper Limit               
    (double) min=20 max=3e+07 step=0.001 default=33333 value=33333 category=Exposure 
    group=429785090

    Exposure Auto Lower Limit               
    (double) min=20 max=3e+07 step=0.001 default=60 value=60 category=Exposure 
    group=429785090

    Exposure Auto Reference                 
    (int) min=0 max=255 step=1 default=128 value=128 category=Exposure

    Exposure Auto                           
    (enum)  default=Continuous category=Exposure
    value=Continuous
    Continuous
    Off

    Exposure                                
    (int) min=20 max=30000000 step=1 default=333 value=333 category=Exposure
```

TIS-Stefan, 2019-05-23:
Hi Yoelk

> So where should I be able to read the default values?
As I wrote: Only in the Technical Reference Manual . I am totally aware, that this is not a helpful and useful answer (blame me..) but since Genicam does not handle default values and Aravis does not handle the XML extension, we have no chance. The only way would be writing an own Linux GigEVision driver. But this takes a lot of time and therefore, is no option. 
The only way is getting the interesting default values from the Technical Reference Manual and use them hardcoded in your program.

Stefan

yoelk, 2019-05-23:
ok, thanks. if you can, please post here the XML file. It will be easier to go over, I guess

TIS-Stefan, 2019-05-23:
Hello

I just found, that the Multiframeoutput mode is available in the camera.  The downloaded XML file from the camera is:
[DMK 39GX265-Z20.zip](https://github.com/TheImagingSource/tiscamera/files/3212036/DMK.39GX265-Z20.zip)
I do not think, that it will be of any help. I can not interprete the ilfe.

Stefan

closed 2019-11-29 by TIS-Stefan