TheImagingSource/tiscamera #283: Error while running tiscamera inside docker.

gaurav181, 2019-05-29:
Dear Tiscamera experts,
   I am getting following error, when I am running tiscamera inside docker

The GIT commit I am setting to is this
git reset --hard 4ac6a7d8c8f32288e54fd3517ef924df524ba468`

root@pdm-qa1:/# /tiscamera/examples/ROS/tis_rosstarter
/tiscamera/examples/ROS/tiscamera.py:1: PyGIWarning: Tcam was imported without specifying a version first. Use gi.require_version('Tcam', '0.1') before import to ensure that the right version gets loaded.
from gi.repository import Tcam, Gst, GLib, GObject
/tiscamera/examples/ROS/tiscamera.py:1: PyGIWarning: Gst was imported without specifying a version first. Use gi.require_version('Gst', '1.0') before import to ensure that the right version gets loaded.
from gi.repository import Tcam, Gst, GLib, GObject
topic_name: None
node_name: None
self.topic_name: 
self.node_name: __name:=tis_camera_node
tcambin serial="31810211" name=source ! video/x-raw,format=BGRx,width=1920,height=1208,framerate=30/1 ! videoconvert ! video/x-raw,format=RGB ,width=1920,height=1208,framerate=30/1! shmsink socket-path=/tmp/ros_mem
Traceback (most recent call last):
File "/tiscamera/examples/ROS/tis_rosstarter", line 12, in <module>
cam = tiscamera.Camera("31810211", 1920, 1208, 30, True, False)
File "/tiscamera/examples/ROS/tiscamera.py", line 78, in _init_
raise RuntimeError("Failed to start video stream.")
RuntimeError: Failed to start video stream.

My docker file content for tiscamera looks like this, running it under Ubuntu16

```
# Install TIS Camera
RUN apt-get update && apt-get install -y g++ cmake pkg-config libudev-dev libudev1 libtinyxml-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev \
    libglib2.0-dev libgirepository1.0-dev libusb-1.0-0-dev libzip-dev uvcdynctrl python-setuptools libxml2-dev libpcap-dev libaudit-dev \
    libnotify-dev autoconf intltool gtk-doc-tools gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-plugins-base gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly libxml2 libpcap0.8 libaudit1 libnotify4
RUN git clone --recursive https://github.com/TheImagingSource/tiscamera.git && cd tiscamera && git reset --hard 4ac6a7d8c8f32288e54fd3517ef924df524ba468 && \
    mkdir build && cd build && cmake -DBUILD_ARAVIS=ON -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr .. && \
    cp /tis_ros/*.py ../examples/ROS && \
    make && make install

Please let me know, if anyone has any input. It is very urgent to me?

gaurav181, 2019-05-29:
```root@pdm-qa1:/deepmap/etc# gst-launch-1.0 tcambin serial="31810211" name=source ! video/x-raw,format=BGRx,width=1920,height=1208,framerate=30/1 ! videoconvert ! video/x-raw,format=RGB ,width=1920,height=1208,framerate=30/1! shmsink socket-path=/tmp/ros_mem
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...```

TIS-Arne, 2019-05-30:
@gaurav181 

Could you please start it like this:

`TCAM_LOG=DEBUG /tiscamera/examples/ROS/tis_rosstarter`

and post the output here?

closed 2019-11-29 by TIS-Stefan