TheImagingSource/tiscamera #284: [DMx 72BUC02] When changing format and binning, the order influences brightness

mihainicolae80, 2019-05-30:
Hi,

When I change the format (resolution/sensor crop size) and binning settings on  DMx 72BUC02, the order in which I change these user controls influences the brightness of the end image (not the brightness user control, but the actual brightness of the image).
Problems:
Starting with the max resolution: 2592x1940, if I first change the resolution to 640x480 and then the binning to 4 (which means binning X2), I get an overexposed image. 
If I first change the binning to 4 and then the resolution to 640x480, the image is "normally" exposed (I'm not really sure which one is the normal one).

I observed this behavior under linux (debian 9) using guvcview and also using the V4L2 linux api in a program I am writing. After the steps above, regardless of the order, the exposure, gain, GainR, GainG, GainB are unchanged and if I change them they increase/decrease the values as expected, making the base image more/less bright.

Thanks!

TIS-Stefan, 2019-05-31:
First of all, binning adds the pixel values. If exposure and gain are not changed, then a brighter image is expected. If automatics are enabled in the pipeline, then they should adjust against the bright images.

Please show your GStreamer Pipeline.

Stefan

mihainicolae80, 2019-05-31:
Hi, I don't think I use GStreamer, at least not directly. I know that binning adds the pixel values, but that behavior is not consequent, as described above. Changing the same paramaters in a different order and obtaining a different image because of that is an expected behavior? And if so, can you point me to some documentation regarding the correct order of setting the V4L2 settings and the User settings? 

Thank you!
Mihai

TIS-Stefan, 2019-05-31:
Hello Mihai

OK, if you do not use our GSteramer models, then you do not have Gain and exposure automatics.

> If I first change the binning to 4 and then the resolution to 640x480, the image is "normally" exposed 

I think, this wont have any effect, because binning can be set only, if the resolution is small enough. That means, you must set 640x480 first and then set binning. Not vice versa. And then the image must become 4 times brighter, because you do not change exposure and gain.

Stefan

mihainicolae80, 2019-05-31:
What you said makes sense but what I find rather peculiar is the fact that I actually CAN first change the binning to 4, which shows an unintelligible image, then change the resolution to 640x480, after which I can see the the image whitout being x4 times brighter or so. What is very weird is the fact that either way, regardless of the order, I still get the full field of view of the camera, the entire sensor is being used, and the image is 640x480 and has that square-ish look after binning, but is not that bright and no other user setting seem to have been changed. Maybe there is some auto-adjustment somewhere but I'm not sure what is it adjusting exactly since the user settings of exposure and gain are untouched.

TIS-Stefan, 2019-05-31:
You can set this property, because the camera does not return, whether it was successful or not. 

If you use 640x480 and binning 4x, then 4x4 pixels are added. Thus you see the full sensor, because 640x4 = (a lot of). And adding 4x4 pixel to one will results in that squarish look. This is expected.
Therefore, there is no effect, if you set binning to 4x4, if you have 2522x1944 resolution. 

Simple words : setting this Override Scan Properties under invalid conditions has no effect.

mihainicolae80, 2019-05-31:
Thanks for the answers, Stefan. I understand the right way to do it but I am yet to figure out the exact difference between the two pictures since all the paramteres that I can control seem to be the same but the resulting pictures do differ only in brightness, it seems. Regardless, there may be a hidden parameter in the firmware or driver of the camera that can not be accessed, but if you think that's not a constructive discussion, I think we can close this issue.

TIS-Stefan, 2019-05-31:
There is nothing hidden. You may attach sample images.

mihainicolae80, 2019-05-31:
I hope this clarifies the problem :)
[image]
[image]

TIS-Stefan, 2019-05-31:
Interesting... you see, that the indicated frame rate in the darker image is higher, than in the brighter? This becomes mysterious...

mihainicolae80, 2019-05-31:
Yes, I noticed that :) But I'm not really sure where to go from here.

TIS-Stefan, 2019-05-31:
You may use a sequence of commands, that is the best for your task.
I also noted the firmware team.
Stefan

mihainicolae80, 2019-05-31:
All right, thanks Stefan!

closed 2019-05-31 by mihainicolae80