TheImagingSource/tiscamera #285: Error  from element /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0: Device lost when using vp9enc and webmmux with the camera [DFK 33UX290] connected to a jetson NANO

kasra0, 2019-06-01:
Hello,

when i launch these 2 commands :

gst-launch-1.0 -e -vvv tcamsrc  ! queue ! videoscale ! videorate ! "video/x-raw, width=(int)1920, height=(int)1080, framerate=(fraction)60/1, stream-format=(string)byte-stream" \
!  videoconvert ! vp9enc  ! "video/x-vp9, width=(int)1920, height=(int)1080, framerate=(fraction)60/1, stream-format=(string)byte-stream"  \
!  webmmux     ! filesink location=/home/kasra/Videos/1920_1080_120_vp9enc_webmmux.webm    sync=false

gst-launch-1.0 -e -vvv tcamsrc  ! queue ! videoscale ! videorate ! "video/x-raw, width=(int)1920, height=(int)1080, framerate=(fraction)60/1, stream-format=(string)byte-stream" \ 
!  videoconvert ! vp9enc  ! "video/x-vp9, width=(int)1920, height=(int)1080, framerate=(fraction)60/1, stream-format=(string)byte-stream"  \
!  matroskamux ! filesink location=/home/kasra/Videos/1920_1080_120_vp9enc_matroskamux.mkv sync=false

**i have this error :  ERROR: from element /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0: Device lost (17910168)**

i attached the logs  for each command  with hight verbosity.

i have no problem with other solutions like : 
    omxh264enc with qtmux or mp4mux 
    omxh265enc with qtmux or mp4mux

i use GStreamer 1.14.1.
camera model :  DFK 33UX290
Board : Jetson NANO
OS : Linux for Tegra.

Best Regards,
Kasra
[vp9_matroska.txt](https://github.com/TheImagingSource/tiscamera/files/3243889/vp9_matroska.txt)
[vp9_webmmux.txt](https://github.com/TheImagingSource/tiscamera/files/3243890/vp9_webmmux.txt)

TIS-Stefan, 2019-06-03:
Hello

Before I investigate into this, I would like to know, whether
gst-launch-1.0 tcambin ! videoconvert ! ximagesink
shows a live video.

Thank you in advance!

kasra0, 2019-06-03:
Hello,

yes 
gst-launch-1.0 tcambin ! videoconvert ! ximagesink
gst-launch-1.0 tcamsrc ! videoconvert ! ximagesink

both works.

And actually these two following commands ( h264 and h265 encoding ) work too.

gst-launch-1.0 -e -vvv tcamsrc  ! queue ! videoscale ! videorate ! "video/x-raw, width=(int)1920, height=(int)1080, framerate=(fraction)60/1, stream-format=(string)byte-stream" !  videoconvert ! omxh265enc control-rate=2 bitrate=100000000 ! "video/x-h265, width=(int)1920, height=(int)1080, framerate=(fraction)60/1,  stream-format=(string)byte-stream"  !  h265parse   ! qtmux ! filesink location=/home/kasra/Videos/1920_1080_60_omxh265enc_qtmux.mp4 sync=false

gst-launch-1.0 -e -vvv tcamsrc  ! queue ! videoscale ! videorate ! "video/x-raw, width=(int)1920, height=(int)1080, framerate=(fraction)60/1, stream-format=(string)byte-stream" !  videoconvert ! omxh264enc control-rate=2 bitrate=100000000 ! "video/x-h264, width=(int)1920, height=(int)1080, framerate=(fraction)60/1,  stream-format=(string)byte-stream"  !  h264parse   ! qtmux ! filesink location=/home/kasra/Videos/1920_1080_60_omxh264enc_qtmux.mp4 sync=false

The  " ERROR: from element /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0: Device lost (17910168)"
appears when using vp9encoding.

thanks,

Best Regards,
Kasra

TIS-Stefan, 2019-06-03:
Hello Kasra

I am afraid, I can not help in this case. You wrote above, camera is working. Only you have some other GStreamer elements, that do not work, thus I would try to find the error there.
However, instead of our tcamsrc / tcambin you can use the videotestsrc and see, what happens.

Stefan

TIS-Arne, 2019-06-03:
@kasra0 

You could try activating the debug output of the tiscamera libraries to see if the issue is caused by our software. Set the following environment variable:
`export TCAM_LOG=DEBUG`
before running your gst-launch commands. 

But like Stefan said, it looks this issue is caused by something else than our software.

/Arne

kasra0, 2019-06-12:
Hi, 
thanks a lot for your advices,
i will test it and come back to you as soon as possible.
( i was in vacation )

Best Regards,
Kasra

kasra0, 2019-11-25:
Hi, 
Sorry, i completly forgot to come back to you.
It was my misunderstanding, now everything is working perfectly.
Thanks guys.
I close the "issue".

closed 2019-11-25 by kasra0