TheImagingSource/tiscamera #286: Exposure ROI functions not present in camera

alejandronespereira, 2019-06-06:
I am using some DFK 27BUP006 cameras and there are some functionalities missing in the python API/tcam-ctrl  that are nevertheless present in the tcam-capture.

On tcam-capture I can set a Brightness reference level and change the region from where the brightness is computed to compensate the exposure and gain, but when doing tcam-ctrl -l this is the result:

Brightness                              (int) min=0 max=4095 step=1 default=168 value=1 category=Exposure
Gain                                    (int) min=4 max=63 step=1 default=4 value=4 category=Exposure
Exposure                                (int) min=50 max=30000000 step=1 default=33333 value=15000 category=Exposure
Gain (dB/100)                           (int) min=0 max=1197 step=1 default=0 value=0 category=Unknown
Trigger Mode                            (bool) default=false value=false category=Special
Software Trigger                        (button) category=Special
Trigger Delay (us)                      (int) min=0 max=10000000 step=10 default=0 value=4400 category=Special
Strobe Enable                           (bool) default=false value=false category=Special
Strobe Polarity                         (bool) default=false value=false category=Special
Strobe Exposure                         (bool) default=false value=false category=Special
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special
GPIO                                    (int) min=0 max=1 step=1 default=0 value=0 category=Special
Offset X                                (int) min=0 max=2336 step=2 default=0 value=16 category=Partial Scan
Offset Y                                (int) min=0 max=1940 step=2 default=0 value=12 category=Partial Scan
Offset Auto Center                      (bool) default=true value=true category=Partial Scan
Override Scanning Mode                  (int) min=0 max=4 step=1 default=0 value=2 category=Unknown

No Brightness reference, no way to set a custom ROI for balance. I am getting fisheye images and I have 4 black corners around the image circle, without being able to change the region, these cameras are useless.

TIS-Stefan, 2019-06-07:
Hello Alejander

Am I right, that I already answered by email to you?

Stefan

alejandronespereira, 2019-08-06:
Yes! Thanks

closed 2019-08-06 by TIS-Stefan