TheImagingSource/tiscamera #288: `tcamsrc` *V0.11.1* hangs on `set_state(Gst.State.NULL)` reporting "Error closing udev monitoring pipe."

jraygauthier, 2019-06-07:
Was not hanging using *V0.9.1*.

Camera is:

```bash
$ tcam-ctrl -l
Available devices:
Model           Type    Serial

DFK 33UX264     V4L2    33710150
```

Here's what happens:

```bash
$ 2>&1 TCAM_LOG=DEBUG GST_DEBUG=WARNING pytest my/camera/tests.py

39:54.410 INFO    contextlib.py                      112  TISCameraPipeline.start - begin
39:54.410 DEBUG   tiscamera_pipeline.py              281  TISCameraPipeline._reload_pipeline
39:54.410 DEBUG   tiscamera_connection.py            84   TISCameraConnection.make_gst_source_element
39:54.411 DEBUG   tiscamera_source_utils.py          51   tcam_make_source: {serial: 33710150}
12295052   <INFO> /build/source/src/LibraryHandle.cpp:72: Could not load library /nix/store/xa1i6f3vnkx1w8dp9pkz2hfh27jhsaih-tiscamera-0.11.1/lib/tcam-0/libtcam-aravis.so
12295070   <INFO> /build/source/src/LibraryHandle.cpp:73:     Reason: /nix/store/xa1i6f3vnkx1w8dp9pkz2hfh27jhsaih-tiscamera-0.11.1/lib/tcam-0/libtcam-aravis.so: cannot open shared object file: No such file or directory
12295074   <INFO> /build/source/src/BackendLoader.cpp:103: Could not load backend /nix/store/xa1i6f3vnkx1w8dp9pkz2hfh27jhsaih-tiscamera-0.11.1/lib/tcam-0/libtcam-aravis.so
12300726   <INFO> /build/source/src/v4l2/V4l2Device.cpp:424: Current framerate is 35 / 1 fps
12301001   <INFO> /build/source/src/v4l2/V4l2Device.cpp:1205: Encountered write only control.
12301219   <INFO> /build/source/src/v4l2/V4l2Device.cpp:874: Detected multiple exposure interfaces. Simplifying
12301255   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:708: Found format: RGGB
12301281   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:708: Found format: RG16
12301308   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:708: Found format: YUYV
12301449   <INFO> /build/source/src/format.cpp:107: No string for fourcc 0x0
12301454   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:708: Found format:
12301479   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:708: Found format: BGR24
12301485   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:255: Returning 5 formats.
12301489   <DEBUG> /build/source/src/PipelineManager.cpp:127: Received 5 formats.
12301519   <DEBUG> /build/source/src/PropertyHandler.cpp:273: Created 52 properties
12301573   <INFO> /build/source/src/gstreamer-1.0/tcamgstbase.cpp:1209: Format has empty fourcc. Ignoring
12301942   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:1394: Changed ctrl Strobe Enable to value 1.
# ...
39:54.460 DEBUG   tiscamera_pipeline.py              221  TISCameraPipeline.src_vid_caps_str: video/x-bayer, format=rggb, width=2448, height=2048
39:54.461 DEBUG   tiscamera_pipeline.py              254  TISCameraPipeline.sink_vid_caps_str: video/x-raw, format=BGR
39:54.462 DEBUG   tiscamera_pipeline.py              268  TISCameraPipeline.videocrop: left=180, top=150, right=180, bottom=150
12311148   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:1394: Changed ctrl Trigger Mode to value 1.
39:54.467 DEBUG   tiscamera_pipeline.py              416  TISCameraPipeline.stop: set_state(Gst.State.PLAYING)
39:54.468 INFO    contextlib.py                      119  TISCameraPipeline.start - end
12312831   <INFO> /build/source/src/ImageSource.cpp:68: Source changed to state STOPPED
12313041   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:565: Stopped stream
12313062   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:169: Requested format change to 'format=RGGB,width=2448,height=2048,framerate=35.000000' 42474752
39:54.469 DEBUG   mcu_acquisition_device_async.py    141  MCUAcquisitionDeviceAsync: start_acquisition async begin
39:54.472 DEBUG   mcu_acquisition_device_blocking_impl.py 530  Start acquisition (post VASP): True
39:54.472 DEBUG   mcu_acquisition_device_async.py    145  MCUAcquisitionDeviceAsync: start_acquisition async end
12315525   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:391: Setting framerate to '35.000000' =  35 / 1
12315611   <INFO> /build/source/src/v4l2/V4l2Device.cpp:424: Current framerate is 35 / 1 fps
12315633   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:235: Active format is: 'format=RGGB,width=2448,height=2048,framerate=35.000000'
12315666   <DEBUG> /build/source/src/PipelineManager.cpp:222: Found device fourcc 'RGGB' - 42474752
12315673   <DEBUG> /build/source/src/PipelineManager.cpp:222: Found device fourcc 'RG16' - 36314752
12315678   <DEBUG> /build/source/src/PipelineManager.cpp:222: Found device fourcc 'YUYV' - 56595559
12315684   <INFO> /build/source/src/format.cpp:107: No string for fourcc 0x0
12315689   <DEBUG> /build/source/src/PipelineManager.cpp:222: Found device fourcc '' - 0
12315694   <DEBUG> /build/source/src/PipelineManager.cpp:222: Found device fourcc 'BGR24' - 33524742
12315704   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:169: Requested format change to 'format=RGGB,width=2448,height=2048,framerate=35.000000' 42474752
12315851   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:391: Setting framerate to '35.000000' =  35 / 1
12316100   <INFO> /build/source/src/v4l2/V4l2Device.cpp:424: Current framerate is 35 / 1 fps
12316141   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:235: Active format is: 'format=RGGB,width=2448,height=2048,framerate=35.000000'
12316152   <INFO> /build/source/src/ImageBuffer.cpp:38: allocating data buffer
# ...
12316207   <INFO> /build/source/src/PipelineManager.cpp:476: Pipeline creation successful.
12316212   <INFO> /build/source/src/PipelineManager.cpp:486: source ->  sink
12316217   <INFO> /build/source/src/ImageSink.cpp:51: Pipeline started playing
12316227   <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:1589: Will use 10 buffers
[... 38 lines cut ...]
[20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_warning_start_and_stop.log](https://github.com/TheImagingSource/tiscamera/files/3266457/20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_warning_start_and_stop.log)

[20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_warning.log](https://github.com/TheImagingSource/tiscamera/files/3266458/20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_warning.log)

[20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_info_start_and_stop.log](https://github.com/TheImagingSource/tiscamera/files/3266459/20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_info_start_and_stop.log)

[20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_info.log](https://github.com/TheImagingSource/tiscamera/files/3266460/20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_info.log)

[20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_debug.log](https://github.com/TheImagingSource/tiscamera/files/3266461/20190607_tiscamerapipeline_stop_hang_w_tcam_debug_w_gst_debug.log)

[ColorizedLogs.zip](https://github.com/TheImagingSource/tiscamera/files/3266467/ColorizedLogs.zip)

[PlainLogs.zip](https://github.com/TheImagingSource/tiscamera/files/3266473/PlainLogs.zip)

Note that in order to get these logs, I was forced to apply the following patch
in order to workaround a segfault:

[tiscamera/pull/287 - Comment out a line that segfaults the
app](https://github.com/TheImagingSource/tiscamera/pull/287)

This is most likely related to [tiscamera/issues/194 - How to properly close a
tcamsrc interface and
pipelines](https://github.com/TheImagingSource/tiscamera/issues/194)

This is a major problem for us, we need a quick fix.

TIS-Stefan, 2019-06-07:
Hello

Since it is already 5pm here and Monday is a holiday, I suggest to go back to the working version.
I suppose, the pipeline you use is somewhere in the zip files or links.

Stefan

TIS-Stefan, 2019-06-11:
Hello

"closing udev monitoring pipe" was an error indeed. It is fixed in the master branch now.

For the other error messages we would like to know, 
* which hardware do you use?
* How is your software built. Or does the same happen with   `gst-launch-1.0 tcambin ! videoconvert  ! ximagesink`

Thank you in advance!

Stefan

jraygauthier, 2019-06-13:
> How is your software built. Or does the same happen with gst-launch-1.0 tcambin ! videoconvert ! ximagesink

Some updates in the description under "Update 2019/06/13". We did not attempt to reproduce the issue with gst-launch.

I should attempt to create a simplified python app that reproduce this at some point.

> which hardware do you use?

I am reproducing this on a pretty standard dev machine:

```bash
$ lscpu 
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
Address sizes:       39 bits physical, 48 bits virtual
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
NUMA node(s):        1
Vendor ID:           GenuineIntel
CPU family:          6
Model:               158
Model name:          Intel(R) Core(TM) i5-7600K CPU @ 3.80GHz
Stepping:            9
CPU MHz:             4000.079
CPU max MHz:         4200.0000
CPU min MHz:         800.0000
BogoMIPS:            7584.00
Virtualization:      VT-x
L1d cache:           32K
L1i cache:           32K
L2 cache:            256K
L3 cache:            6144K
NUMA node0 CPU(s):   0-3
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
```

but our target machine is not much different in specs.

jraygauthier, 2019-06-14:
> For the other error messages we would like to know,

I'm curious about what are the errors you are referring to?

TIS-Arne, 2019-06-21:
@jraygauthier 

We are unable to reproduce this issue. Could you please send a small script that triggers the issue for you?

jraygauthier, 2019-06-21:
@TIS-Arne: Hi, I am on vacation this week, back the next. Leave me some time to create a sample script to reproduce. Indeed it might be that recent changes @TIS-Stefan pushed to `master` in response to this issue fix the problem (TBD). I shall do likewise for #287.

jraygauthier, 2019-06-26:
@TIS-Stefan :

Applying the following changsets on top of V0.11.1: 

 -  `28d8190b10e342da87a8e808f73130ad92fd4f87`: gst: Do not call src_stop when not running
 -  `c38c63b753939c099d398788b8a522f9e8b6a9b6`: v4l2: Fix pipe file descriptor handling

I saw the following critical gstreamer errors when running my unit tests:

```bash 
$ ./launch-unittest.sh | bcat

(pytest:25483): GStreamer-CRITICAL **: 11:24:21.365: Failed to deactivate pad tcamsrc32:src, very bad

(pytest:25483): GStreamer-CRITICAL **: 11:26:22.917: Failed to deactivate pad tcamsrc61:src, very bad

# ... A stream of those here.

(pytest:25483): GStreamer-CRITICAL **: 11:37:43.445: Failed to deactivate pad tcamsrc586:src, very bad

(pytest:25483): GStreamer-CRITICAL **: 11:37:48.038: Failed to deactivate pad tcamsrc593:src, very bad
```

Whenever one of these error occurs, one the test fails with a timeout awaiting on expected camera frames (which were produced when using V0.9.1).I 

It however does not seem to hang anymore.

Running with debug, I no longer can see the `Error closing udev monitoring pipe` it now outputs the following errors:

```bash
44:34.555 DEBUG   tiscamera_pipeline.py              416  TISCameraPipeline.start: set_state(Gst.State.PLAYING)
44:34.556 INFO    contextlib.py                      119  TISCameraPipeline.start - end
2343867    <INFO> /build/source/src/ImageSource.cpp:68: Source changed to state STOPPED
2343891    <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:567: Stopped stream
2343919    <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:171: Requested format change to 'format=RGGB,width=2448,height=2048,framerate=35.000000' 42474752
2343938    <ERROR> /build/source/src/v4l2/V4l2Device.cpp:188: Error while calling VIDIOC_REQBUFS to empty buffer queue. Device or resource busy
0:00:07.939177595 29250      0x22ff850 ERROR                tcamsrc gsttcamsrc.cpp:1106:send_log_to_bus: Backend reported error: /build/source/src/v4l2/V4l2Device.cpp:188: Error while calling VIDIOC_REQBUFS to empty buffer queue.  
2344007    <ERROR> /build/source/src/v4l2/V4l2Device.cpp:221: Error while setting format 'Device or resource busy'
0:00:07.939234727 29250      0x22ff850 ERROR                tcamsrc gsttcamsrc.cpp:1106:send_log_to_bus: Backend reported error: /build/source/src/v4l2/V4l2Device.cpp:221: Error while setting format ' '
0:00:07.939250658 29250      0x22ff850 ERROR                tcamsrc gsttcamsrc.cpp:977:gst_tcam_src_set_caps: Unable to set format in device
0:00:07.939275670 29250      0x22ff850 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<tcamsrc24> error: Internal data stream error.
0:00:07.939287087 29250      0x22ff850 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<tcamsrc24> error: streaming stopped, reason not-negotiated (-4)

# Some shutdown code on timeout.

44:45.931 INFO    contextlib.py                      112  TISCameraPipeline.stop - begin
44:45.931 DEBUG   tiscamera_pipeline.py              456  TISCameraPipeline.stop: set_state(Gst.State.PAUSED)
44:45.931 DEBUG   tiscamera_pipeline.py              458  TISCameraPipeline.stop: set_state(Gst.State.NULL)

(pytest:29250): GStreamer-CRITICAL **: 11:44:45.932: Failed to deactivate pad tcamsrc59:src, very bad

# ...

44:36.727 DEBUG   tiscamera_pipeline.py              416  TISCameraPipeline.start: set_state(Gst.State.PLAYING)
2596448    <INFO> /build/source/src/ImageSource.cpp:68: Source changed to state STOPPED
2596593    <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:567: Stopped stream
44:36.728 INFO    contextlib.py                      119  TISCameraPipeline.start - end
2596621    <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:171: Requested format change to 'format=RGGB,width=2448,height=2048,framerate=35.000000' 42474752
2596641    <ERROR> /build/source/src/v4l2/V4l2Device.cpp:188: Error while calling VIDIOC_REQBUFS to empty buffer queue. Device or resource busy
0:00:10.110474214 29250      0x22ff850 ERROR                tcamsrc gsttcamsrc.cpp:1106:send_log_to_bus: Backend reported error: /build/source/src/v4l2/V4l2Device.cpp:188: Error while calling VIDIOC_REQBUFS to empty buffer queue.  
2596707    <ERROR> /build/source/src/v4l2/V4l2Device.cpp:221: Error while setting format 'Device or resource busy'
0:00:10.110529072 29250      0x22ff850 ERROR                tcamsrc gsttcamsrc.cpp:1106:send_log_to_bus: Backend reported error: /build/source/src/v4l2/V4l2Device.cpp:221: Error while setting format ' '
0:00:10.110542985 29250      0x22ff850 ERROR                tcamsrc gsttcamsrc.cpp:977:gst_tcam_src_set_caps: Unable to set format in device
0:00:10.110567603 29250      0x22ff850 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<tcamsrc31> error: Internal data stream error.
0:00:10.110579875 29250      0x22ff850 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<tcamsrc31> error: streaming stopped, reason not-negotiated (-4)
44:36.731 DEBUG   mcu_acquisition_device_async.py    141  MCUAcquisitionDeviceAsync: start_acquisition async begin

# Some shutdown code on timeout.

44:45.931 INFO    contextlib.py                      112  TISCameraPipeline.stop - begin
44:45.931 DEBUG   tiscamera_pipeline.py              456  TISCameraPipeline.stop: set_state(Gst.State.PAUSED)
44:45.931 DEBUG   tiscamera_pipeline.py              458  TISCameraPipeline.stop: set_state(Gst.State.NULL)

(pytest:29250): GStreamer-CRITICAL **: 11:44:45.932: Failed to deactivate pad tcamsrc59:src, very bad

# ...
```

TIS-Edgar, 2019-06-28:
The pad error has been fixed and has already been pushed to master. Please let us know if this fixes your issues.

jraygauthier, 2019-06-28:
Hi @TIS-Edgar, I do not have access to my whole ext trigger kit today and limited to soft triggers. Using the latest, it seems to be better using only soft trigs runnin on top of kernel 4.19 and with my patch to comment out the problematic log. I will have to validate running all my automated tests new week.

I think I might have been bitten by either #296 or #295.

jraygauthier, 2019-07-02:
Hi again @TIS-Edgar . I can still reproduce systematically using b454a8bb06dbcb9e3d3434a03a72d3d2fa44c11e even only with software trigs on kernel 4.19.

See "Update 2019/07/02" section in the description for the script used to reproduce.

TIS-Edgar, 2019-07-04:
Hi there,
To avoid confusion. 
The pipeline does not hang but receives segfaults. Is that correct?
Is there a reason you are manually deleting the pipeline and source?

It is enough to simply set both to `None`. set_state(Gst.State.NULL) will handle almost all of the required cleanup.

For comparison:
When you compile tiscamera with `BUILD_TESTS=ON` your build directory will contain a test in `tests/integration/start_stop`. 
When you execute `start-stop-runner-sh --caps <caps> --serial` it will repeatedly call an executable, run for ~5 seconds and stop without having problems.

So this segfault has something to do with the way your script is handling the building/destruction of the pipeline/src.  How that can cause this I still do not know.
Is this a blocking issue for you or can you work around it until it is fixed?

jraygauthier, 2019-07-04:
@TIS-Edgar : 

 > The pipeline does not hang but receives segfaults. Is that correct?

For this issue, the pipeline really hangs after multiple iterations and does not segfault (on top of master, with #295 patch and on kernel 4.19). The segfault is the separate #295 issue. I have another separate issue for when used on top of kernel 4.14 #296 which leads to yet another kind of error.

Note however that on my home machine, I had no success in reproducing with this script. It might be some deadlock or timing issue which is dependent on the particular hw.

 >  Is there a reason you are manually deleting the pipeline and source?

To reproduce what is occurring in the actual application, when running unit tests, etc. The pipeline and the  source are not kept alive forever but only during the acquisition process.

This issue **is the only one for which we have no workaround** (but to revert to using 0.9.1). The hang prevent us from running or tests. 

In the upcoming days I will have much less time to work on these issues though as we're releasing and I'm working on another aspect of the application / environment. I should come back to this at some point later.

jraygauthier, 2020-02-20:
@TIS-Edgar 

## Reproducing the reentrancy issue with more logs

[tiscamera-unstable-patched-logs]: https://github.com/jraygauthier/tiscamera/commit/5ab0e793ff88a97f09e7728455a2e6c3026d704e

Running with the [tiscamera-unstable-patched-logs] version.

### Non hanging case

```log
03:36.833 DEBUG   tiscamera_pipeline.py              504  TISCameraPipeline.stop: set_state(Gst.State.PAUSED)
03:36.833 DEBUG   tiscamera_pipeline.py              506  TISCameraPipeline.stop: set_state(Gst.State.NULL)
0:00:05.798763228 31741      0x1c56d50 WARN                 tcamsrc gsttcamsrc.cpp:1252:gst_tcam_src_stop: gst_tcam_src_stop: init
0:00:05.798820033 31741      0x1c56d50 WARN                 tcamsrc gsttcamsrc.cpp:1257:gst_tcam_src_stop: gst_tcam_src_stop -> notify_all
0:00:05.798837518 31741      0x1c56d50 WARN                 tcamsrc gsttcamsrc.cpp:1262:gst_tcam_src_stop: gst_tcam_src_stop -> unique_lock
0:00:05.798850894 31741      0x1c56d50 WARN                 tcamsrc gsttcamsrc.cpp:1264:gst_tcam_src_stop: gst_tcam_src_stop -> stop_stream
2251480    <INFO> /build/source/src/ImageSource.cpp:68: Source changed to state STOPPED
2251943    <INFO> /build/source/src/v4l2/V4l2Device.cpp:585: notifying callbacks about lost device
0:00:05.804453533 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1050:gst_tcam_src_device_lost_callback: gst_tcam_src_device_lost_callback: init
0:00:05.804594361 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1057:gst_tcam_src_device_lost_callback:<tcamsrc9> error: Device lost (25910066)
0:00:05.804657283 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1065:gst_tcam_src_device_lost_callback:<tcamsrc9> error: Device lost
0:00:05.804693342 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1069:gst_tcam_src_device_lost_callback: gst_tcam_src_device_lost_callback -> gst_tcam_src_stop
0:00:05.804707804 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1252:gst_tcam_src_stop: gst_tcam_src_stop: init
0:00:05.804720392 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1257:gst_tcam_src_stop: gst_tcam_src_stop -> notify_all
0:00:05.804738695 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1262:gst_tcam_src_stop: gst_tcam_src_stop -> unique_lock
2252357    <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:567: Stopped stream
2252393    <INFO> /build/source/src/ImageSink.cpp:55: Pipeline stopped playing
0:00:05.806360619 31741      0x1c56d50 WARN                 tcamsrc gsttcamsrc.cpp:1266:gst_tcam_src_stop: gst_tcam_src_stop -> unlock
0:00:05.806400496 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1264:gst_tcam_src_stop: gst_tcam_src_stop -> stop_stream
0:00:05.806416961 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1266:gst_tcam_src_stop: gst_tcam_src_stop -> unlock
0:00:05.806429407 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1269:gst_tcam_src_stop: gst_tcam_src_stop -> gst_element_send_event
0:00:05.806714122 31741      0x1c56d50 WARN                 tcamsrc gsttcamsrc.cpp:1269:gst_tcam_src_stop: gst_tcam_src_stop -> gst_element_send_event
0:00:05.806748030 31741      0x1c56d50 WARN                 tcamsrc gsttcamsrc.cpp:1273:gst_tcam_src_stop: gst_tcam_src_stop: end
0:00:05.807435647 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1273:gst_tcam_src_stop: gst_tcam_src_stop: end
0:00:05.807484769 31741      0x1ce7d20 WARN                 tcamsrc gsttcamsrc.cpp:1071:gst_tcam_src_device_lost_callback: gst_tcam_src_device_lost_callback: end
2264005    <INFO> /build/source/src/CaptureDeviceImpl.cpp:159: Closed device DFM 37UX264-ML.
# Everything alright here.
```

### Hanging case

```log
03:44.888 DEBUG   tiscamera_pipeline.py              504  TISCameraPipeline.stop: set_state(Gst.State.PAUSED)
03:44.889 DEBUG   tiscamera_pipeline.py              506  TISCameraPipeline.stop: set_state(Gst.State.NULL)
0:00:05.703139131 31789      0x1247b50 WARN                 tcamsrc gsttcamsrc.cpp:1252:gst_tcam_src_stop: gst_tcam_src_stop: init
0:00:05.703188087 31789      0x1247b50 WARN                 tcamsrc gsttcamsrc.cpp:1257:gst_tcam_src_stop: gst_tcam_src_stop -> notify_all
0:00:05.703204712 31789      0x1247b50 WARN                 tcamsrc gsttcamsrc.cpp:1262:gst_tcam_src_stop: gst_tcam_src_stop -> unique_lock
0:00:05.703218308 31789      0x1247b50 WARN                 tcamsrc gsttcamsrc.cpp:1264:gst_tcam_src_stop: gst_tcam_src_stop -> stop_stream
2336395    <INFO> /build/source/src/ImageSource.cpp:68: Source changed to state STOPPED
2336822    <INFO> /build/source/src/v4l2/V4l2Device.cpp:585: notifying callbacks about lost device
2336891    <DEBUG> /build/source/src/v4l2/V4l2Device.cpp:567: Stopped stream
0:00:05.704613046 31789      0x12da120 WARN                 tcamsrc gsttcamsrc.cpp:1050:gst_tcam_src_device_lost_callback: gst_tcam_src_device_lost_callback: init
2336935    <INFO> /build/source/src/ImageSink.cpp:55: Pipeline stopped playing
0:00:05.704712388 31789      0x1247b50 WARN                 tcamsrc gsttcamsrc.cpp:1266:gst_tcam_src_stop: gst_tcam_src_stop -> unlock
0:00:05.704735024 31789      0x1247b50 WARN                 tcamsrc gsttcamsrc.cpp:1269:gst_tcam_src_stop: gst_tcam_src_stop -> gst_element_send_event
0:00:05.704758802 31789      0x1247b50 WARN                 tcamsrc gsttcamsrc.cpp:1273:gst_tcam_src_stop: gst_tcam_src_stop: end
[... cut ...]
    `gst_tcam_src_stop` context and avoid calling `gst_tcam_src_stop` again when
    this is the case.

Some questions to be answered if reentrancy is really meant to be a property of the `gst_tcam_src_stop` function:

 -  As mentionned above: shoud `std::recursive_mutex` be used instead `std::mutex`?

 -  Unprotected `gst_element_send_event(GST_ELEMENT(self), gst_event_new_eos());`?

    Won't it be called twice in the reentrant case. Could it be an issue?

 -  Potential double finalization: `gst_tcam_src_finalize`?

 -  Use of condition variable `self->cv`?

TIS-Edgar, 2020-02-24:
Hi,

That is a lot of work you put into that.
I will answer your questions to the best of my knowledge.

- should std::recursive_mutex be used instead std::mutex?
No. A recursive_mutex has the purpose of allowing the same thread that already holds the lock to lock the mutex multiple times. Since the call to `gst_tcam_src_stop` from the device_lost_callback is not recursive in any way this will not change anything. The EOS event is handled in a different thread thus the normal mutex suffices. 

- Unprotected gst_element_send_event(GST_ELEMENT(self), gst_event_new_eos());
I assume with unprotected you mean not locked by the mutex? The end-of-stream event if handled by the event loop one of the central parts of gstreamer. 
Once such an event is received all polling the source will stop i.e. gst_tcam_src_create will not be called any more and all elements down the pipeline will flush their data and transmit the eos event.
Sending multiple such events should make no difference as they are handled in a different thread context and all elements handle such an event exactly once.

- Potential double finalization: gst_tcam_src_finalize?
finalize is called when the plugin is destroyed. not the instance of the tcamsrc element.

The condition_variable self->cv  is used to enforce a waiting position in `gst_tcam_src_create`.
This function has the task to fill gstbuffer with image data once an image is ready.
This image 'ready' state is signaled in `gst_tcam_src_sh_callback`.
The reason for this is that the capture thread (V4l2Device.cpp::work_thread ->sh_callback) is independent from the gstreamer thread (create). Both are independent of the device_lost thread as that is simply for the case of "something went wrong, stop the stream".

Instead of de facto disabling device_lost may I suggest the following:

```
    self->is_running = FALSE;
    gst_element_send_event(GST_ELEMENT(self), gst_event_new_eos());
    // gst_tcam_src_stop(GST_BASE_SRC(self));
```
This will cause the pipeline itself to call gst_tcam_src_stop and is likely the better implementation in comparison to the current state.

One bug we recently discovered prevents a correct shutdown and _may_ be related to your issue.
[This patch](https://gist.github.com/TIS-Edgar/ebd6f5039d119909b052c85b55e180c8) addresses both. I'd understand if at this point you simply prefer to keep you solution and be done with it.

jraygauthier, 2020-02-24:
> Since the call to gst_tcam_src_stop from the device_lost_callback is not recursive in any way this will not change anything. The EOS event is handled in a different thread thus the normal mutex suffices.

I just noticed that indeed the device lost event is called from a totally separate thread. It does indeed makes the use of `std::recursive_mutex` unnecessary. 

> I assume with unprotected you mean not locked by the mutex?

Yes I was.

> Sending multiple such events should make no difference as they are handled in a different thread context and all elements handle such an event exactly once.

Alright. That should make it fine then.

> finalize is called when the plugin is destroyed

Alright, so unrelated.

> The condition_variable self->cv is used to enforce a waiting position in gst_tcam_src_create

Alright too.

However, with the assumption of the above being correct, this still does not explains why calling `gst_tcam_src_stop` directly from the `device_lost_callback` would hang. There must be something else that is wrong and should be understood.

I tested the [patch](cb133de28d978616659b04467a3f425c611a7025) you provided and it seem to fix the hang too.

Some more things I noticed by looking at the source code in an attempt to make sure I understood the various above answers: 

 -  `V4l2Device::is_stream_on` should at least be atomic as it is mutated from multiple threads and read from even more threads.

 -  Same thing applies to `_GstTcamSrc::is_running` which now can be set from the device lost thread too.

 -  It is really hard to understand how each states in the various instances (`V4l2Device`, `_GstTcamSrc`) are shared by the various threads and what kind of model has been taken to ensure thread-safety. For example, the various atomic variables in `V4l2Device` 're access has been made atomic individually. Is is sufficient under some assumption or should there been atomicity as a whole? 

    Some comments over each of these states / state groups would greatly helper a reader understand how developer ensured to preserve correctness of the program.

TIS-Edgar, 2020-02-26:
You are right with wanting to set a few more flags to atomic. I will apply a few changes to the development branch to address that.

The documentation about internal setups is indeed lacking.
I will add some comments as well as some descriptions to the documentation to make things more 'accessible'.

All programs basically have the following threads:

- the device index thread: started when tcambin/src query the available devices. once a device is open. It is also used for device-lost events.
- Your main thread: starts the stream by telling gstreamer to go into PLAYING
- the gstreamer main stream: puts tcamsrc into PLAYING
- the capture thread: tcamsrc starts this one. It waits until the backend (v4l2, aravis, libusb) says an image is ready and passes it (together with meta information like capture time) to a callback. This notifies gst_tcam_src_create about an image.
- the 'device lost' thread: started by tcamsrc to get notifications if the device does not respond/ is lost
- the streaming thread: this one calls gst_tcam_src_create waiting for new images to iterate over all used elements until the end of the pipeline or a queue is reached. If your streaming thread is identical to your display thread you may encounter problems.

All additional threads are either started in another context by the application and have nothing to do with the image capture or are started by a single gstreamer element for faster image processing. tcamwhitebalance is such an example as it launches work threads to evenly distribute the application of whitebalance values onto the image across all cores.

jraygauthier, 2020-02-26:
Thanks for this overview, will keep a reference to it.

Is the patch fixing this issue already part of the development branch? Can you provide a ref to the dev branch changeset?

If so (both of the above), I think the issue can be closed.

TIS-Edgar, 2020-02-27:
Is the patch fixing this issue already part of the development branch?

It is now!
Relevant commits are:  
c3e3d26d4595dc6861440221010660af02987cc4
628b6ff55885e31fe381e52712d5c2e08f5a961f

For an overview over the changes affecting the development branch, please refer to the [CHANGELOG.md](https://github.com/TheImagingSource/tiscamera/blob/development/CHANGELOG.md) of the development branch.
If there are any open questions concerning this issue, lacking documentation or the changelog, please let me know.

jraygauthier, 2020-02-27:
Excellent. Thanks for the prompt support @TIS-Edgar. This issue has been resolved for us. Closing.

closed 2020-02-27 by jraygauthier