TheImagingSource/tiscamera #289: Can't disable auto gain on DFK23U445 by any means

pbfy0, 2019-06-13:
I've been unable to disable auto gain on a DFK23U445. The example-02 program says that it has changed the auto gain setting (from True to False), but it has no effect and running the program again shows that it has reverted to False. I tried installing the UVC extension using both uvcdynctrl with usb3.xml and tcam-uvc-extension-loader with usb23.json. Both installed some additional properties, but failed to install the auto gain control (kernel says that the UVC control does not exist). I'm wondering if there's a firmware update I need to disable auto gain, or if this camera simply lacks the ability to disable it.

TIS-Stefan, 2019-06-14:
Did you also disable Auto Exposure?

pbfy0, 2019-06-14:
As far as I can tell, the Auto Exposure control also isn't exposed, but manually setting the exposure control seems to take effect; setting the exposure low makes gain go way up.

TIS-Edgar, 2019-06-19:
The DFK23U445 does not have internal auto algorithms. tcam-ctrl only shows the properties the camera itself has. If you want to disable auto-gain you will have to use the tcamprop interface or disable it in the tcamautoexposure gstreamer element. 
At this point in time tiscamera has __no__ caching. That means all sofware properties like auto gain will be reset to their default values.

closed 2019-06-28 by TIS-Edgar