TheImagingSource/tiscamera #290: `tcamsrc` *V0.9.1* stuck on auto exposure even disregarding ext trigger on "DMM 37UX264-ML" when in `ExposureActive` mode

jraygauthier, 2019-06-13:
-  Using 0.9.1, when I list tcam properties, I see `Exposure Auto` property twice:

    ```bash
    $ ./print-tcam-props.sh | grep 'Exposure Auto  '
    Exposure Auto                           (bool) default=true value=true category=Exposure
    Exposure Auto                           (bool) default=true value=true category=Exposure

    ```

 -  Still using 0.9.1, when I attempt to change to manual exposure, it fails:

    ```bash
    $ ./set-tcam-props.sh "Exposure Auto=false"
    tcam-ctrl -p -s "Exposure Auto=false" "12910635"
    Found property!
    Setting Exposure Auto to FALSE
    WARNING: More than one value found for "Exposure Auto". Selecting the first one.
     -> Exposure Auto="true"
    ERROR: `actual_value="true"` does not match `expected_value="false"`.
    
    $ ./print-tcam-props.sh | grep 'Exposure Auto  '
    Exposure Auto                           (bool) default=true value=true category=Exposure
    Exposure Auto                           (bool) default=true value=true category=Exposure

    ```

 -  Thinking the previous might be effective on the camera even tough this is
    not reflected in the property, I attempt to change it when my pipeline is
    running (I can see the stream using ximagesink):

    ```bash
    $ ./set-tcam-props.sh "Exposure=500"
    tcam-ctrl -p -s "Exposure=500" "12910635"
    Found property!
     -> Exposure="333"
    ERROR: `actual_value="333"` does not match `expected_value="500"`.
    ```

    The change is still not reflected by the property.

    Worst, **the change is not reflected on the real time stream**, whatever
    value I use.

All of the previous issues are fixed by 0.11.1. However, the following bug
is preventing us from adopting this newer release:

 -  [`tcamsrc` *V0.11.1* hangs on `set_state(Gst.State.NULL)` reporting "Error closing udev monitoring pipe." · Issue #288 · TheImagingSource/tiscamera](https://github.com/TheImagingSource/tiscamera/issues/288)

We would really like to move to 0.11.1 as with 0.9.1, we're stuck with kernel
4.14 (does not work with kernel 4.19 nor 5.0).

TODO: New issue + refer to that issue.

From 0.9.1 to 0.11.1 some properties names and values were changed:

 -  Prop: `Exposure` -> `Exposure Time (us)`
 -  Prop value for `Trigger Polarity`: `RisingEdge` -> `Rising Edge`
 -  Prop value for `Trigger Exposure Mode`: `ExposureActive` -> `Exposure Active`

I confirmed using `launch-gstreamer-rx-pipeline.sh` and signal generator for
which I changed the duty cycle of my trig in real time:

 -  For "DFK 33UX264" with fw-942 on tiscamera-0.9.1, I observed effective
    change in exposure proportionnaly to the increase in duty cycle.

 -  For "DMM 37UX264-ML" with fw-977 on tiscamera-0.9.1, the image oscillate
    (clearly is effectively in auto exposure mode) **completly disregarding my
    changes in duty cycle**.

TIS-Stefan, 2019-06-14:
Hello 

We would like to know, whether you use the Master branch and pulled from that.

jraygauthier, 2019-06-14:
Hi, no I did not try the fix you committed yet. We were working on having something stable for the release (on top of 0.9.1).

jraygauthier, 2019-06-26:
There is no "DMM 37UX264-ML" specific issue. This was a misconfiguration error on my part. Now properly configured, the ext trig expose is effective. See edit in description. So 0.9.1 would remain acceptable for us as a workaround for 0.11.1.

jraygauthier, 2020-02-27:
Now that #288 has been fixed (now moved to patched 0.11.1), this is no longer an issue for us. @TIS-Stefan: should I leave this open or close?

TIS-Stefan, 2020-02-27:
We close that

closed 2020-02-27 by TIS-Stefan