TheImagingSource/tiscamera #291: Program error at sample code, 04-list-formats.py in examples/python/

ryoji1415, 2019-06-24:
I have some error at running of sample code, 04-list-formats.py.
Following is my environments and error code.
Please give me some help.

## Details for PC
Surface laptop (64bit)
## OS
Ubuntu (64bit)
## OS version
18.04 LTS
## Tiscamera version
0.11.1
## Aravis version
0.6.3
## Process of build
### 1. Clone git file
    git clone https://github.com/TheImagingSource/tiscamera.git
### 2. Install build dependencies
    sudo ./scripts/install-dependencies.sh --compilation --runtime
### 3. Build tiscamera
    cd tiscamera
    mkdir build && cd build
    cmake -DBUILD_ARAVIS=ON -DBUILD_TOOLS=ON ..
    make
    sudo make install
### 4. Start the gige-daemon
    sudo systemctl daemon-reload
    sudo systemctl enable gige-daemon.service
    sudo systemctl start gige-daemon.service
    sudo systemctl status gige-daemon.service
### 5. Install aravis
    cd aravis-0.6.3
    ./configure
    make
    sudo make install
## Error program
/examples/python/04-list-formats.py
## Details for Error
    video/x-bayer rggb - 256x16 - Traceback (most recent call last):
    File "04-list-formats.py", line 72, in print_formats
    framerates = structure.get_value("framerate")
    TypeError: unknown type GstValueList
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
    File "04-list-formats.py", line 132, in <module>
    main()
    File "04-list-formats.py", line 126, in main
    print_formats(source)
    File "04-list-formats.py", line 76, in print_formats
    substr = fmt.to_string()[fmt.to_string().find("framerate="):]
    AttributeError: 'str' object has no attribute 'to_string'

TIS-Edgar, 2019-06-24:
Hi,

There was indeed a bug in the 04-list-formats.py example.
I pushed a fix to the master branch.
Please update your repository to the latest version of the master branch.

ryoji1415, 2019-06-25:
Hi Edgar,

After I tried to run by updated program, I had no error.
Thank you for your quickly support.

ryoji1415, 2019-06-25:
Hi Edgar,

Sorry, I overlooked one error at running updated sample program.
Please give me more help.

    video/x-bayer rggbTraceback (most recent call last):
    File "04-list-formats.py", line 132, in <module>
    main()
    File "04-list-formats.py", line 126, in main
    print_formats(source)
    File "04-list-formats.py", line 60, in print_formats
    width = structure.get_value("width")
    TypeError: unknown type GstIntRange

TIS-Edgar, 2019-06-25:
Hi there,

I missed checking the example with a GigE camera that has ranges.
Please pull and verify that it works. I am hopeful to have addressed all issues with this example.

ryoji1415, 2019-06-25:
Hi Edgar,

After I tried this example, there was no error in 04-list-formats.py.
I appreciate for your support.
But, I found other error like below in 05-set-format.py.
I want your help for this error.

    Traceback (most recent call last):
    File "05-set-format.py", line 96, in <module>
    sys.exit(main())
    File "05-set-format.py", line 54, in main
    structure.set_value("framerate", Gst.Fraction(30, 1))
    TypeError: object() takes no parameters

TIS-Edgar, 2019-06-26:
Hi,

I pushed a fix for this problem. The rest of the examples should not have these problems as they do less 'manual' work such as creating GstStructure objects.

If there are still problems, please let me know.

ryoji1415, 2019-06-26:
Hi,

There are still problem like below in 05-set-format.py.
I'm sorry, but please give me more help.

    Traceback (most recent call last):
    File "05-set-format.py", line 109, in <module>
    sys.exit(main())
    File "05-set-format.py", line 66, in main
    caps.append_structure(structure)
    TypeError: argument structure: Expected Gst.Structure, but got __main__._ResultTuple

TIS-Edgar, 2019-06-26:
Can you give me your camera model?

ryoji1415, 2019-06-26:
My camera model is DFK33GP1300.

TIS-Edgar, 2019-06-26:
Can you change the line:

    structure = structure.from_string(struc_string)

to 

    structure, end = structure.from_string(struc_string)

I will have to test a few different ubuntu versions to ensure that there is nothing else I am missing.

ryoji1415, 2019-06-27:
Thank you for your suggestion.
After changing, there is no error.
I can watch live streaming.
If I have some problem, I want your help again.

closed 2019-06-27 by TIS-Edgar