TheImagingSource/tiscamera #293: Can't take clearly color image

ryoji1415, 2019-06-27:
I want to save color image, but I capture only by monochrome like below because I don't know how to convert to color image.
By the way, in live streaming, I can see by color.
I try to realize it by reference to 07-appsink.py in python sample program.
Please tell me how to get color image from "buffer_map", "pixel_data" in 07-appsink.py and so on.
[image]

TIS-Edgar, 2019-06-28:
Hi there,

It seems like the data is stored in a wrong fashion thus losing information.
As the 07-appsink.py example states:

    # this is only one pixel
    # when dealing with formats like BGRx
    # pixel_data will have to consist out of
    # pixel_offset   => B
    # pixel_offset+1 => G
    # pixel_offset+2 => R
    # pixel_offset+3 => x
    pixel_data = buffer_map.data[pixel_offset]

This means to access a single pixel you will have to read 3 (sub)pixel in the form of the 3 bytes for R, G and B.

To find the mistake in the code I would have to see it.
Are you trying to manually save it or are you using a gstreamer pipeline to handle the process for you?

ryoji1415, 2019-06-28:
I am using a gstreamer pipeline.

TIS-Edgar, 2019-06-28:
What does that pipeline look like? How are you feeding images into it?

ryoji1415, 2019-06-28:
Maybe, I make you misunderstand due to my lack of explanation.
Monochrome image shown in first question was pulled out by stopping video gotten in sample code, 08-save-stream.py.
The pipeline used in my program is following:

    pipeline = Gst.parse_launch("tcambin name=source"
                                " ! videoconvert"
                                " ! appsink name=sink")

I want to save color image in this callback function.

    def callback(appsink, user_data):

      sample = appsink.emit("pull-sample")

      if sample:
        caps = sample.get_caps()
        gst_buffer = sample.get_buffer()

        try:
            (ret, buffer_map) = gst_buffer.map(Gst.MapFlags.READ)

            video_info = GstVideo.VideoInfo()
            video_info.from_caps(caps)

            stride = video_info.finfo.bits / 8

            pixel_offset = int(video_info.width / 2 * stride +
                               video_info.width * video_info.height / 2 * stride)
            pixel_data = buffer_map.data[pixel_offset]

            global framecount
            framecount += 1

        finally:
            gst_buffer.unmap(buffer_map)

      return Gst.FlowReturn.OK

main function of callback:

    sink = pipeline.get_by_name("sink")

    sink.set_property("emit-signals", True)

    user_data = " My save directory path "

    sink.connect("new-sample", callback, user_data)

    pipeline.set_state(Gst.State.PLAYING)

TIS-Edgar, 2019-06-28:
Do both pipeline have the same settings?
The example 08-save-stream.py sets the format to 

    "video/x-raw,format=BGRx,width=640,height=480,framerate=30/1"

Whereas 07-appsink.py does not set a format, thus using the tcambin will use the largest format available. That would certainly explain the video output you are seeing.

ryoji1415, 2019-07-01:
I changed the format in 08-save-stream.py for my camera to 

    " ! video/x-raw,format=BGRx,width=1280,height=1024,framerate=5/1"

ryoji1415, 2019-07-02:
Hi Edgar,

I'm sorry, but this is an urgent matter.
To meet the customer's request ASAP, I want your help for this matter.

TIS-Edgar, 2019-07-02:
What is your current status? Any progress?
What does your code look like exactly?
I need information to help.

ryoji1415, 2019-07-02:
I want to save color image regularly, so I make following program by reference to 07-appsink.py.

    import sys
    import gi
    import time
    import datetime
    import os
    import cv2
    import numpy as np
    
    gi.require_version("Tcam", "0.1")
    gi.require_version("Gst", "1.0")
    gi.require_version("GstVideo", "1.0")

    from gi.repository import Tcam, Gst, GstVideo

    framecount = 0

    def callback(appsink, user_data):
        sample = appsink.emit("pull-sample")

        if sample:

            caps = sample.get_caps()
            gst_buffer = sample.get_buffer()

            try:
                (ret, buffer_map) = gst_buffer.map(Gst.MapFlags.READ)

                video_info = GstVideo.VideoInfo()
                video_info.from_caps(caps)

                stride = video_info.finfo.bits / 8

                pixel_offset = int(video_info.width / 2 * stride +
                                   video_info.width * video_info.height / 2 * stride)

                pixel_data = buffer_map.data[pixel_offset]
                timestamp = gst_buffer.pts

                img_array = np.asarray(bytearray(buffer_map.data), dtype=np.uint8)
                img = img_array.reshape((video_info.height, video_info.width))

                global framecount

                save_name = "sample.png"
                save_path = os.path.join(user_data, save_name)
                cv2.imwrite(save_path, img)

                framecount += 1

            finally:
                gst_buffer.unmap(buffer_map)

        return Gst.FlowReturn.OK

    def main():

        Gst.init(sys.argv)  # init gstreamer
        serial = 24810049 # my camera serial

        pipeline = Gst.parse_launch("tcambin name=source"
                                " ! videoconvert"
                                " ! appsink name=sink")

        # test for error
        if not pipeline:
            print("Could not create pipeline.")
            sys.exit(1)

        # The user has not given a serial, so we prompt for one
        if serial is not None:
            source = pipeline.get_by_name("source")
            source.set_property("serial", serial)

        sink = pipeline.get_by_name("sink")

        # tell appsink to notify us when it receives an image
        sink.set_property("emit-signals", True)

        user_data = "/home/ryoji/tiscamera/examples/python/sample_data" # save folder path

        # tell appsink what function to call when it notifies us
        sink.connect("new-sample", callback, user_data)

        pipeline.set_state(Gst.State.PLAYING)

        print("Press Ctrl-C to stop.")

        try:
            while True:
                time.sleep(1)
        except KeyboardInterrupt:
            pass
        finally:
            pipeline.set_state(Gst.State.NULL)

    if __name__ == "__main__"
          main()

But, color image cannot be got due to  following error message.
So, I want to give how to get color image.

    Traceback (most recent call last):
    File "07-appsink.py", line 78, in callback
    img = img_array.reshape((video_info.height, video_info.width))
    ValueError: cannot reshape array of size 5242880 into shape (1024,1280)

TIS-Edgar, 2019-07-02:
The reshape call is incomplete.
This line should give you a better result:

    img = img_array.reshape((video_info.height, video_info.width, -1))

ryoji1415, 2019-07-02:
I can get color image!! Thank you very much!!
For my understanding, I want you tell me.
    
    img = img_array.reshape((video_info.height, video_info.width, -1))
Is this equal to below?

    img = img_array.reshape((video_info.height, video_info.width, 4))

TIS-Edgar, 2019-07-02:
If you look at the [numpy documentation for reshape](https://docs.scipy.org/doc/numpy/reference/generated/numpy.reshape.html):

    One shape dimension can be -1. In this case, the value is 
    inferred from the length of the array and remaining dimensions.

-1 basically says: Try to automatically determine the correct values.
4 says that each pixel is 4 bytes aka B G R x

If you explicitly request another format with different layout that value might change.
For example for BGR that value should be 3. If you work with greyscale images that value should be 1.

ryoji1415, 2019-07-02:
I understood, I really appreciate your kindly help.

closed 2019-07-04 by TIS-Edgar