TheImagingSource/tiscamera #294: FPS drops significantly when the shutter speed is set to below 16666 us

skywhale, 2019-06-28:
OS: Ubuntu 19.04
TIS version: 0.10.0_development/dbb13d0_rev_2525
Camera type: DFM 37UX178-ML

Command:
```
TCAM_LOG=DEBUG chrt --rr 99 tcam-capture
```

Using the `tcam-capture` tool, I noticed that camera FPS behaves strange at around the shutter speed of 16666 us. I'm taking those FPS measurements from the status bar of the tool, but we observed similar symptoms with our custom application that uses `rscam` Rust crate (thus bypassing gstreamer).

Shutter speed and FPS correlation
(all other parameters are set equal. auto gain and WB are off)
17000 us: ~58 FPS
16666 us: ~58 FPS
16650 us: ~50 FPS
16630 us: ~43 FPS
16600 us: ~38 FPS
16570 us: ~58 FPS
16000 us: ~33 FPS

We tested two 37UX178's and the issue happens with both. Suspecting something possibly wrong with clock speed computation? Could you try if you can reproduce the issue?

TIS-Edgar, 2019-06-28:
Hi,
We are unable to reproduce this issue with the specified commit or with the current master.

What hardware are you using? It may be a displaying issue.

skywhale, 2019-07-06:
Thanks Edgar.

This is our setup:
CPU: i9-9900K
GPU: GeForce 2080 Ti
Display: LG 24UD58-B over DisplayPort
USB cable: 3m 3.0 USB cable from iDS

Thanks for trying to reproduce the issue on your side. Did you use the same camera model?

We are going to try reproducing the issue with other workstations to see if this is something specific to our hardware setup.

TIS-Edgar, 2019-07-10:
We are still unable to reproduce this. The hardware and operating system should work fine.

You are running at 3,072×2,048 @ 60 fps correct?
Can you observe this when running qv4l2?

To ensure that the display is not a problem, run this pipeline:
`gst-launch-1.0 -v tcamsrc serial=SERIAL ! bayer2rgb ! videoconvert ! fpsdisplaysink video-sink=fakesink`
Disabling of auto exposure etc can be done at runtime via qv4l2.

TIS-Edgar, 2019-07-10:
Does it only happen when running with real time threads or always?

skywhale, 2019-07-19:
I'm sorry for my slow response. I'm back working on this issue!

Yes, we are running at 3,072×2,048 @ 60 fps. Some new findings:

1. It happens with or without real-time threads.

2. Using the gstreamer pipeline with fpsdisplaysink, we are still able to reproduce the issue. For record, these are the commands I ran.
```
gst-launch-1.0 -v tcamsrc serial=18910230 ! video/x-bayer,format=rggb,width=3072,height=2048,framerate=60/1 ! bayer2rgb ! videoconvert ! fpsdisplaysink video-sink=fakesink

# Change the shutter speed while running gstreamer
tcam-ctrl -p -s "Exposure Time (us)=16000" 18910230
tcam-ctrl -p -s "Exposure Time (us)=16666" 18910230
...
```

shutter speed (us) vs. FPS
[image]

3. I am not able to reproduce this with `qv4l2`, because the FPS is consistently below 30, no matter what we use for the shutter speed. I'm assuming this is a display issue.

4. I tried the same test scenario on two other workstations we have, and the issue was reproduced on a workstation with similar configuration (i9 & 2080 ti), but *not* on another workstation with different configuration (i7 & dual 1080 ti).

Based on these findings, I started to suspect this may be due to some scheduling / polling issue at V4L2 layer. I'm interested in hearing your insights.

TIS-Stefan, 2019-07-19:
Hello

I tested the camera and your commands too and I am not able to reproduce the issue.
I do not think, it is an issue of your computer. Also it is not an issue of the camera, I checke the same hardware. Therefore, I stumble about your point 4. 

What happens, if you do not use the bayer2rgb?
Or is there a limitation of data transfer on your USB controllers? As soon as you use longer exposure times, the frame rate goes under 60 fps and so less bytes per second are transfered. I must admit, that would be totally new to me.

Stefan

skywhale, 2019-07-19:
Thank you Stefan!

We nailed down the hardware difference between the workstations that reproduce the issue and that doesn't, and it turned out to be the USB3.0 PCIe board that we are using.

StarTech PEXUSB3S44V
https://www.startech.com/jp/en/Cards-Adapters/USB-3.0/Cards/PCI-Express-USB-3-Card-4-Dedicated-Channels-4-Port~PEXUSB3S44V

red: using a USB3.0 port on the motherboard
blue: using one of the 4 ports of StarTech PEXUSB3S44V
[image]

We are still very curious, though, where the sudden FPS drop comes from, just below the shutter speed of 16666 us. It's hard to believe that the transfer speed limit of the StarTech board is just exactly between 60FPS and 59FPS, but I guess it is actually the case??

We'll look into this a bit more. Your suggestion of turning off bayer2rgb sounds worth trying, as I'm still suspecting some timing issue.

skywhale, 2019-07-19:
Tested without `bayer2rgb` and `videoconvert`.

red: with `bayer2rgb` and `videoconvert`
blue: without `bayer2rgb` and `videoconvert`
[image]

I was first confused to see that FPS is higher *with* `bayer2rgb` and `videoconvert` compared to without, at shutter speed less than 16666us, but I guess that is due to the data loss happening and some recovery mechanism of `videoconvert`?

skywhale, 2019-07-19:
[image]

Polled usb bandwidth every 100ms using `usbtop`, and the result seems to indicate that FPS drop does come from data loss happening with the StarTech board. The blue bars are clustered at around 377 MB/s. Does the raw RGGB format used by the camera takes 1 byte per pixel?

TIS-Stefan, 2019-07-19:
Hello

bayer2rgb creates a lot of CPU load, therefore, I am also surprised, that the fps goes down.
Good to read, that the frame drop is the reason.
The 37U cameras provides raw data, which is indeed 1 byte per pixel on GRAY8 formats. 

I use USB 3.0 controllers with Intel and Fresco Logic controllers. They usually work fine.

The test is somewhat tricky. We must drop the incomplete frames in order to avoid segmentation faults.
I am not sure, whether the tcamsrc exports the frame drops, but I faintly remember, there was a property for incomplete image counter.

Stefan

TIS-Stefan, 2019-07-19:
Hi

The Startech board uses the Renesas/NEC - µPD720202 chipset. It is slower that current Intel and Fresco Login Chipsets.

Stefan

skywhale, 2019-07-19:
Thank you! Could you share the specific model # that you have used in the past?

TIS-Stefan, 2019-07-19:
You can try this one:
https://www.inateck.com/inateck-ktu3fr-2o2i-usb-3-0-pci-express-card.html

skywhale, 2019-07-19:
We'll give it a try!

Thank you for helping investigate the issue.

closed 2019-07-19 by skywhale