TheImagingSource/tiscamera #295: `tcamsrc` *V0.11.1* segfaults on `TCAM_LOG=WARNING` preventing us from obtaining debug log

jraygauthier, 2019-06-28:
A segmentation fault occur with V0.11.1 (and latest `master`) as soon as
`TCAM_LOG` is set over `WARNING` level (e.g.: `DEBUG`, `INFO` but not `ERROR`).

```bash
$ ./reproduce_bug_51_385_segfault_on_warning_log_nix.sh 
INFO:__main__:init_gstreamer()
INFO:__main__:====== Iteration 0 =======
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src.get_device_serials()
INFO:__main__:tcam_src.set_property("serial", 33710150)
105421     <WARNING> /build/source/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201)
105773     <WARNING> /build/source/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Override Scanning Mode'(199e257)
105823     <WARNING> /build/source/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Trigger Global Reset Release'(199e261)
INFO:__main__:tcam_src.set_state(Gst.State.NULL)
INFO:__main__:====== Iteration 1 =======
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src.get_device_serials()
INFO:__main__:tcam_src.set_property("serial", 33710150)
142146     <WARNING> /build/source/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201)
/home/rgauthier/dev/retnia/hw/camera_tools/reproduce_bug_51_385_segfault_on_warning_log.sh: line 6:   329 Segmentation fault      ${SCRIPT_DIR}/reproduce_bug_51_385_segfault_on_warning_log.py
```

A patch has been sumitted via PR: [Comment out a line that segfaults the app by jraygauthier · Pull Request #287 · TheImagingSource/tiscamera](https://github.com/TheImagingSource/tiscamera/pull/287).

When you apply the patch from this PR, the segfault no longer occur:

```bash
$ ./reproduce_bug_52_385_segfault_on_warning_log_nix_patched.sh
INFO:__main__:init_gstreamer()
INFO:__main__:====== Iteration 0 =======
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src.get_device_serials()
INFO:__main__:tcam_src.set_property("serial", 33710150)
INFO:__main__:tcam_src.set_state(Gst.State.NULL)
INFO:__main__:====== Iteration 1 =======
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src.get_device_serials()
INFO:__main__:tcam_src.set_property("serial", 33710150)
INFO:__main__:tcam_src.set_state(Gst.State.NULL)
INFO:__main__:====== Iteration 2 =======
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src.get_device_serials()
INFO:__main__:tcam_src.set_property("serial", 33710150)
INFO:__main__:tcam_src.set_state(Gst.State.NULL)
INFO:__main__:====== Iteration 3 =======
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src.get_device_serials()
INFO:__main__:tcam_src.set_property("serial", 33710150)
INFO:__main__:tcam_src.set_state(Gst.State.NULL)
```

Script bundle attached: [reproduce_bug_51_385_segfault_on_warning_log_scripts_bundle.zip].

Code from `reproduce_bug_51_385_segfault_on_warning_log.py`:

```py
#!/usr/bin/env python
import logging

try:
    from src.video.gstreamer import Tcam, Gst, GLib, GObject, init_gstreamer
except ModuleNotFoundError:
    # If we're here, it means that this is executed from outside of
    # our code base. Define missing symbols.
    import gi
    import sys

    gi.require_version("Tcam", "0.1")
    gi.require_version("Gst", "1.0")
    gi.require_version("GstVideo", "1.0")

    from gi.repository import Tcam, Gst, GLib, GObject  # noqa E402

    def init_gstreamer():
        Gst.init(sys.argv)

        # Force the loading of tcamsrc gstreamer plugin.
        tmp_tcamsrc = Gst.ElementFactory.make("tcamsrc")
        if tmp_tcamsrc is not None:
            tmp_tcamsrc.set_state(Gst.State.NULL)
        else:
            print('Could not load the `tcamsrc` gstreamer plugin')

LOGGER = logging.getLogger(__name__)

def setup_module() -> None:
    LOGGER.info("init_gstreamer()")
    init_gstreamer()  # Make sure gstreamer has been initialized.
    # init_gstreamer()  # Make sure gstreamer has been initialized.

def _make_tcam_src() -> Gst.Element:
    LOGGER.info("tcam_src = Gst.ElementFactory.make(\"tcamsrc\")")
    tcam_src = Gst.ElementFactory.make("tcamsrc")
    # It might be that gstreamer has not been initialized.
    assert tcam_src is not None
    LOGGER.info("tcam_src.get_device_serials()")
    connected_serials = tcam_src.get_device_serials()
    if not connected_serials:
        raise AssertionError("No connected camera!")
    first_serial = connected_serials[0]
    LOGGER.info("tcam_src.set_property(\"serial\", %s)", first_serial)
    tcam_src.set_property("serial", first_serial)

    return tcam_src

# Reproducing ASP00051-385.
def _reproduce_bug() -> None:
    # Ok, segfault. occurs about 1 run out of 2 sometimes less, sometimes more.
    #
    # It seems to be essential to have at least 4 iterations to get the
    # segfault.
    #
    # Note that this seems to segfault only when running on top of kernel 4.14
    # and not on 4.19.
    for i in range(0, 4):
        LOGGER.info("====== Iteration %s =======", i)
        LOGGER.info("tcam_src = Gst.ElementFactory.make(\"tcamsrc\")")
        tcam_src = _make_tcam_src()

        # Cleanup.
        LOGGER.info("tcam_src.set_state(Gst.State.NULL)")
        tcam_src.set_state(Gst.State.NULL)
        # NOTE: This is essential to reproduce the segfault as it forces the src
        # to be gced.
        del tcam_src

def main() -> None:
    logging.basicConfig(level="INFO")

    setup_module()
    _reproduce_bug()

main()

```

[reproduce_bug_51_385_segfault_on_warning_log_scripts_bundle.zip]: https://github.com/TheImagingSource/tiscamera/files/3339601/reproduce_bug_51_385_segfault_on_warning_log_scripts_bundle.zip

TIS-Edgar, 2019-07-01:
Hi,

I think your script finally allowed me to reproduce something.
There might actually be 2 separate issues at hand.
1. Message handling. The error you mentioned triggers the bug but is not the cause.
   The actual culprit is likely `send_log_to_bus` in gsttcamsrc.cpp
   I am still investigating a good fix. I will keep you updated.

2. The logging seems to have a timing issue.
To verify this on your end, could you change the function `_reproduce_bug` to include a sleep?

        LOGGER.info("tcam_src = Gst.ElementFactory.make(\"tcamsrc\")")
        tcam_src = _make_tcam_src()

        time.sleep(1)
        # Cleanup.
        LOGGER.info("tcam_src.set_state(Gst.State.NULL)")
        tcam_src.set_state(Gst.State.NULL)

Changing the state destroys the element before the creation thread in the background is able to finish.
I am not yet sure how to fix this.

TIS-Edgar, 2019-07-05:
I think this issue is solved.
I pushed a commit (d675b13e28f3f0f4087a59eb3e6d388777d62dda) to master that seems to solve the problem. At least your test scripts do not cause segfaults on my machine anymore.

Please verify that this is also the case on your machines.

The Pull Request you offered got rid of the line that triggered the bug. 
Background explanation (ignore if you do not care):
The tiscamera libraries have a static Logger object that handles the entire logging process.
This object has a callback function that the tcamsrc uses to get notified about critical issues.
The problem is that this static object is initialized once and only cleaned up when the libraries are closed. That means when a start/stop cycle runs the callback might still point at the previous iteration of the tcamsrc while it is actually the current iteration that should be called.
As to why it only triggered irregularly: I suspect the increase of the log level is enough to push the time delays to a degree that is sufficient to make the life cycle of gstreamer objects an issue.

This whole ordeal has to be fixed in a better manner but that will require some time, so consider this a "it works" fix.

jraygauthier, 2020-02-19:
This is effectively fix by d675b13e28f3f0f4087a59eb3e6d388777d62dda.

closed 2020-02-19 by jraygauthier