TheImagingSource/tiscamera #296: tiscamera V0.11.1 on linux kernel 4.14: "<ERROR>...V4l2Device.cpp:1557: Buffer has wrong size"

jraygauthier, 2019-06-28:
After the first iteration (pipeline creation, start and stop), I no longer
receive frames. Afterward, unless I disconnect / reconnect the camera, the
problem occurs immediatly.

Does not have these issues when running on kernel 4.19.

Interestingly, we have a bit of the reverse issue for a version:

 -  ASP00051-370: tiscamera V0.9.1: `tcam-ctrl -p` return no properties under
    linux kernel 4.19 nor under 5.0

I encountered this problem when I forgot to revert back to kernel 4.19
after toggle to 4.14 as a workaround for the ASP00051-370.

I'm not sure what is your goal in terms of supported kernel version? Is 4.14
deemed too old? 

I notice it is possible to get 0.9.1 working (with some tweaks)
on top of the 0.11.1 new dev rule system. 

Were 0.11.1 proper support for kernel
4.14 avaible, I would be able to safely switch between version by touching
only user space, no reboot required.

Note that this is minor for us as we most likely will move to 0.11.1 once
blocking issues will be fixed and are allowed to use newer kernels. 

However, this might affect other users stuck with older kernels.

Here's an except of what I have running on kernel 4.19:

```bash
$ ./reproduce_bug_51_384_hangs_on_set_state_null_nix.sh
INFO:__main__:init_gstreamer()
INFO:__main__:====== Iteration 0 =======
INFO:__main__:tcam_src = Gst.ElementFactory.make("tcamsrc")
INFO:__main__:tcam_src.get_device_serials()
INFO:__main__:tcam_src.set_property("serial", 33710150)
INFO:__main__:_configure_tcam_source: {use_soft_trig: True}
INFO:__main__:_make_pipeline
INFO:__main__:pipeline.set_state(Gst.State.PLAYING)
INFO:__main__:time.sleep(0.1)
INFO:__main__:------ Cycle 0 ------
INFO:__main__:time.sleep(0.08)
INFO:__main__:Trigger 0
INFO:__main__:tcam_src.set_tcam_property("Software Trigger", True)
INFO:__main__:time.sleep(0.15)
INFO:__main__:_on_new_frame_thread_safe: begin
INFO:__main__:_on_new_frame_thread_safe: new frame sent: {w: 2448, h: 2048}
INFO:__main__:_on_new_frame_thread_safe: end
INFO:__main__:time.sleep(0.003)
# ...
INFO:__main__:Trigger 5
INFO:__main__:tcam_src.set_tcam_property("Software Trigger", True)
INFO:__main__:time.sleep(0.14)
INFO:__main__:_on_new_frame_thread_safe: begin
INFO:__main__:_on_new_frame_thread_safe: new frame sent: {w: 2448, h: 2048}
INFO:__main__:_on_new_frame_thread_safe: end
INFO:__main__:time.sleep(0.003)
INFO:__main__:time.sleep(0.09)
INFO:__main__:------ Cycle 1 ------
INFO:__main__:time.sleep(0.08)
INFO:__main__:Trigger 0
INFO:__main__:tcam_src.set_tcam_property("Software Trigger", True)
INFO:__main__:time.sleep(0.15)
INFO:__main__:_on_new_frame_thread_safe: begin
INFO:__main__:_on_new_frame_thread_safe: new frame sent: {w: 2448, h: 2048}
INFO:__main__:_on_new_frame_thread_safe: end
INFO:__main__:time.sleep(0.003)
# ...
[... 288 lines cut ...]
INFO:__main__:------ Cycle 0 ------
INFO:__main__:time.sleep(0.08)
INFO:__main__:Trigger 0
INFO:__main__:tcam_src.set_tcam_property("Software Trigger", True)
INFO:__main__:time.sleep(0.15)
INFO:__main__:time.sleep(0.003)
INFO:__main__:Trigger 1
INFO:__main__:tcam_src.set_tcam_property("Software Trigger", True)
INFO:__main__:time.sleep(0.15)
INFO:__main__:time.sleep(0.003)
INFO:__main__:Trigger 2
INFO:__main__:tcam_src.set_tcam_property("Software Trigger", True)
INFO:__main__:time.sleep(0.15)
INFO:__main__:time.sleep(0.003)
INFO:__main__:Trigger 3
INFO:__main__:tcam_src.set_tcam_property("Software Trigger", True)
INFO:__main__:time.sleep(0.15)
INFO:__main__:time.sleep(0.003)
# ...
```

I have to disconnect / reconnect to camera to recover from the problem.

[reproduce_bug_51_384_hangs_on_set_state_null_scrit_bundle.zip](https://github.com/TheImagingSource/tiscamera/files/3340379/reproduce_bug_51_384_hangs_on_set_state_null_scrit_bundle.zip)
[Logs.zip](https://github.com/TheImagingSource/tiscamera/files/3340380/Logs.zip)

jraygauthier, 2019-07-02:
It seems that I forgot setting the trigger mode in the script / bundle above. Updated the bundle accordingly. Note that the bug is still present.

jraygauthier, 2020-02-27:
Now that #288 has been fixed, this is no longer an issue for us as we were able to move to default distro kernel 4.19. We're closing the matching issue in our bug tracker. @TIS-Edgar: should I close this one as well or is there any plan for tiscamera supporting previous version of the linux kernel?

TIS-Edgar, 2020-02-27:
If this issue is of no concern to you anymore, close it.

Ubuntu Release 16.04 LTS as the reference system for the tiscamera release 0.11.1 has the kernel v4.4. So that should be considered the minimal supported version.

jraygauthier, 2020-02-27:
Excellent. No longer an issue for us (kernel 4.19) Closing.

closed 2020-02-27 by jraygauthier