TheImagingSource/tiscamera #297: Is it possible to overcome ethernet disconnections

yoelk, 2019-06-28:
Hello,
We use the tiscamera package via python.
When the camera is connected and we're getting frames, I'd like to know if it's possible to overcome cable disconnections.
Meaning, if I disconnect the ethernet cable to the camera and connect it again, is it possible for the GST pipeline (or better said the `tcambin` source) to still be active? At the moment, if I do this, I stop getting frames and I get an error message on the GST bus (as explained here: https://github.com/TheImagingSource/tiscamera/issues/236)

So is there something I can do to revive the `tcambin` source, or do I have to stop it and re-open a new one?
Thanks,
Joel

TIS-Edgar, 2019-07-01:
Hi yoelk,

It depends on why you need this.

The backend needs to be reopened for the image retrieval to continue.
There is currently no code for this as this is considered a responsibility of the user application.

If you want your pipeline to be independent of the gst_state of the tcambin i'd suggest splitting the pipeline into to parts:

    tcambin ! capsfilter ! appsink
    appsrc ! <you pipeline>

The appsink callback simply forwards the gstbuffer to the appsrc, thus making image handling independent from image retrieval.

Would that suffice or are there other concerns that need to be addressed with this?

yoelk, 2019-07-01:
Thanks Edgar! 
If I understood correctly, I'll just need to restart the appsrc pipeline. I'll give it a try.

closed 2019-07-01 by yoelk