TheImagingSource/tiscamera #298: Extend FOV

eloibove, 2019-07-03:
Hi, I am currently using a DFK33UX252. The field of view I get at 640x480 with 'Override Scanning Mode'=5 is not enough for my application. I understand that to use the binning feature the size of the image has to be at most half of the sensor resolution, which in my case is 1024x768. Then, I think that with this resolution and binning by half the FOV should be enough. However, this resolution is not available... 
I was wondering if there is a way to use this resolution as in issue #183 . I think that this resolution is available in Windows but not in Linux, and I am using Ubuntu 16.04.
Thank you in advance.

TIS-Stefan, 2019-07-10:
Hi
Unfortunately V4L2 does support a limited list of resolutions, which must be saved in the USB descriptor of the camera, only. Therefore, not all possible resolutions are available. Unfortunately the USB descriptor has a limited size of memory.

Thus, we are not able to solve this issue to you needs.

Stefan

eloibove, 2019-07-10:
Hi Stefan, thank you anyway!

eloibove, 2019-09-24:
Hi Stefan, I was wondering if it is possible to change the available resolutions that are saved in the USB descriptor of the cameras since we don't need the 1920x1080 resolution and a resolution of 1024x768 would really be the solution to our problem. 

If it is possible, could you provide instructions to do it?

Thank you very much! 

Eloi

TIS-Stefan, 2019-09-25:
Hello Eloi

Is it possible, your company is already in contact about that by email with me? I have  same case with another programmer. Same camera, same request.
Which means, we are working on that.

Stefan

eloibove, 2019-09-25:
Hi Stefan. Yes, I contacted yesterday via email. It's good to know you are working on that! Please, keep me updated about any further news. 
Thank you very much,

Eloi

TIS-Stefan, 2019-09-25:
ok, lets close the case here.

closed 2019-09-25 by TIS-Stefan