TheImagingSource/tiscamera #301: Is there any way to use DFK23U445 with Indi driver

gbsdavid, 2019-07-21:
Hi, 

I got a DFK23U445 and plan tu use it for autoguiding under linux. 
The camera is fully fonctionning with tis-camera software or piping with gst-launch-1.0 tcambin ! videoconvert ! ximagesink but I need it to be recognize by Ekos or PHD2.

I've readed that there is two indi driver : V4L2 and DMK for imaging source camera but none of them are  working for me.

Did someone know how to do that ?

Regards,

TIS-Stefan, 2019-07-22:
Hello 

The DFK23U445 is v4l2 compatible, therefore PHD2 should be able to see it. I suggest to ask the PHD authors about this issue.

I am sorry for not having a better answer

Stefan

gbsdavid, 2019-07-22:
Hi,

I finally found a solution on the indi forum, it was former on the imaging source forum but this forum seems to have disapear ?
If I add this 
echo "199e 8412" | sudo tee /sys/bus/usb/drivers/uvcvideo/new_id

the uvc driver is good and the camera is now fully recognize by indi.

closed 2019-11-29 by TIS-Stefan