TheImagingSource/tiscamera #302: What is the shutter speed of imx178 and can it be changed?

marcel1991, 2019-07-21:
I am wondering how I can find out the shutter speed/rolling shutter time, so the time between the first image row is captured and the last. Does this depend on the frame rate? So if I use a smaller framerate this time is longer? Or is it always as fast as possible? Does it depend on the exposure time somehow?

And what is the actual shutter time for the imx178 sensor?

TIS-Stefan, 2019-07-22:
Hello

You can read the exposure time from the camera. 
If the camera is connected to an USB 3.0 controller, the exposure time determines the time between first and last row. If the camera is USB 37U model and connected to an USB 2.0 port, the time between first and last row is determined by frame rate.

> And what is the actual line time for the imx178 sensor when I use it in trigger mode?
The time you have set. Or what the automatic sets.

Your questions are somewhat general, I miss
- exact camera model
- software used

You may install our repository and use tcam-capture. This should answer your questions regarding exposure time.

Stefan

marcel1991, 2019-07-22:
So the exposure time is the time between the first row and the last row, as well? I thought exposure time is the time each row is exposed to the scene. Is the same timing used for USB3 cameras?

I use the tiscamera repository examples. There I can set the exposure of course. But until now I thought this would only change the time each row is exposed to the scene and not the timeshift between each row for the rolling shutter.

TIS-Stefan, 2019-07-23:
Of course the exposure times sum up per line. The only property to lower the rolling shutter effect is lower exposure time until the line delivery to computer exceeds the speed of USB 3.0 and sensor.

TIS-Stefan, 2019-11-29:
"is lower exposure time" I am not sure, whether this is correct.
The rolling shutter effect is determined by the frame rate, because this specifices the time between first and last line being read out. The higher the frame rate, the short the time span between first and last line. 
Thus, highest frame rate gives less rolling shutter effect.
I am sorry, if my previous answer was confusing.

closed 2019-11-29 by TIS-Stefan