TheImagingSource/tiscamera #303: gstreamer crash on list properties

midjji, 2019-07-22:
ubuntu 18.04, 
followed the build instructions, installed with root
gst-launch-1.0 tcambin ! videoconvert ! ximagesink
works 

plugin 1 usb tis camera. dfk37bux178, or dmk33ux252
python 00-list-devices.py   
sys:1: Warning: cannot register existing type 'ArvInterface'
sys:1: Warning: cannot add private field to invalid (non-instantiatable) type '<invalid>'
sys:1: Warning: g_once_init_leave: assertion 'result != 0' failed
sys:1: Warning: g_type_register_static: assertion 'parent_type > 0' failed
sys:1: Warning: g_object_new_with_properties: assertion 'G_TYPE_IS_OBJECT (object_type)' failed
then hangs forever. Same for the c example.

midjji, 2019-07-22:
warnings for the c
(00-list-devices:5313): GLib-GObject-WARNING **: 16:56:19.040: cannot register existing type 'ArvInterface'

(00-list-devices:5313): GLib-GObject-WARNING **: 16:56:19.040: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(00-list-devices:5313): GLib-CRITICAL **: 16:56:19.040: g_once_init_leave: assertion 'result != 0' failed

(00-list-devices:5313): GLib-GObject-CRITICAL **: 16:56:19.040: g_type_register_static: assertion 'parent_type > 0' failed

(00-list-devices:5313): GLib-GObject-WARNING **: 16:56:19.040: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(00-list-devices:5313): GLib-CRITICAL **: 16:56:19.040: g_once_init_leave: assertion 'result != 0' failed

(00-list-devices:5313): GLib-GObject-CRITICAL **: 16:56:19.040: g_object_new_with_properties: assertion 'G_TYPE_IS_OBJECT (object_type)' failed

midjji, 2019-07-22:
build without aravis, same result

TIS-Stefan, 2019-07-23:
Hi

I can reproduce the issue. But if you start the gige-daemon wit
`gige-daemon start`
It works fine. 
(gige-daemon does not point to gige camera only, it also handles USB cameas.)

Stefan

midjji, 2019-07-25:
workaround works.

closed 2019-11-29 by TIS-Stefan