TheImagingSource/tiscamera #304: odd properties for dfk 37bux178, incorrect format?

midjji, 2019-07-25:
querying for the properties of the dfk 37bux178 I get the following. 
Two problems
1) the camera should support much higher framerates
2) attempting to use the raw bayers i.e. rggb or rggb16 simply makes it hang. regardless of resolution and framerate. 

RGBx works though, both for high and low framerates. 

size = 640 x 480 color formats = {rggb,} frame rates = {60/1,30/1,15/1,5/1,1/1,}
size = 1920 x 1080 color formats = {rggb,} frame rates = {15/1,5/1,1/1,}
size = 2048 x 2048 color formats = {rggb,} frame rates = {5/1,1/1,}
size = 3072 x 2048 color formats = {rggb,} frame rates = {5/1,1/1,}
size = 640 x 480 color formats = {rggb16,} frame rates = {30/1,15/1,5/1,1/1,}
size = 1920 x 1080 color formats = {rggb16,} frame rates = {7/1,5/1,1/1,}
size = 2048 x 2048 color formats = {rggb16,} frame rates = {2/1,1/1,}
size = 3072 x 2048 color formats = {rggb16,} frame rates = {2/1,1/1,}
size = 640 x 480 color formats = {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR,} frame rates = {60/1,30/1,15/1,5/1,1/1,}
size = 1920 x 1080 color formats = {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR,} frame rates = {15/1,5/1,1/1,}
size = 2048 x 2048 color formats = {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR,} frame rates = {5/1,1/1,}
size = 3072 x 2048 color formats = {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR,} frame rates = {5/1,1/1,}

There is also an error in dmesg, regarless of option used: 

[260884.993584] ------------[ cut here ]------------
[260884.993588] Unknown pixelformat 0x00000000
[260884.993636] WARNING: CPU: 7 PID: 21715 at /build/linux-6ZmFRN/linux-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1291 v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
[260884.993637] Modules linked in: cpuid btrfs zstd_compress xor raid6_pq ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs libcrc32c uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media rfcomm cmac bnep binfmt_misc snd_hda_codec_hdmi nls_iso8859_1 intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper cryptd intel_cstate snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel eeepc_wmi asus_wmi wmi_bmof snd_hda_codec sparse_keymap snd_hda_core snd_hwdep snd_pcm arc4 snd_seq_midi snd_seq_midi_event snd_rawmidi rtl8821ae btcoexist snd_seq rtl_pci rtlwifi mac80211 snd_seq_device snd_timer snd intel_rapl_perf input_leds cfg80211 btusb btrtl btbcm
[260884.993727]  btintel bluetooth mei_me soundcore serio_raw ecdh_generic mei shpchp wmi nvidia_uvm(POE) acpi_pad mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops psmouse drm r8169 ahci ipmi_devintf mii libahci ipmi_msghandler video
[260884.993784] CPU: 7 PID: 21715 Comm: tisconfig Tainted: P        W  OE    4.15.0-47-generic #50-Ubuntu
[260884.993786] Hardware name: ASUSTeK COMPUTER INC. G11CB/G11CB, BIOS 0601 09/08/2015
[260884.993798] RIP: 0010:v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
[260884.993801] RSP: 0018:ffffac2806377c30 EFLAGS: 00010282
[260884.993805] RAX: 0000000000000000 RBX: ffffac2806377d68 RCX: 0000000000000006
[260884.993807] RDX: 0000000000000007 RSI: 0000000000000086 RDI: ffff8e0b43dd6490
[260884.993810] RBP: ffffac2806377c40 R08: 0000000000000001 R09: 000000000000176b
[260884.993812] R10: ffff8e091844be00 R11: 0000000000000000 R12: ffff8e0a3b847400
[260884.993814] R13: ffffffffc1eca900 R14: ffff8e091844be40 R15: ffff8e0a3b847400
[260884.993818] FS:  00007faddfbe2780(0000) GS:ffff8e0b43dc0000(0000) knlGS:0000000000000000
[260884.993820] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[260884.993823] CR2: 00005581887b5048 CR3: 000000046436c002 CR4: 00000000003606e0
[260884.993825] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[260884.993828] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[260884.993829] Call Trace:
[260884.993844]  v4l_enum_fmt+0x74/0x110 [videodev]
[260884.993856]  __video_do_ioctl+0x370/0x380 [videodev]
[260884.993863]  ? _cond_resched+0x19/0x40
[260884.993874]  video_usercopy+0x24d/0x640 [videodev]
[260884.993884]  ? video_ioctl2+0x20/0x20 [videodev]
[260884.993893]  ? enqueue_task_fair+0xa1/0x7f0
[260884.993903]  video_ioctl2+0x15/0x20 [videodev]
[260884.993912]  v4l2_ioctl+0xc8/0xf0 [videodev]
[260884.993924]  do_vfs_ioctl+0xa8/0x630
[260884.993931]  ? _do_fork+0x153/0x400
[260884.993937]  ? get_unused_fd_flags+0x30/0x40
[260884.993942]  SyS_ioctl+0x79/0x90
[260884.993949]  do_syscall_64+0x73/0x130
[260884.993955]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[260884.993959] RIP: 0033:0x7faddc6875d7
[260884.993961] RSP: 002b:00007ffc4b9e29c8 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[260884.993965] RAX: ffffffffffffffda RBX: 0000000000000004 RCX: 00007faddc6875d7
[260884.993967] RDX: 00007ffc4b9e2ad0 RSI: 00000000c0405602 RDI: 000000000000000c
[260884.993969] RBP: 00000000c0405602 R08: 00007fadd332c4cd R09: 000055818855a010
[260884.993971] R10: 0000000000000000 R11: 0000000000000246 R12: 00007ffc4b9e2ad0
[260884.993974] R13: 000000000000000c R14: 00000000c0405602 R15: 00007ffc4b9e2ad0
[260884.993976] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00 3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 d7 22 ea c1 e8 5f e9 df c1 <0f> 0b 80 7b 0c 00 0f 85 e2 f3 ff ff 8b 43 2c 48 c7 c1 7e 16 ea 
[260884.994053] ---[ end trace e914dd22b4461ac8 ]---

midjji, 2019-07-25:
Ok so part of the issue is that the device connects as if it was usb2 instead of three. 
Why does it do this? Is the rest of the issue just some half broken fall back mode? 

I'll try to replace the cable. Replacing the device with another identical device garners the same result.

TIS-Stefan, 2019-07-25:
If the camera connects as USB 2.0 only, then it detected your USB port being an USB 2.0 port. Thus, the first question is, which computer brand and model you use.
Also you can run 
lsusb -t
That shows connections and their speed.

midjji, 2019-07-25:
The port is usb 3, based on both lsusb and  the performance of other tis
cameras on that port, transfer speed etc.

However lsusb says it is connected to a usb2 device when I plug it in.

Wondering if the cable could be bad, but I tried two.

Btw, I'm trying to synch a bunch of cameras using external trigger but it
seems the gstreamer component, or something earlier is silently dropping
frames. The effect is alleviated if I reduce resolution well below
maximum/cameras, but I can't be sure it doesn't happen as the dropping is
silent. Is there a way to avoid that, or just any lower level API I could
use?

I'd really like to get the hw frameid, is there anyway to get/set that at
all in Linux? I can't be the only one asking for this. The cameras are
basically useless for slam, isometry, stereo etc without it.

I just want to be able to select camera by serial, set the minimum image
properties, use hw trigger and read frames with id and getting to know when
images are dropped. Should I even use the gstreamer for this or is there
another API?

Oh and when I'm wishing, is there a way to tell the camera to use more
error correction, or even TCP style transfer at lower throughput? I'm not
sure, but it seems to be a bit noisy.

midjji, 2019-07-25:
I'll check the specific mb model and what usbsubsystem it uses for usb
first thing tomorrow, but givent that three other tis cams show no problem
with that port I doubt it's on the computer side.

I just realize I heard the same thing from a customer a while back. They
got a different set of properties for some of the laptops using the same
cameras, which is what I'm seeing. It didn't seem laptop model dependent
though and they just tossed those specific laptops. Makes me wonder if
there is some voltage or margin which isn't quite meet somewhere.

TIS-Stefan, 2019-07-26:
Which computer brand and model do you use?
Do you use our USB 3.0 C cables?
Currently the time stamp and frame number can not be queried from the camera, because UVC and v4l2 do not support that.
(Annotation: the 37U series has no frame counter. It toggles 0 and 1 between frames only)

Stefan

midjji, 2019-07-27:
Tried a third cable which makes the camera be detected as usb3, but its
still not working correctly.

Motherboard: Asus h170-1/G11C8/DP-M8,

/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    |__ Port 1: Dev 25, If 0, Class=Miscellaneous Device, Driver=, 5000M
    |__ Port 1: Dev 25, If 1, Class=Miscellaneous Device, Driver=, 5000M
    |__ Port 1: Dev 25, If 2, Class=Miscellaneous Device, Driver=, 5000M
    |__ Port 1: Dev 25, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 25, If 4, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 3: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid,
1.5M
    |__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid,
1.5M
    |__ Port 7: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 7: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M

Now the properties are the expected:
size = 640 x 480 color formats = {rggb,} frame rates =
{280/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1,}
size = 1920 x 1080 color formats = {rggb,} frame rates =
{130/1,120/1,60/1,30/1,15/1,5/1,1/1,}
size = 2048 x 2048 color formats = {rggb,} frame rates =
{70/1,60/1,30/1,15/1,5/1,1/1,}
size = 3072 x 2048 color formats = {rggb,} frame rates =
{60/1,30/1,15/1,5/1,1/1,}
size = 640 x 480 color formats = {rggb16,} frame rates =
{280/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1,}
size = 1920 x 1080 color formats = {rggb16,} frame rates =
{90/1,60/1,30/1,15/1,5/1,1/1,}
size = 2048 x 2048 color formats = {rggb16,} frame rates =
{45/1,30/1,15/1,5/1,1/1,}
size = 3072 x 2048 color formats = {rggb16,} frame rates =
{30/1,15/1,5/1,1/1,}
size = 640 x 480 color formats = {RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR,}
frame rates = {280/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1,}
size = 1920 x 1080 color formats =
{RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR,} frame rates =
{130/1,120/1,60/1,30/1,15/1,5/1,1/1,}
size = 2048 x 2048 color formats =
{RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR,} frame rates =
{70/1,60/1,30/1,15/1,5/1,1/1,}
size = 3072 x 2048 color formats =
{RGBx,xRGB,BGRx,xBGR,RGBA,ARGB,BGRA,ABGR,} frame rates =
{60/1,30/1,15/1,5/1,1/1,}

However, the bayer modes rggb and rggb16 do not work.

[427955.022404] ------------[ cut here ]------------
[427955.022407] Unknown pixelformat 0x00000000
[427955.022450] WARNING: CPU: 2 PID: 30595 at
/build/linux-6ZmFRN/linux-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1291
v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
[427955.022451] Modules linked in: cdc_acm cpuid btrfs zstd_compress xor
raid6_pq ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs libcrc32c uvcvideo
videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev
media rfcomm cmac bnep binfmt_misc snd_hda_codec_hdmi nls_iso8859_1
intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm
irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc
aesni_intel aes_x86_64 crypto_simd glue_helper cryptd intel_cstate
snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel eeepc_wmi
asus_wmi wmi_bmof snd_hda_codec sparse_keymap snd_hda_core snd_hwdep
snd_pcm arc4 snd_seq_midi snd_seq_midi_event snd_rawmidi rtl8821ae
btcoexist snd_seq rtl_pci rtlwifi mac80211 snd_seq_device snd_timer snd
intel_rapl_perf input_leds cfg80211 btusb
[427955.022547]  btrtl btbcm btintel bluetooth mei_me soundcore serio_raw
[... 41 lines cut ...]
[427955.022746]  ? get_unused_fd_flags+0x30/0x40
[427955.022752]  SyS_ioctl+0x79/0x90
[427955.022758]  do_syscall_64+0x73/0x130
[427955.022764]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[427955.022768] RIP: 0033:0x7f00a68385d7
[427955.022770] RSP: 002b:00007ffcd17daf88 EFLAGS: 00000246 ORIG_RAX:
0000000000000010
[427955.022774] RAX: ffffffffffffffda RBX: 0000000000000004 RCX:
00007f00a68385d7
[427955.022776] RDX: 00007ffcd17db090 RSI: 00000000c0405602 RDI:
000000000000000c
[427955.022779] RBP: 00000000c0405602 R08: 00007f009d4dd4cd R09:
0000000000000000
[427955.022781] R10: 0000560904027e48 R11: 0000000000000246 R12:
00007ffcd17db090
[427955.022783] R13: 000000000000000c R14: 00000000c0405602 R15:
00007ffcd17db090
[427955.022786] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00
3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 d7 22 ea c1 e8 5f e9 df c1
<0f> 0b 80 7b 0c 00 0f 85 e2 f3 ff ff 8b 43 2c 48 c7 c1 7e 16 ea
[427955.022862] ---[ end trace e914dd22b4461b2b ]---

The RGBx mode works, but only at around 5hz for full resolution.

Is the camera misrepresenting what it supports?

TIS-Stefan, 2019-07-29:
You are right. The camera detects the USB connection as USB 2.0. 

Did we not already discussed case somewhere else?

Stefan

TIS-Stefan, 2019-07-29:
> However, the bayer modes rggb and rggb16 do not work.
What does this mean? Since the camera has this modes only, I understand you do not get any image at all. The camera provides bayer raw data only. The RGB colors are created by software using the GStreamer bayer2rgb module.
What did you do for testing the the formats?
I have the suspect, there is an issue between the camera and your USB 3.1 Gen 2 controller. 
Can you show the output of "lspci"? Thus I can see the chipset of the controller

I must admit, I am guessing only and I have no idea, what really causes the issue.
In order to exclude the USB cables, you may try to use the camera on another computer, if possible. If that works fine, then it points to the issue between camera and USB 3.1 Gen 2 controller.

Stefan

midjji, 2019-07-29:
I use the tiscamera gstreamer api to list which formats the camera
supports.
It answers with the list above. I then use set property to set image
format.
I append the code I used to query the gst_element_factory_make("tcambin",
"tcambin-source")
for the properties. Could there be a missmatch between what is exposed by
tcambin and what the camera supports?
Is there an alternative to using the intermediate gstreamer layer?

One more possibly related issue,
When I set external trigger the external trigger fps must be lower than the
fps property set.
Which is confusing since once external trigger is set, why would what
framerate used for non trigger mode matter at all?

00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen
Core Processor Host Bridge/DRAM Registers (rev 07)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen
Core Processor PCIe Controller (x16) (rev 07)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev
06)
00:14.0 USB controller: Intel Corporation 100 Series/C230 Series Chipset
Family USB 3.0 xHCI Controller (rev 31)
00:16.0 Communication controller: Intel Corporation 100 Series/C230 Series
Chipset Family MEI Controller #1 (rev 31)
00:17.0 RAID bus controller: Intel Corporation SATA Controller [RAID mode]
(rev 31)
00:1c.0 PCI bridge: Intel Corporation 100 Series/C230 Series Chipset Family
PCI Express Root Port #5 (rev f1)
00:1c.5 PCI bridge: Intel Corporation 100 Series/C230 Series Chipset Family
PCI Express Root Port #6 (rev f1)
00:1c.6 PCI bridge: Intel Corporation 100 Series/C230 Series Chipset Family
PCI Express Root Port #7 (rev f1)
00:1d.0 PCI bridge: Intel Corporation 100 Series/C230 Series Chipset Family
PCI Express Root Port #9 (rev f1)
00:1f.0 ISA bridge: Intel Corporation H170 Chipset LPC/eSPI Controller (rev
31)
00:1f.2 Memory controller: Intel Corporation 100 Series/C230 Series Chipset
Family Power Management Controller (rev 31)
00:1f.3 Audio device: Intel Corporation 100 Series/C230 Series Chipset
Family HD Audio Controller (rev 31)
00:1f.4 SMBus: Intel Corporation 100 Series/C230 Series Chipset Family
SMBus (rev 31)
01:00.0 3D controller: NVIDIA Corporation GK110BGL [Tesla K40c] (rev a1)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd.
RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 11)
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821AE
802.11ac PCIe Wireless Network Adapter
04:00.0 USB controller: ASMedia Technology Inc. ASM1142 USB 3.1 Host
Controller

midjji, 2019-07-29:
gst_element_set_state(tcambin_, GST_STATE_READY);

    gst_element_get_state(tcambin_, nullptr, nullptr, GST_CLOCK_TIME_NONE);

    std::vector<VideoFormatCaps> ret;

    GstPad *pad = gst_element_get_static_pad(tcambin_, "src");

    assert(pad);

    GstCaps *caps = gst_pad_query_caps(pad, nullptr);

    assert(caps);

    for(uint i = 0; i < gst_caps_get_size(caps); ++i)

    {

        VideoFormatCaps fmt;

        GstStructure* s = gst_caps_get_structure(caps, i);

        if (s)

        {

            if (!g_strcmp0(gst_structure_get_name(s), "ANY"))

                continue;

            const GValue *width = gst_structure_get_value(s, "width");

            const GValue *height = gst_structure_get_value(s, "height");

            assert(width && height);

            if (G_VALUE_HOLDS_INT(width))

            {

                assert(G_VALUE_HOLDS_INT(height));

                fmt.size.width = g_value_get_int(width);

                fmt.size.height = g_value_get_int(height);

            }

            else if (GST_VALUE_HOLDS_INT_RANGE(width))

            {

                assert(GST_VALUE_HOLDS_INT_RANGE(height));

                fmt.size_min.width = gst_value_get_int_range_min(width);

                fmt.size_min.height = gst_value_get_int_range_min(height);

                fmt.size_max.width = gst_value_get_int_range_max(width);

                fmt.size_max.height = gst_value_get_int_range_max(height);

            }

            else

            {

                assert(false && "Invalid or missing width/height");

            }

            // Handle color formats

            const GValue *format = gst_structure_get_value(s, "format");

            // TODO: Support jpeg

            if (format == nullptr)

                continue;

            if (G_VALUE_HOLDS_STRING(format))

            {

fmt.formats.push_back(std::string(gst_structure_get_string(s,
"format")));

            }

            else if (GST_VALUE_HOLDS_LIST(format))

            {

                for (uint i=0; i < gst_value_list_get_size(format); i++)

                {

                    const GValue *val = gst_value_list_get_value(format, i);

                    assert(val && G_VALUE_HOLDS_STRING(val));

                    fmt.formats.push_back(std::string(g_value_get_string(val)));

                }

            }

            else

            {

                assert(FALSE && "Invalid or missing format");

            }

            // Handle frame rates

            const GValue *framerates = gst_structure_get_value(s, "framerate");

            assert(framerates);

            if (GST_VALUE_HOLDS_LIST(framerates))

            {

                for (uint i=0; i < gst_value_list_get_size(framerates); i++)

                {

                    const GValue *val = gst_value_list_get_value(framerates, i);

                    assert(val && GST_VALUE_HOLDS_FRACTION(val));

                    FrameRate rate;

                    rate.numerator = gst_value_get_fraction_numerator(val);

                    rate.denominator = gst_value_get_fraction_denominator(val);

                    fmt.framerates.push_back(rate);

                }

            }

            else if (GST_VALUE_HOLDS_FRACTION_RANGE(framerates))

            {

                const GValue *min =
gst_value_get_fraction_range_min(framerates);

                const GValue *max =
gst_value_get_fraction_range_max(framerates);

                assert(min && max);

                fmt.framerate_min.numerator =
gst_value_get_fraction_numerator(min);

                fmt.framerate_min.denominator =
gst_value_get_fraction_denominator(min);

                fmt.framerate_max.numerator =
gst_value_get_fraction_numerator(max);

                fmt.framerate_max.denominator =
gst_value_get_fraction_denominator(max);

            }

            else

            {

                assert(FALSE && "Invalid or missing framerate");

            }

            ret.push_back(fmt);

        }

    }

midjji, 2019-07-29:
Just to be clear, setting image format to RGBx gives me a 8 bit rgb image,
converted by gstreamer I guess?

TIS-Stefan, 2019-07-30:
First of all, please do not put three issues at once into one text. I wont be able to handle this.

> tcambin-source")
> for the properties. Could there be a missmatch between what is exposed by
> tcambin and what the camera supports?
> 

No.

`Is there an alternative to using the intermediate gstreamer layer?`
No. It is exactly as specified  in Gstreamer documentation

> One more possibly related issue,
> When I set external trigger the external trigger fps must be lower than the
> fps property set.
> Which is confusing since once external trigger is set, why would what
> framerate used for non trigger mode matter at all?

This is a typical misunderstanding. The frame rate determines the speed the sensor an generate images. If you set the frame rate to 10 fps, then it is clocked for creating a maximum speed of 10 frames, not faster.  I hope, that explains this.

> Just to be clear, setting image format to RGBx gives me a 8 bit rgb image,
> converted by gstreamer I guess?
 
Yes. The bayer2rgb GStreamer module for debayering is inserted automatically by the tcambin. 

Stefan

midjji, 2019-08-01:
Allright, single issue: setting rggb or rggb16, causes camera setup to fail
v4l to throw invalid format and no images to come.

Setting rgbx in full resolution at max framerate works fine with the new
cable. But I want the raw.

The names of the image format in the datasheet for the camera  are
different from what it reports, I'll try them today. Can you replicate the
issue?

midjji, 2019-08-01:
the issue was that I set video/x-raw instead of video/x-bayer

closed 2019-08-01 by midjji

TIS-Stefan, 2019-08-05:
Hello

I made an article at https://theimagingsource.deskpro.com/en/kb/articles/linux-set-video-formats-in-camera-and-memory that explains, how to set video formats for color and Bayer Raw data using GStreamer and the tcambin. 
I hope, it makes things somewhat better to understand.

Stefan