TheImagingSource/tiscamera #305: Unifying of python module installation path leads to bitbake build fails

aobergschwandtner, 2019-08-01:
The changes in the commits d82f9e1b7a03ab3c5c8096c3f50f0ad0067aad3d and dc0708658e52e07d335f98ef0fea29bbd6e7e90e seem to lead to invalid paths for the python modules which results in failed builds with bitbake.

If I revert both commits, the build works fine again.

TIS-Edgar, 2019-08-02:
Hi,

Could you provide me with your bitbake setup so that I can reproduce this issue?

aobergschwandtner, 2019-08-06:
Hi Edgar,
sorry for my delayed answer regarding this topic.
Please find attached by bitbake recipe which hopefully helps you to reproduce.

Best regards,
Andreas

Hi,

Could you provide me with your bitbake setup so that I can reproduce this issue?

—

# Copyright (C) 2017-2019 SKIDATA AG
# Richard Leitner (LERI) <richard.leitner@skidata.com>

DESCRIPTION = "The Imaging Source Gstreamer 1.0 Plugins"
HOMEPAGE = "http://www.theimagingsource.com/"
LICENSE = "Apache-2.0"
SECTION = ""
DEPENDS = "gstreamer1.0 gstreamer1.0-plugins-base libtinyxml eudev libusb pkgconfig libzip"
PR = "r0"
PV = "0.11.1.sd01"

inherit pkgconfig cmake gobject-introspection

SRC_URI = "${SD_ELINUX_GIT_MIRROR}/tiscamera.git;branch=master;protocol=ssh;name=git"

SRCREV_git = "v-tiscamera-${PV}"

LIC_FILES_CHKSUM = "file://LICENSE;md5=76f9b634bc5499ed996452ef6f4456dc"

# tiscamera build configuration
EXTRA_OECMAKE = "\
	-DBUILD_ARAVIS=OFF \
	-DBUILD_GST_1_0=ON \
	-DBUILD_TOOLS=ON \
	-DBUILD_V4L2=ON \
	"

# workaround for invalid/not detected resources
EXTRA_OECMAKE += "\
	-DINTROSPECTION_SCANNER=${WORKDIR}/recipe-sysroot/usr/bin/g-ir-scanner-wrapper \
	-DINTROSPECTION_COMPILER=${WORKDIR}/recipe-sysroot/usr/bin/g-ir-compiler-wrapper \
	-DINTROSPECTION_GENERATE=${WORKDIR}/recipe-sysroot/usr/bin/g-ir-generate \
	"

S = "${WORKDIR}/git"

PACKAGES += "tiscamera-udev tiscamera-tools tiscamera-tools-python"
FILES_tiscamera-udev = "\
	/etc/udev/rules.d \
	/usr/bin/tcam-uvc-extension-loader \
	"
FILES_tiscamera-tools = "\
	/usr/bin \
	/usr/share/applications \
	"
FILES_tiscamera-tools-python = "/usr/lib/python3.7"

# manually specify -dev files because otherwise *.gir is included here.
FILES_${PN}-dev = "\
	/usr/include \
	/usr/lib/libtcam-algorithms.so \
	/usr/lib/libtcam-dfk73.so \
	/usr/lib/libtcamprop.so \
	/usr/lib/libtcam.so \
	"
FILES_${PN} = "\
	/usr/lib/girepository-1.0/ \
	/usr/lib/gstreamer-1.0/libgst* \
	/usr/lib/libt* \
	/usr/lib/pkgconfig \
	/usr/lib/tcam-0 \
	/usr/share/gir-1.0/*.gir \
	/usr/share/theimagingsource \
	/usr/share/tiscamera/images \
	/usr/share/uvcdynctrl/ \
	"

# we need the dev-so skip for /usr/lib/tcam-0/libtcam-v4l2.so
INSANE_SKIP_${PN} = "dev-so"

TIS-Edgar, 2019-08-08:
Hi,

I am not sure why the install path is incorrect when using bitbake.
Without the commits you mentioned at least tcam-gigetool would still not install correct.

You can set the variable `TCAM_INSTALL_PYTHON3_MODULES` to overwrite any auto determined paths. That should allow you to at least continue without problems.

closed 2019-11-29 by TIS-Stefan