TheImagingSource/tiscamera #307: Gray8

vishal-prgmr, 2019-08-14:
Greetings,

Using the simple snap c++ example for capturing images using a DFK 33GX273. I cannot set the format to GRAY8 in `cam.set_capture_format("BGRx", FrameSize{640,480}, FrameRate{30,1});
` . Only BGRx works. 

Anything I should watch out for?

TIS-Stefan, 2019-08-14:
hello 
Unfortunately this is correct. 
Why do you want to use GRAY8?

vishal-prgmr, 2019-08-19:
Hi. 
No reason for using Gray8. I was playing around the APIs and samples and stumbled upon this error. That's why decided to post it here.

TIS-Stefan, 2019-08-19:
Hmmm...

I tried several pipelines, as shown at https://theimagingsource.deskpro.com/en/kb/articles/linux-set-video-formats-in-camera-and-memory and GRAY8 worked fine. At least, with my USB cameras. I just tried 
` gst-launch-1.0 tcambin ! video/x-raw, format=GRAY8, widht=640,height=480, framerate=30/1 ! videoconvert ! ximagesink`
with a GigE Camera and it works fine on my machine too.

It may is possible, that the TCamCamera class does not accept this parameter.

closed 2019-08-22 by vishal-prgmr