TheImagingSource/tiscamera #308: Hardware Trigger - C++

vishal-prgmr, 2019-08-14:
Greetings,

Where do I set the trigger mode to Hardware trigger in the c++ Linux SDK? 
Camera model: DFK 33GX273.

TIS-Stefan, 2019-08-14:
Simply enable the Trigger Mode" property. As shown in the samples. The trigger mode is for hard- and software trigger.

vishal-prgmr, 2019-08-19:
Greetings, could you specify the sample that you are referring too.

TIS-Stefan, 2019-08-19:
Hello

This one:
https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Save%20an%20image%20using%20OpenCV%20in%20C%2B%2B

And the other ones, that show, how to set properties:
https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples

Stefan

vishal-prgmr, 2019-08-19:
@TIS-Stefan , Thank you for your reply. What should the Trigger Mode be set to(ON,OFF or the G_type_boolean)? 

I was referring to the example https://github.com/TheImagingSource/tiscamera/blob/master/examples/c/06-softwaretrigger.c

vishal-prgmr, 2019-08-19:
@TIS-Stefan , Thank you for your reply. 

I was referring to [https://github.com/TheImagingSource/tiscamera/blob/master/examples/c/06-softwaretrigger.c](url). 

What should the Trigger Mode be set to ?

TIS-Stefan, 2019-08-19:
HI

Since you use a GigE Camera, you may use "On" and "Off".

Stefan

vishal-prgmr, 2019-08-19:
HI

Does enabling the Trigger Mode override the boolean trigger signal "SaveNextImage" (in the software trigger example that you mentioned above)?

TIS-Stefan, 2019-08-19:
Hello Vishay
The trigger mode enables the trigger mode in the camera. That means, the camera stops sending images. It will send an image only, if a hardware trigger, e.g. a TTL signal occurs or if you send a software trigger by software trigger to the camera.
The trigger mode is used to make an image on an external signal, e.g. a light barrier creates that, because a specimen has moved in front of the camera.

You use a callback to handle the image. The callback handling has the advantage, that you must not wait for an image in your main thread, thus your program does not block in waiting for an image.

The sample [Save an image using OpenCV in C++](https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/Save%20an%20image%20using%20OpenCV%20in%20C%2B%2B/main.cpp) implements the callback function `GstFlowReturn new_frame_cb(GstAppSink *appsink, gpointer data)`.  This function is called every time a new image comes in. It looks somewhat complex, but most part is getting the image from the GstBuffer and convert it to OpenCV for saving or processing. 

The callback is set to the TCamCapture object by a call to 

```
// Register a callback to be called for each new frame
cam.set_new_frame_callback(new_frame_cb, &CustomData); 

```
After that, the live stream is started and you can enable the trigger mode
```
// Start the camera
cam.start();
    std::shared_ptr<Property> TriggerMode = NULL;
    try
    {
        TriggerMode = cam.get_property("Trigger Mode");
        TriggerMode->set(cam,"On");
    }
    catch(std::exception &ex)    
    {
        printf("Error %s : %s\n",ex.what(), "Trigger Mode");
    }
```
You will see, that the callback is called immediately after `cam.start()` at least once. Then the trigger mode comes into affect. As far as I know, the trigger mode should be enabled only after `cam.start()`, otherwise the application will hang, because the pipeline does not start.
I hope, this explains this a little bit better.

vishal-prgmr, 2019-08-22:
@TIS-Stefan ,

Thank You :)

closed 2019-08-22 by vishal-prgmr