TheImagingSource/tiscamera #309: Linux - DFK 42BUC03 - Missing parameters (Strobe Enable)

edwardpercy, 2019-08-19:
**No Strobe Enable Parameter** (Using Linux Ubuntu 16.04 - TIS Camera: DFK 42BUC03)

Hi, I am trying to set the parameter 'Strobe Enable' to true but the parameter is unknown and does not show as an option when running 'tcam-ctrl -p'. I am able to get it to work using Windows so I know its compatible with my camera. I tried running UVC Extension Units as per the tutorial (sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0) but get:

` /home/geoslam/tiscamera/data/uvc-extensions/tisEUVC.xml: error: video0: unable to map **'Strobe Enable'** control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)`

`/home/geoslam/tiscamera/data/uvc-extensions/tisEUVC.xml: error: video0: unable to map 'Strobe Enable' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)`

Any help is greatly appreciated as I have been struggling with this all day. Thanks

**I also followed the UDEV Rules tutorial. However running tcam-ctrl -p still doesn't show Strobe Enable as an option:**

Found 13 propert(y/ies)
Brightness                         
category=Exposure
Gain                                  
category=Exposure
Exposure                           
Focus                                  
Trigger Mode                       
Software Trigger                  
Gain Red                                
Gain Green                            
category=Exposure
Gain Blue                               
Binning                                 
X Offset                              
Y Offset                                
Trigger Global Reset Shutter        

**(v4l2-ctl --all) shows the driver info:** 

Driver Info (not using libv4l2):
        Driver name   : uvcvideo
	Card type     : DFK 42BUC03
	Bus info      : usb-0000:00:14.0-13.3
	Driver version: 4.15.18
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1280/960
	Pixel Format      : 'GRBG'
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 1228800
	Colorspace        : Default
	Transfer Function : Default
	YCbCr Encoding    : Default
	Quantization      : Default
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 960
	Default     : Left 0, Top 0, Width 1280, Height 960
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1280, Height 960
Selection: crop_bounds, Left 0, Top 0, Width 1280, Height 960
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness (int)    : min=0 max=255 step=1 default=12 value=12
                           gain (int)    : min=34 max=255 step=1 default=16 value=36
              exposure_absolute (int)    : min=1 max=300000 step=1 default=127 value=240
                 focus_absolute (int)    : min=0 max=1000 step=1 default=0 value=0
                        privacy (bool)   : default=0 value=0
                        trigger (bool)   : default=0 value=0
               software_trigger (button) :
                          gainr (int)    : min=0 max=63 step=1 default=1 value=36
                          gaing (int)    : min=0 max=63 step=1 default=1 value=36
                          gainb (int)    : min=0 max=63 step=1 default=1 value=36
                        binning (int)    : min=1 max=1 step=1 default=1 value=1
                       x_offset (int)    : min=0 max=514 step=2 default=0 value=0
                       y_offset (int)    : min=0 max=480 step=2 default=0 value=0
   trigger_global_reset_shutter (bool)   : default=0 value=0

edwardpercy, 2019-08-20:
Hi, thanks for the quick response. Is this planned for the future or is there any way it can be added?

TIS-Arne, 2019-08-20:
Hi Edward, 

This Python script will enable strobe for the DFK 42BUC03: 

https://gist.github.com/TIS-Arne/00a00e5cdc7a0d10cc31e50a41179de9

It is currently not possible to export this feature as a device property that can be controlled from the tiscamera software. This is a firmware limitation and unfortunately I can not tell whether there will be an update to the firmware any time soon. The aforementioned script will unload the device driver, enable the strobe functionality and load the device driver again. The strobe pulse will then be enabled until the camera gets power cycled. Maybe this helps?

/Arne

edwardpercy, 2019-08-20:
Hi Arne, thanks that worked perfectly! do you have a C/C++ version?

TIS-Arne, 2019-08-20:
Edward, 

I am sorry, I do not have a C version. You could just execute this script once after the camera is connected, eg. via a udev rule. This way there should not be a need for a C version.

edwardpercy, 2019-08-20:
Ah ok, no problem. Thanks for the help!

closed 2019-08-20 by edwardpercy