TheImagingSource/tiscamera #310: Gstreamer mp4 output issues (08-Save-Stream.c) - DFK 42BUC03 - Ubuntu 16.04

edwardpercy, 2019-08-20:
Hi, when running the 08-Save-Stream.c example program I have no issues. However when I try to change the output to use mp4 format within the 08-Save-Stream.c example code there is no output (0kb). Any help is greatly appreciated, I have tried many encoder/mux/bayer but no luck, thanks!
**When I run the command below it saves the .mp4 correctly:**

`gst-launch-1.0 tcambin! video/x-raw,format=BGRx,width=1280,height=960,framerate=10/1 ! videoconvert ! x264enc ! mp4mux -e ! filesink location=test.mp4`

**When I put this into the 08-Save-Stream.c example no data is saved:**

GstElement* pipeline = gst_parse_launch("tcambin name=bin"
                                            " ! video/x-raw,format=BGRx,width=1280,height=960,framerate=10/1"
                                            " ! videoconvert"
                                            " ! x264enc"
                                            " ! mp4mux -e"
                                            " ! filesink name=fsink", &err);

**The original 08-Save-Stream parameters work fine and produce a live video and save .avi:**

GstElement* pipeline = gst_parse_launch("tcambin name=bin"
                                            " ! video/x-raw,format=BGRx,width=640,height=480,framerate=30/1"
                                            " ! tee name=t"
                                            " ! queue"
                                            " ! videoconvert"
                                            " ! ximagesink"
                                            " t."
                                            " ! queue"
                                            " ! videoconvert"
                                            " ! avimux"
                                            " ! filesink name=fsink", &err);

TIS-Stefan, 2019-08-21:
Hello Edward

The video format must be converted into something, the x264enc understands. You may try

`gst-launch-1.0 tcambin  ! video/x-raw, format=BGRx,width=1280,height=960,framerate=10/1 ! videoconvert ! video/x-raw, format=I420,width=1280,height=960,framerate=10/1  ! queue ! x264enc ! mp4mux ! filesink location=test.mp4
`
Stefan

edwardpercy, 2019-08-21:
Hi Stefan, thanks again for the quick response. The mp4 saves but when I try to play it I get: "This file contains no playable streams" There is a file size now though.

TIS-Stefan, 2019-08-21:
I must admit, I tried that once successfully and I am not sure which player I used for that. 
Let me search the usage of the x264enc in the internet and try to find a solution. You can do that in parallel too.

Stefan

edwardpercy, 2019-08-21:
Hi Stefan, I tried adding the -e flag to the mp4mux and it saves the video in a format that can be read however the video is jumpy so something is still configured wrong.
Thanks fr the help, I appreciate it

edwardpercy, 2019-08-21:
Also when implementing this into C I also have an issue where the video won't be saved unless I have been recording for a while (approx >10 secs) otherwise nothing will be saved. Any idea what could be causing this?

TIS-Stefan, 2019-08-21:
Hi Edward

thank you very much for pointing me to the -e parameter of the mp4mux. I did not knew that. 
Running the capture pipeline on me Corei5 computer, I get a nice smooth video. No lags, no jumps.
As first measurement, we should add a queue module before videoconvert. Thus, the modules are distributed on the cores of your computer.

`gst-launch-1.0 tcambin ! video/x-raw, format=BGRx,width=1280,height=960,framerate=10/1 ! queue ! videoconvert ! video/x-raw, format=I420,width=1280,height=960,framerate=10/1 ! queue ! x264enc ! mp4mux -e ! filesink location=test.mp4
`
Then I would try to change the parameters of the x264enc in hope of reducing CPU load.

Which computer model do you use?

Stefan

TIS-Stefan, 2019-08-21:
Edward

I forgot the programming question.. Which pipeline do you use in C?
However, there should be no difference between gst-launch and your C code.

edwardpercy, 2019-08-21:
Running Ubuntu 16.04 with 15-4590 CPU
Here is some footage below: you can see it is jittery
[test.zip](https://github.com/TheImagingSource/tiscamera/files/3524537/test.zip)

edwardpercy, 2019-08-21:
Adding the -e flag to the mp4mux stops the C program from outputting. I can send the entire code and see if you can get it working on your end?

edwardpercy, 2019-08-21:
Hi I just noticed that when recording at framerate=10/1 the outputted test.mp4 headers state that it is 10fps but when running at framerate=15/1 the headers state 13fps

TIS-Stefan, 2019-08-21:
Hi Edward

I use Ubuntu 18.04. I am not sure, whether this makes a different. 
The pipeline creates a nice, smooth video on my computer, even if I move around.
How smooth is the video, if you simply show a live display? 

`gst-launch-1.0 tcambin ! video/x-raw, format=BGRx,width=1280,height=960,framerate=10/1 ! videoconvert ! ximagesink`

Is it competele smooth? If yes, we do not have issues with data transfer.

Stefan

edwardpercy, 2019-08-21:
Hi Stefan, the live video displays smooth

I managed to get the video to record smooth using the pipeline below. However there is still an issue where the .mp4 header reads duration=0 so playback goes into negative time (Although the video plays perfectly). Any idea why duration is recording as 0? I have used the EOS -e flag 

      GstElement* pipeline = gst_parse_launch("tcambin name=bin num-buffers=1000"
                                            

                                            " ! video/x-raw, format=BGRx,width=1280,height=960,framerate=25/1" 
                                            " ! tee name=t"
                                            " ! queue"
                                            " ! videoconvert"
                                            " ! video/x-raw, format=I420,width=1280,height=960,framerate=25/1"
                                            " ! videoscale"
                                            " ! videorate"
                                            " ! queue"
                                            " ! x264enc tune=zerolatency bitrate=5000 speed-preset=superfast qp-min=18"
                                            " ! h264parse"
                                            " ! mp4mux faststart-file=/data/local/tmp/faststart_file fragment-duration=100 faststart=true"                                     
                                            " ! filesink name=fsink async=0 -e", &err);

TIS-Edgar, 2019-08-22:
Hi Edward,

Take a look at the man page for gst-launch-1.0.
The `-e` is a flag for gst-launch not for a gstreamer element.
It is there to work around the quirks a commandline pipeline has in comparison to a program pipeline.

Pipelines for file saving require a reliable EOS signal. This so that the muxer is able to correctly write meta data like video length etc. into the resulting file. These things are typically determined when no new buffers arrive aka End-Of-Stream.

The `-e` in the c pipeline causes a syntax error. It never really plays. To confirm that you can increase the debug level `--gst-debug-level=2` or `--gst-debug=*:2`

Assuming the pipeline description is the only thing you changed in the 08 example you have two sources for EOS signals in the c code.
- The `num-buffers=1000` property of tcambin. If you take a look at `gst-inspect-1.0 tcambin` you will see that tcambin has no such property. To use that you need to use the tcamsrc directly.
- `gst_element_set_state(pipeline, GST_STATE_NULL);` 

When working bot would send the EOS signal causing the meta data to be written. In cases of errors, etc. where an orderly shutdown is prevented a corruption of the meta data is to be expected.

I hope that clears up some misunderstandings.

edwardpercy, 2019-08-22:
Hi Edgar, thank you for clearing that one up makes more sense now.
The only thing I have changed in the 08 example is the pipeline to:

`GstElement* pipeline = gst_parse_launch("tcambin name=bin num-buffers=1000"                                        

                                        " ! video/x-raw, format=BGRx,width=1280,height=960,framerate=25/1" 
                                        " ! tee name=t"
                                        " ! queue"
                                        " ! videoconvert"
                                        " ! video/x-raw, format=I420,width=1280,height=960,framerate=25/1"
                                        " ! videoscale"
                                        " ! videorate"
                                        " ! queue"
                                        " ! x264enc tune=zerolatency bitrate=5000 speed-preset=superfast qp-min=18"
                                        " ! h264parse"
                                        " ! mp4mux faststart-file=/data/local/tmp/faststart_file fragment-duration=100 faststart=true"                                     
                                        " ! filesink name=fsink async=0", &err);`

This outputs the video perfectly except that there is no duration stored in the meta-data headers. IT stores information such as bitrate/framerate/resolution etc but no duration (Causing the playback to enter negative values). Because the other headers are being written this suggests EOS is working? But how come there is no duration being stored.

edwardpercy, 2019-08-22:
Hi something else I found whilst debugging were lots of lines saying "posting on bust state-changed message: ... , time 99:99:99.9999999999, seq-num 258"

Any idea why all the times are at 99:99:99....?
Using the default pipeline I didn't get this

TIS-Edgar, 2019-08-22:
A workaround is to add the following code right before the set_state NULL.

```
    gst_element_send_event(pipeline, gst_event_new_eos());

    // wait for the EOS to traverse the pipeline and is reported to the bus
    GstBus *bus = gst_element_get_bus(pipeline);
    gst_bus_timed_pop_filtered(bus, GST_CLOCK_TIME_NONE, GST_MESSAGE_EOS);
    // this stops the pipeline and frees all resources
```

I am still investigating how this issue is exactly caused, but that allows you to keep going.

The fixme messages are caused by the gstreamer backend. They are FYI and have no influence on the quality/reliability of the stream.

The state change timestamps are something I would have to look into.

edwardpercy, 2019-08-22:
Thank you so much! that did the trick. Let me know if you figure out the cause of this. My final question is for the first 1 second of the video the frames are still (Almost like it is paused). I have attached a video so you can better understand.
[test.zip](https://github.com/TheImagingSource/tiscamera/files/3529606/test.zip)

Thanks for all the help it is much appreciated!

TIS-Edgar, 2019-08-22:
I am not sure where that delay comes from.
Does it also appear when you simply display the stream
instead of recording it?

The following _may_ help:
- The property `n-threads` for videoconvert 
- The property `sync=false` for filesink
- queues can be a hindrance

edwardpercy, 2019-08-22:
Thanks Edgar I will give it a go. Also how do I add n-threads because I keep getting: "no property n-threads in element videoconvert0"

edwardpercy, 2019-08-22:
Hi Edgar. With that extra EOS code you supplied earlier combined with the pipeline I was using yesterday allowed it to work. It seems the issues are solved. Thanks for your help

edwardpercy, 2019-08-22:
> gst_element_send_event(pipeline, gst_event_new_eos());
>
>     // wait for the EOS to traverse the pipeline and is reported to the bus
>     GstBus *bus = gst_element_get_bus(pipeline); [...]

This was the main thing I was missing. Not sure why it is needed but thanks!

closed 2019-08-22 by edwardpercy