TheImagingSource/tiscamera #311: Udev issues when symlinking cameras

blazini36, 2019-08-22:
I recently swapped out computers that I use to work with my 2 DFK33UX264's. I did a fresh install of Mint 19 and Tiscamera. I noticed there have been quite a few changes in tiscamera and Mint19 vs Mint18 such as the v4l2 controls and the way that they are read out.

I believe I have most of that sorted but I'm having issues with udev symlinking. Previously I did this with a single .rules file containing only the lines:
SUBSYSTEM=="video4linux", ATTRS{idVendor}=="199e", ATTRS{idProduct}=="9089", ATTRS{serial}=="<serialN>", SYMLINK+="topcam"  
SUBSYSTEM=="video4linux", ATTRS{idVendor}=="199e", ATTRS{idProduct}=="9089", ATTRS{serial}=="<serialN>", SYMLINK+="botcam"

I did not use the 80-theimagingsource-camera.rules file. To load the UVC extensions I just dropped the usb3.xml file into the /usr/share/uvcdynctrl/data/199e directory, I notice the 80~.rules file has another way of doing this with a "tcam-extension loader"

Right now for some reason with the tiscamera install and 2 cameras are plugged in I get 4 kernel video devices (video0-3) shown in /dev along with my 2 symlinks, "topcam" and "botcam". On boot they do not always link the correct video device, they randomly link to the 2 phantom devices from time to time.

What I need to do is create a symlink that is based on the serial number of the camera that will always be linked to the botcam/topcam video devices. I'm not sure how to get this working right on this install.

TIS-Edgar, 2019-08-29:
Hi there,

We deprecated the use of uvcdynctrl as the maintainability of extension units was not feasable.

If you have BUILD_TOOLS enabled tcam-uvc-extension-loader will be built and installed.
Within your build directory in data/udev is the generated rules file that contains the absolute paths necessary for it to work. You can simply copy the extension unit command out of it.

As for the double entries:

They are generated by the kernel due to the fact that the camera has 2 descriptor interfaces. This is something that can not be removed.

Since all cameras will always have 2 interfaces take the first one as that will always be the one you want. i.e.  0,2,4 etc.

If you require serial associations for dev nodes look into /dev/v4l/by-id there you can find symlinks.

blazini36, 2019-09-05:
I've tried the best I can to review your suggestions but It seems to create some new issues that I'm not sure I can resolve all issues without a further suggestion.

>They are generated by the kernel due to the fact that the camera has 2 descriptor interfaces. This is something that can not be removed.

That's not something that happened with the same cameras on the old PC install so I assume this second interface is created by Tcam?

>Since all cameras will always have 2 interfaces take the first one as that will always be the one you want. i.e. 0,2,4 etc.

I assume you mean to take the kernel name /dev/videoN which is lowest for that camera? That doesn't solve the original problem that the kernel will instantiate physical devices in a manor that is not consistent. The whole purpose of a Udev symlink that was based on serial was to say "serial #1234 links to topcam" and "serial #4321 links to botcam" because regardless serial #1234 may instantiate as /dev/video0 or /dev/video1. Now the case is that serial #1234 may be /dev/video0+1 or /dev/video2+3. The real problem is that both video0 and video1 links will be attached to serial number info but only one is attached to the useable v4l2 instance, the other has no v4l2-ctrl or image attached as far as I can see. The compounding issue is that the V4l2 attachment can be on either video0 or video1 just as inconsistent as the kernel instantiation.

So it appears I can work around this by using the "v4l....by-id-index-0" symlink because it does reference the cameras unique serial and it does seem to be consistent that the v4l2-ctrl usable interface is always attached to the "index-0" symlink. The problem I encounter with that is that OpenCV needs to refer to a device index, and that's a problem for the reasons above. Currently I'm thinking that querying the v4l link with "readlink" in program will expose the current video device but I'm wondering if that's the best approach.

```
~$ readlink '/dev/v4l/by-id/usb-The_Imaging_Source_Europe_GmbH_DFK_33UX264_28810325-video-index0' 
../../video2
```

upon boot that kernel device could be any one of 2-4 instances, upon re-plugging the camera it almost certainly will change

TIS-Edgar, 2019-09-05:
I think I finally found a way to reliably create a correct symlink.
The rule would look like this:
```
ACTION=="add", SUBSYSTEM=="video4linux", ATTRS{idVendor}=="199e",\
    ENV{ID_V4L_CAPABILITIES}==":capture:", \
    ATTRS{serial}=="17810181",  \
    GROUP="video", MODE="0666", TAG+="uaccess", TAG+="udev-acl", \
    SYMLINK+="topcam"
```

The important line is `ENV{ID_V4L_CAPABILITIES}==":capture:", `
It looks at the udev attributes and selects the capture interface for the symlink.
To see the most important properties execute:
```
udevadm info --query=all --name=/dev/video0
```

For all other solutions you'd have to write your own program and iterate the devices with libudev.

blazini36, 2019-09-05:
I wrote the following file as a 80....rules file in /etc/udev/rules.d/ copying out what looked necessary in the example .rules file from the tiscamera /data directory. It's not producing the "topcam" and "botcam" symlinks in /dev as the rules I posted above does. Any suggestions?

> 
# This file assures that The Imaging Source USB cameras are correctly recognized

# if no driver has claimed the interface yet, load uvcvideo
ACTION=="add", SUBSYSTEM=="usb", \
ATTRS{idVendor}=="199e", ATTRS{idProduct}=="8101", \
RUN+="/sbin/modprobe -b uvcvideo"

# add the imaging source VID and PID to the list of devices supported by uvcvideo
ACTION=="add", SUBSYSTEM=="drivers", \
               ENV{DEVPATH}=="/bus/usb/drivers/uvcvideo", \
               RUN+="/bin/sh -c 'echo 199e 8101 > /sys/bus/driver/uvcvideo'"

# give users permission to interact with cameras
SUBSYSTEM=="usb", ATTRS{idVendor}=="199e", GROUP="video", MODE="0666", TAG+="uaccess", TAG+="udev-acl"

# load uvc extensions for USB3 33 cameras, requires absolute path
ACTION=="add", SUBSYSTEM=="video4linux", \
               ATTRS{idVendor}=="199e", ATTRS{idProduct}=="90??", \
               RUN+="/usr/bin/tcam-uvc-extension-loader --device=/dev/%k -f /usr/share/theimagingsource/tiscamera/uvc-extension/usb33.json"

##specific camera rules here
#topcam
ACTION=="add", SUBSYSTEM=="video4linux", ATTRS{idVendor}=="199e",\ ENV{ID_V4L_CAPABILITIES}==":capture:", \ ATTRS{serial}=="28810325", \ GROUP="video", MODE="0666", TAG+="uaccess", TAG+="udev-acl", \ SYMLINK+="topcam"

#botcam
ACTION=="add", SUBSYSTEM=="video4linux", ATTRS{idVendor}=="199e",\ ENV{ID_V4L_CAPABILITIES}==":capture:", \ ATTRS{serial}=="43710275", \ GROUP="video", MODE="0666", TAG+="uaccess", TAG+="udev-acl", \ SYMLINK+="botcam
<

It does appear to load the extensions without the usb3.xml file in /usr/share/uvcdynctrl/data/199e, the cameras list v4l2-ctrls that would be unavailable as a generic V4l2 device.

TIS-Edgar, 2019-09-06:
If you want to have the camera specific rules in one line you have to remove the `\`.
They are there to tell udev that the rule continues on the next line.

blazini36, 2019-09-07:
I probably should have known that about the "\". Anyway I fixed that in udev, the links are created, but they are pointed at "fuse". I'm not quite sure what that's about.
```
# This file assures that The Imaging Source USB cameras are correctly recognized

# if no driver has claimed the interface yet, load uvcvideo
ACTION=="add", SUBSYSTEM=="usb", \
ATTRS{idVendor}=="199e", ATTRS{idProduct}=="8101", \
RUN+="/sbin/modprobe -b uvcvideo"

# add the imaging source VID and PID to the list of devices supported by uvcvideo
ACTION=="add", SUBSYSTEM=="drivers", \
               ENV{DEVPATH}=="/bus/usb/drivers/uvcvideo", \
               RUN+="/bin/sh -c 'echo 199e 8101 > /sys/bus/driver/uvcvideo'"

# give users permission to interact with cameras
SUBSYSTEM=="usb", ATTRS{idVendor}=="199e", GROUP="video", MODE="0666", TAG+="uaccess", TAG+="udev-acl"

# load uvc extensions for USB3 33 cameras, requires absolute path
ACTION=="add", SUBSYSTEM=="video4linux", \
               ATTRS{idVendor}=="199e", ATTRS{idProduct}=="90??", \
               RUN+="/usr/bin/tcam-uvc-extension-loader --device=/dev/%k -f /usr/share/theimagingsource/tiscamera/uvc-extension/usb33.json"

# specific camera rules here
#topcam
ACTION=="add", SUBSYSTEM=="video4linux", ATTRS{idVendor}=="199e", \ 
               ENV{ID_V4L_CAPABILITIES}==":capture:", \ 
               ATTRS{serial}=="28810325", \ 
               GROUP="video", MODE="0666", TAG+="uaccess", TAG+="udev-acl", \ 
               SYMLINK+="topcam"

#botcam
ACTION=="add", SUBSYSTEM=="video4linux", ATTRS{idVendor}=="199e",\ 
               ENV{ID_V4L_CAPABILITIES}==":capture:", \ 
               ATTRS{serial}=="43710275", \ 
               GROUP="video", MODE="0666", TAG+="uaccess", TAG+="udev-acl", \ 
               SYMLINK+="botcam" 
```
```
~$ ls -l /dev/topcam  
lrwxrwxrwx 1 root root 4 Sep  6 18:37 /dev/topcam -> fuse
```
```
~$ readlink /dev/topcam
fuse
```

blazini36, 2019-09-15:
Any ideas on this? I realized that the above udev file pretty much links devices randomly. Recently I found that the topcam symlink was actually linked to my NVME drive!

TIS-Stefan, 2019-09-18:
Hello

Sorry for not answering. I needed some time to understand your issue. I have the same on my Ubuntu 18.04: Connecting a camera to the computer and I get
/dev/video0
/dev/video1

This even happens with the USB 2.0 cameras, which definitely have only one interface : UVC only. 

I removed the files from uvcdynctrl, but I suppose, that has no effect on the issue at all, because the issue happens somewhere in udev.

I also removed the /etc/udev/rules.d/aravis.rules, but this also does not make an effect. There are still two /dev/video* shown.

We have to continue searching.

Stefan

TIS-Stefan, 2019-09-18:
Hi
I have learned, that the double devices are caused by the Kernel. The second device is for an UVC Meta Data Stream, but this is unused.
Thus you may make your simlinks to /dev/video0 and /dev/video2.

Stefan

blazini36, 2019-09-18:
Hi,

that doesn't really work because /dev/videoN is instantiated by the kernel at random. So serial#1234 may be video0, right now, on next boot it may be video2. This is OK if you are ignoring /dev/videoN and symlinking to a serial number, but now you must refer back to video0 or video2 because video1 and video3 exist?

This creates a somewhat convoluted issue with simply identifying individual cameras reliably. When there is one video device per hardware camera it is the typical Linux Kernel issue of just udev symlinking based on a hardware feature that is unchangeable. Now with 2 devices per camera a rule must be written on the user's end to enumerate the lower of the 2 kernel instantiated video devices?

My programmer friend's solution to this is to read which link has v4l2-ctrl output on it and simlink that camera but that depends on my original symlink of identifying which set of links is attached to which serial number. You can see this becomes very convoluted to manage. The v4l2-by-id symlink is probably the only thing reasonable to use in this case but the user must be aware to use the lower instance of the by-id link. I think there should be a better way.

If it's a kernel issue I have not seen it in 4.9. I believe the current build is running 4.18 so I suppose this came about between the 2 versions? I can easily test this on a machine running 5.3, but I assume you are saying the issue will present itself on any machine based on kernel version independant of tiscamera?

TIS-Stefan, 2019-09-18:
Regarding Kernel versions:
I use 4.19 and I did not saw this issue on older Kernels, therefore, that must have been invented in between, as you think.

The best way to identify the cameras is using the serial numbers and query the dev node programmatically. The following C++ sample project shows, how to enumerate the cameras, video formats, frame rates and also how to set properties.
[v4l2_cameratest.zip](https://github.com/TheImagingSource/tiscamera/files/3626349/v4l2_cameratest.zip)

You may pick from it, what you need to open the cameras. 
However, this tool will list the cameras twice too.

Stefan

closed 2019-11-29 by TIS-Stefan