TheImagingSource/tiscamera #312: Setting Properties - C++

vishal-prgmr, 2019-08-26:
Greetings,

When I am trying to set the properties "TriggerPolarity" and "Exposure Value", I get the error "Failed to Query Property: TriggerPolarity"

I set the properties as follows : 

```
//Exposure Value
    std::shared_ptr<Property> ExposureValue = NULL;
    try
    {
        ExposureValue = cam.get_property("Exposure Time (us)");
        ExposureValue->set(cam,333);
    }
    catch(std::exception &ex)
    {
        printf("Error %s : %s\n",ex.what(), "Exposure Value");
    }

//Trigger polarity
    std::shared_ptr<Property> TriggerPolarity = NULL;
    try
    {
        TriggerPolarity = cam.get_property("Trigger Polarity");
        TriggerPolarity->get(cam,"RisingEdge");
    }
    catch(std::exception &ex)
    {
        printf("Error %s : %s\n",ex.what(), "Trigger Polarity");
    }
  
```

TIS-Stefan, 2019-08-26:
Hello

I am sorry to ask, but which camera model do you use?
Did you check for "Trigger Polarity" availability and same name "Trigger Polarity" by a call to 
`tcam-ctrl -c <serial>`
?
Maybe your camera does not support this

Stefan

vishal-prgmr, 2019-08-26:
Hi
@TIS-Stefan , the model is DFK 33GX273. Yes, i am sure the camera supports this because i referred to the Parameter name from the manual available on the website for my camera model.

TIS-Stefan, 2019-08-26:
Ups, sorry,
I meant tcam-ctrl -p
Just run "tcam-ctrl" without parameters in order to see, which parameters are allowed.

Stefan

vishal-prgmr, 2019-08-26:
Ran tcam-ctrl -p and got the actual parameter names. 
Its not "Exposure Time (us)" for setting Exposure value. The correct parameter name was "**Exposure**". 
And for Trigger Polarity, the correct parameter name was "**Trigger Activation**".

Now the properties are queried without any errors. 

Thank You

closed 2019-08-26 by vishal-prgmr