TheImagingSource/tiscamera #314: 'Simple Snap Image in C++' program hangs

OpalJellyfish, 2019-09-04:
I tried running the  **Simple Snap Image in C++** program given in the _Linux-tiscamera-Programming-Samples_ repo. When I try running the program, it often works correctly immediately after making it:
```
$ ./simple-snapimage 
Start Snap
pipeline refcount at cleanup: 1
```

However, the program tends to hang afterwards and I need to terminate the program by pressing Ctrl + c:
```
$ ./simple-snapimage 
Start Snap
^C 
```

For reference, my OS is Ubuntu 16.04.6 LTS.

TIS-Stefan, 2019-09-05:
Hello

Do you know the line of code, where the sample hangs? 

Thank you in advance!

Stefan

OpalJellyfish, 2019-09-05:
Hello Stefan,

The sample seems to hang at this line: `scanf("%c",dummyvalue);`

If I replace these lines:
char dummyvalue[10];
scanf("%c",dummyvalue);
with `getchar();`, the sample hangs at `getchar();` instead.

If I just comment out the lines dealing with waiting for the enter key so the code looks like this:
```
    else
    {
        printf("Timeout at snapImage()\n");
    }

    cam.stop();
```, then the program hangs at `cam.stop();` sometimes [since the program is multithreaded, the program seems to hang at different places for different executions].

Now if I undo all of the changes I have made and just change `cam.set_capture_format("BGRx", FrameSize{640,480}, FrameRate{30,1});` to `cam.setCaptureFormat("BGRx", FrameSize{5472,3648}, FrameRate{5,1});` as you mentioned, I keep getting:
> Timeout at snapImage()

I will see if I can try to run the program on Ubuntu 18.04.

OpalJellyfish, 2019-09-05:
I tried running the sample unmodified on Ubuntu 18.04.3 LTS and I got the following output:
> Start Snap
> Timeout at snapImage()
> Press enter key to end program.
> pipeline refcount at cleanup: 1

Then I tried changing 
`cam.set_capture_format("BGRx", FrameSize{640,480}, FrameRate{30,1});` to `cam.set_capture_format("BGRx", FrameSize{5472,3648}, FrameRate{5,1});`
 and I get the same output:
> Start Snap
> Timeout at snapImage()
> Press enter key to end program.
> pipeline refcount at cleanup: 1

TIS-Stefan, 2019-09-06:
There is a comment about the snapImage line saying that the timeout should be set accordingly to the frame rate. E.g. if the frame rate is 30 fps, the timeout should be double of 1/fps = 1/30 * 2  = 1/15 = 66ms. Therefore, I wonder, why 60ms was too long for 30 frames per second. Only point could be you get no images or your exposure time is very long.

If you use 5 frames per second, the timeout should be around 500ms. 
You may try a pipeline at the command line
`gst-launch-1.0 tcambin ! video/x-raw, format= BGRx,width=640, height=480, framerate=30/1 ! videoconvert ! ximagesink`
This should give an image. If so, all is fine. Try a longer timeout.
I am not sure, whether the ximagesink handles the maximum size image. We commented the line, in which the ximagesink is set to "cam" when using that format.

BTW: the tcam-capture software shows an image from the camera too?

Stefan

OpalJellyfish, 2019-09-06:
I commented out the line: 
`cam.enable_video_display(gst_element_factory_make("ximagesink", NULL));`

and 

>BTW: the tcam-capture software shows an image from the camera too?
Where is this tcam-capture software located? Is it this example?

OpalJellyfish, 2019-09-06:
You mentioned that my exposure time could be too long, so I tried running the **tiscamera/examples/02-set-properties example** and I got the following:
> Exposure Auto has value: true
> Could not query Gain Auto
> Could not query Exposure
> Exposure Auto has value: false
> Could not query Gain Auto
> Could not query Exposure
Why could I not query Exposure?

I set the color format to:
`cam.set_capture_format("BGRx", FrameSize{5472,3648}, FrameRate{5,1});`
and set the timeout to:
`if( cam.snapImage(500) )`
but 'Timeout at snapImage()' was still printed out.

Even if I try a longer timeout as you recommended:
`if( cam.snapImage(10000) )`
I still get 'Timeout at snapImage()' printed out.

>I am not sure, whether the ximagesink handles the maximum size image. We commented the line, >in which the ximagesink is set to "cam" when using that format.
I commented out the line: 
`cam.enable_video_display(gst_element_factory_make("ximagesink", NULL));`
and tried the same timeouts I did above but I still get 'Timeout at snapImage()'.

When I tried running 
`$ gst-launch-1.0 tcambin ! video/x-raw, format= BGRx,width=640, height=480, framerate=30/1 ! videoconvert ! ximagesink`
I got images and the following output:
> Setting pipeline to PAUSED ...
> Pipeline is live and does not need PREROLL ...
> INFO:
> Working with src caps: video/x-raw, framerate=(fraction)30/1, width=(int)640, height=(int)480, format=(string)GRAY16_LE
> Setting pipeline to PLAYING ...
> New clock: GstSystemClock
> 
> ^Chandling interrupt.
> Interrupt: Stopping pipeline ...
> Execution ended after 0:00:07.839163043
> Setting pipeline to PAUSED ...
> Setting pipeline to READY ...
> Setting pipeline to NULL ...
> Freeing pipeline ...

Just to check, I tried running 
`$ gige-daemon start`
and I got 
>Daemon is already running!

And when I tried running 
`$ tcam-capture`
images from the camera were displayed and I saved an image as a .tiff file.

TIS-Stefan, 2019-09-09:
I am lost... It works fine on our x86 computers Ubuntu 18.04. Timeout 1000ms. 
If cam.enable_video_display(gst_element_factory_make("ximagesink", NULL)); is not commented out, does a small window opens showing a live video? According to the error message I would say not.
So the question is: Why does tcam capture work? Why does the same pipeline work on the command line, which does not work in the program?
Did you enable the "Trigger Mode"? I do not think so, otherwise you wrote that in the text here.

You can set a breakpoint in tcamimage.cpp at line 64 and check, wether the callback is called at all.
Also you can enable debug logging with
`export TCAM_LOG=DEBUG`
and send the output to me.

You may try in the pipeline "GRAY8" instead of "BGRx"

OpalJellyfish, 2019-09-09:
1. I got the exposure program to work CONTINUE TYPING!

2. If `cam.enable_video_display(gst_element_factory_make("ximagesink", NULL));` is not commented out, no window showing a live video opens. 

3. "Did you enable the "Trigger Mode"? I do not think so, otherwise you wrote that in the text here."
I am confused, are you asking me if I enabled the trigger mode in the Simple Snap Image in C++ example?

4. Using gdb, I set a breakpoint in tcamimage.cpp at line 64 and the callback does not seem to be called at all.

OpalJellyfish, 2019-09-09:
1. I got the exposure program to work CONTINUE TYPING!

2. If `cam.enable_video_display(gst_element_factory_make("ximagesink", NULL));` is not commented out, no window showing a live video opens. 

3. "Did you enable the "Trigger Mode"? I do not think so, otherwise you wrote that in the text here."
I am confused, are you asking me if I enabled the trigger mode in the Simple Snap Image in C++ example?

4. Using gdb, I set a breakpoint in tcamimage.cpp at line 64 and the callback does not seem to be called at all.$ gdb ./simple-snapimage

OpalJellyfish, 2019-09-09:
1. If `cam.enable_video_display(gst_element_factory_make("ximagesink", NULL));` is not commented out, no window showing a live video opens. 

2. "Did you enable the "Trigger Mode"? I do not think so, otherwise you wrote that in the text here."
I am confused, are you asking me if I enabled the trigger mode in the Simple Snap Image in C++ example?

3. Using gdb, I set a breakpoint in tcamimage.cpp at line 64 and the callback does not seem to be called at all. 

Specifically I ran 
`$ gdb ./simple-snapimage`
and did the following:

> (gdb) b tcamimage.cpp:64
> Breakpoint 1 at 0x280e0: file /home/ramya/tiscamera-Linux-Programming-Samples/Simple Snap Image in C++/tcamimage.cpp, line 64.
> (gdb) r
> Starting program: /home/ramya/tiscamera-Linux-Programming-Samples/Simple Snap Image in C++/build/simple-snapimage 
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
> [New Thread 0x7fb42d91d0 (LWP 15998)]
> [New Thread 0x7fb3ad81d0 (LWP 15999)]
> [New Thread 0x7fb32d71d0 (LWP 16000)]
> [New Thread 0x7fb29251d0 (LWP 16001)]
> [Thread 0x7fb29251d0 (LWP 16001) exited]
> Start Snap
> Timeout at snapImage()
> Press enter key to end program.
> pipeline refcount at cleanup: 1
> [Thread 0x7fb42d91d0 (LWP 15998) exited]
> [Thread 0x7fb32d71d0 (LWP 16000) exited]
> [Thread 0x7fb3ad81d0 (LWP 15999) exited]
> [Inferior 1 (process 15991) exited normally]

4. I enabled debug logging as you mentioned. Here is the output I am getting after running `$ ./simple-snapimage`:

> 116315     <DEBUG> Logger:107: 
> The following library versions are used:
> 	Tcam:	0.11.1_master/2042bba_rev_2617
> 	Aravis:	0.6_version_2042bba
> 138211     <DEBUG> Logger:107: 
> The following library versions are used:
> 	Tcam:	0.11.1_master/2042bba_rev_2617
> 	Aravis:	0.6_version_2042bba
> Start Snap
> Timeout at snapImage()
> Press enter key to end program.
> pipeline refcount at cleanup: 1

5. Trying "GRAY8" instead of "BGRx" did not change anything. Running 
`$ ./simple-snapimage`
still gives me 

> Start Snap
> Timeout at snapImage()
> Press enter key to end program.
> pipeline refcount at cleanup: 1

TIS-Arne, 2019-10-01:
@OpalJellyfish 
Hi, Sorry for not coming back to you earlier. I missed this github ticket. 

Which JetPack Version are you using on the Xavier?

closed 2019-11-29 by TIS-Stefan