TheImagingSource/tiscamera #315: Binning cannot be set on DFK AFU420

ekorneev-os, 2019-09-06:
We are working with DFK AFU420 cameras, USB3, using Linux SDK. We are able to communicate and control the cameras with tiscamera library but unable to set horizontal and vertical binning properties in Python. This is a problem when we attempt to set a lower resolution for the camera stream, and predictably only a small portion of the frame is streamed when binning is set to 1. When we attempt to set the binning properties - no errors are returned, but upon checking the horizontal and vertical binning values are always 1.  Here is the example of the code:

import TIS

Tis = TIS.TIS(47810390, 960, 600, 30, True)

Tis.Set_Property("Binning Horizontal",  8)
Tis.Set_Property("Binning Vertical", 8)

print('horiz',Tis.Get_Property("Binning Horizontal").value)
print('vert', Tis.Get_Property("Binning Vertical").value)

TIS-Stefan, 2019-09-06:
Hi
Binning can be set only, if the live video does not run.  I have a C++ sample setting binning:

  _pTCamCamera->pause();
   setEnumProperty(_PropertyName.c_str(), Selected.c_str() );
   _pTCamCamera->start();

ekorneev-os, 2019-09-06:
Thank you for the reply. How can we set binning in Python?

-----------------------------------------
Egor Korneev
ph: 608.504.2475
CEO | Founder | Researcher
Ordinal Science - AI Innovation Lab
website | linkedin

TIS-Stefan, 2019-11-29:
I am sorry, I missed your question from 6th September.

The principle of setting binning in Python is the same as doing in C++.
Are you able to implement that in Python?

Stefan

closed 2021-02-19 by TIS-Stefan