TheImagingSource/tiscamera #316: Documentation of camera properties

davidsean, 2019-09-11:
The function of most camera properties are obvious by the name, like `Exposure Time (us)` or `Gain`. Others require a bit more imagination and guess work. For example, on my  DMM 27UJ003-ML, the property `Override Scanning Mode` can have value 0, 1 or 2 (I'd like to try getting high frame-rate, lower resolution, full field of view images, but first need to know how these modes are defined).

Additionally I am looking into using the camera's output to trigger the illumination, so properties like: `Trigger Delay (us)`, `Strobe Delay`, `Strobe Duration`, `Strobe Exposure`, `GPIO` `GPOut`, `Trigger Global Reset Release` are good examples of what I would like to understand more.

Where can I find documentation about these properties ? 

I am adding below the properties I currently can see as a result of ` tcam-ctrl -p 12910212`
```Found 19 propert(y/ies)
Brightness                              (int) min=0 max=4095 step=1 default=-8193 value=168 category=Exposure
Gain                                    (int) min=100 max=383 step=1 default=57343 value=383 category=Exposure
Exposure Time (us)                      (int) min=100 max=30000000 step=1 default=33333 value=1000000 category=Unknown
Gain (dB/100)                           (int) min=0 max=583 step=1 default=0 value=0 category=Unknown
Trigger Mode                            (bool) default=true value=false category=Special
Software Trigger                        (button) category=Special
Trigger Delay (us)                      (int) min=0 max=10000000 step=10 default=0 value=150 category=Special
Strobe Enable                           (bool) default=true value=true category=Special
Strobe Polarity                         (bool) default=false value=false category=Special
Strobe Exposure                         (bool) default=false value=false category=Special
Strobe Duration                         (int) min=0 max=1023 step=1 default=0 value=0 category=Special
Strobe Delay                            (int) min=0 max=0 step=1 default=0 value=0 category=Special
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special
GPIO                                    (int) min=0 max=1 step=1 default=0 value=0 category=Special
Offset X                                (int) min=0 max=3776 step=2 default=0 value=0 category=Partial Scan
Offset Y                                (int) min=0 max=2668 step=2 default=0 value=0 category=Partial Scan
Offset Auto Center                      (bool) default=true value=true category=Partial Scan
Override Scanning Mode                  (int) min=0 max=2 step=1 default=0 value=0 category=Unknown
Trigger Global Reset Release            (bool) default=false value=false category=Special```

TIS-Stefan, 2019-09-12:
Hello
The Override Scanning Mode is tricky and used for enabling binning and skipping. It can be set to a value only, if the live stream / pipeline is stopped. Also a suitable resolution must be set, if the resolution is too big, there wont be an effect.
Unfortunately, the Override Scanning Mode value is different for each sensor, thus, here is trial and error and no documentation.
For the documentation of the properties I recommend to have a look into the "Technical Reference Manual" of the 33U cameras, e.g. https://www.theimagingsource.com/products/industrial-cameras/usb-3.0-color/dfk33uj003/
There is an explanation of camera properties. Not all of them are in your 27UJ003, because the 33UJ003 is somewhat smarter. However, in the Technical Reference Manual is a good description of camera properties.
I hope, this helps a little bit.
My Override Scanning Mode answer is not satisfying, I think about writing a small sample program, that allows to set that property and see the result. Then the values and their effect are more or less clear. What do you think?

Stefan

davidsean, 2019-09-17:
Although the behaviour Override Scanning mode is different, it would be great to have some examples to experiment with. If one wants 2x skipping (or binning) should the original resolution (for example 2560x1440) or the reduced resolution (1260x720) be used as a gstreamer caps?

Am I correct to assume that the frame rates will be affected by the Scanning modes?  Is is possible to have a wide-field of view with higher frame rates (but lower resolution) using skipping? I played around with changing the values, but the gstreamer-1.0 caps (from `tcam-ctrl -c`) do not seem to change.

TIS-Stefan, 2019-09-18:
Hello David

> If one wants 2x skipping (or binning) should the original resolution (for example 2560x1440) or the reduced resolution (1260x720) be used as a gstreamer caps?

Yes, the resolution must be set before override scanning mode. 

> Am I correct to assume that the frame rates will be affected by the Scanning modes? 
In Windows yes, in Linux for GigE cameras too. USB cameras use v4l2, which gets its formats and frame rates from the USB descriptor in the camera. That means, formats and frame rates are hard coded. Also: no frame rate change.
There might be missing the optimal reduced resolutions in the cameras. We found this for another model. That will be matter of changes with the next firmware releases.

By this week I have no time to create a sample code / test program  for Overide Scanning Mode, but I will have a look in the next week.

Stefan

TIS-Stefan, 2019-09-25:
Hello David

I tested the override scanning mode values for this DMM 27UJ003.
0 , 1: no binning 
2 : binning 2x
The stream has to be stopped, when setting the override scanning mode:
```
    gst_element_set_state(pipeline_, GST_STATE_READY);
    gst_element_get_state(pipeline_, NULL, NULL, GST_CLOCK_TIME_NONE);
```
Then the property can be set and the pipeline can be restarted:
```
gst_element_set_state(pipeline_, GST_STATE_PLAYING);
gst_element_get_state(pipeline_, NULL, NULL, GST_CLOCK_TIME_NONE);
```    
Before binning / Override Scanning Mode comes into effect, the video format resolution must be set to half of maximum width and height.

Stefan

closed 2019-11-29 by TIS-Stefan