TheImagingSource/tiscamera #318: RPI 4B and DMM 72BUC02-MLPB

junsujang, 2019-09-27:
I have installed tiscamera on a new raspberrypi 4B, and I am having trouble connecting DMM 72BUC02-MLPB cameras through USB. I have soldered the wires to a USB cable with type-A head, and while it works on my ubuntu and other sbcs, it is almost always not able to make a proper connection. Thus, when I run list-devices.py from examples, I do not find the camera. I get the following dmesg:

`[ 3276.687368] uvcvideo: Found UVC 1.00 device DMM 72BUC02-MLPB (199e:8307)
[ 3276.687851] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 1: -71 (exp. 1).
[ 3276.688073] uvcvideo: Failed to query (GET_INFO) UVC control 6 on unit 1: -71 (exp. 1).
[ 3276.688286] uvcvideo: Failed to query (GET_INFO) UVC control 17 on unit 1: -71 (exp. 1).
[ 3276.688497] uvcvideo: Failed to query (GET_INFO) UVC control 2 on unit 3: -71 (exp. 1).
[ 3276.688711] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 3: -71 (exp. 1).
[ 3276.689856] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[ 3276.690085] uvcvideo: Failed to query (129) UVC probe control : -71 (exp. 26).
[ 3276.690109] uvcvideo: Failed to initialize the device (-5).
[ 3276.722115] usb 1-1.1: USB disconnect, device number 45
`

giorgio-ne, 2019-09-29:
I have a similar issue on Raspbian buster on both a Raspberry Pi 4B anda Raspberry Pi 3B+. The camera is a USB3 Skyris 445M manufactured by TIS for Celestron
Here is the output of lsusb and dmesg:

pi@Rpi3:~ $ lsusb
Bus 001 Device 005: ID 199e:8442 The Imaging Source Europe GmbH 
Bus 001 Device 004: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@Rpi3:~ $ dmesg|tail
[   58.903505] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   58.903514] usb 1-1.3: Product: SKYRIS 445M
[   58.903524] usb 1-1.3: Manufacturer: Celestron
[   58.903533] usb 1-1.3: SerialNumber: 4414355
[   59.184856] uvcvideo: Found UVC 1.10 device SKYRIS 445M (199e:8442)
[   59.185673] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[   59.185857] uvcvideo: Failed to query (129) UVC probe control : 26 (exp. 34).
[   59.185866] uvcvideo: Failed to initialize the device (-5).
[   59.186198] usbcore: registered new interface driver uvcvideo
[   59.186210] USB Video Class driver (1.1.1)

giorgio-ne, 2019-09-29:
OK, my issue was fixed by flashing the camera firmware to using file Usb3Cam.413.fwpack downloaded from:
https://s3.amazonaws.com/celestron-site-support-files/support_files/firm-413.zip

TIS-Stefan, 2019-09-30:
Hello
The correct firmware file for your camera is in the repository 
tiscamera/data/firmwares/usb2/dfk72uc02_3012.euvc

Update instructions are at https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-CMOS-Cameras

You should clone and build the tiscamera repository.

Stefan

junsujang, 2019-09-30:
Hello Stefan,

I have already gone through all the steps to install the firmware on the camera about two weeks ago (that is why the cameras all work on other computers besides RPI4).

TIS-Stefan, 2019-10-01:
The Skyris camera is quite old, made in 2014. Thus the EEPROM can be too small and the firmware update fails. After successful firmware update the hardware ID should start with 199e:83xx, but yours is 199e:8442, thus I think, the update failed.
----------------------------------------------------------------------

The other DMM 72BUC02-MLPB seems to have the correct firmware. Since the camera with your soldered cable works fine on all other cameras, the RP4 seems to make a situation. I can not guess, why and since a standard USB 2.0 cable can not be used, I can not really help.
Thus, you may try an USB 2.0 hub between camera and RP4 and see, what happens.
I am sorry, but that is all I can do and suggest.

Stefan

junsujang, 2019-10-01:
Hello Stefan,

Thank you for your opinion on this matter. I think the solution with RPI4 would probably be using 2 USB3.0 cameras. I will update the issue if I get to try that out.

Thank you,
-Junsu

closed 2019-10-01 by junsujang