TheImagingSource/tiscamera #319: Trying to make a stereo imaging pair using DMM 72BUC02-MLPB

junsujang, 2019-09-27:
I am trying to use two DMM 72BUC02-MLPB cameras to capture images at 5FPS synchronized by an external trigger from a function generator input (5V square wave). At the moment, on a RPI 4B, I can only get upto 3FPS max with synchronized and non-missing images when running both cameras at the same time. 

Please find my code and commands below:

[stereo.zip](https://github.com/TheImagingSource/tiscamera/files/3664682/stereo.zip)

Any suggestions on how one should go about using 2 USB cameras as a stereo pair will be greatly appreciated.

TIS-Stefan, 2019-09-30:
If you need higher frame rates, then you must use lower resolution. The USB 2.0 bandwidth is exhausted, if one camera runs will full resolution 2592x1944@6 fps. Thus even 3 fps is near to 50% of the bandwidth. 

Stefan

junsujang, 2019-09-30:
Hello Stefan,

Thank you for your advice. I wish I had one more USB3 camera (I already have one) that I can test to see if I could use two USB3 cameras with 6FPS. 

On a separate note, would you still recommend using tiscamera for stereoimaging?

Thanks!

TIS-Stefan, 2019-10-01:
Hello 

From my experience, the RP4 has a good USB 3.0 speed. Thus I think, running two 5MP cameras with 5 or 6 fps could work. Question is only the CPU load of the PI. If this is too high, you will encounter the same issues.

Stefan

closed 2019-10-01 by junsujang