TheImagingSource/tiscamera #320: Framerate and external trigger

junsujang, 2019-09-30:
I have noticed that if you set the framerate of the camera lower than the external trigger frequency in the code, you do not end up getting the FPS that you are trying to get. May I ask what is the mechanism behind setting up the FPS on the software provided in tiscamera library?

TIS-Stefan, 2019-10-01:
The frame rate determines the speed, that is used to send the image data out. If you trigger faster that 1/fps seconds, then triggers occur, while the camera is busy and there these triggers are ignored.

Stefan

junsujang, 2019-10-01:
Hmm, does that mean that the data output from the camera can actually be delayed if the external trigger is not synchronized with the camera?

For example, can we consider the following case: let's set the camera to be 4 FPS and only use external trigger with 2Hz.

```
time, t (s)        0.0    0.1    0.2    0.3     0.4    0.5    0.6    0.7    0.8     0.9     1.0
cam data out        o                o                    o              o
trigger                       X                                    X

```
In the following case, will the data be only be ouput at t=0.25 and t=0.75 instead of t=0.15 and t=0.65?

Is there a way to ensure that the data is immediately sent when there is an external trigger?

TIS-Stefan, 2019-10-02:
> Hmm, does that mean that the data output from the camera can actually be delayed if the external trigger is not synchronized with the camera?
No. 
In your sample case, you will get the images excatly when the trigger occurs. Otherwise, you would not be able to trigger with low or changing frequencies

> Is there a way to ensure that the data is immediately sent when there is an external trigger?`
This is definitely ensured.

Stefan

junsujang, 2019-10-02:
In this case, am I understanding it correctly that when I set the framerate to the highest, the datarate to communicate the image over is the maximum? i.e. say that if I used 4FPS setting, this will send data at 20MB/s (arbitrarily chosen) instead of 2FPS which would send data at 10MBs?

TIS-Stefan, 2019-10-07:
Hello

you may let me know, which camera model you use.
Also, the maximum frame rate available with the specified video format fits into the bandwidth of the used bus.
You can use 
tcam-ctrl -c serial
in order to get a list of video formats and their frame rates.

Stefan

closed 2019-11-29 by TIS-Stefan