TheImagingSource/tiscamera #321: Can't change exposure setting to manual from auto

ryoji1415, 2019-10-02:
I want to take image using exposure time set by manual. 
I added ___camera.set_tcam_property("Exposure Auto", False)___ in my program.
But, exposure time was adjusting by auto.
So, I tried sample program, 02-set-properties.py. Following is the result :

    Expposure Auto has value: Continuous
    Gain Auto has value: Continuous
    Exposure has value: 19863
    Expposure Auto has value: Continuous
    Gain Auto has value: Continuous
    Exposure has value: 3000

I guess "Exposure Auto value" on line 4 should be changed because Exposure Auto value was changed to "False" at camera.set_tcam_property("Exposure Auto", False).
Could you please give me any advice?

## PC
Ubuntu 18.04
python 3.6.8

## Camera
DFK 33GP1300

TIS-Stefan, 2019-10-02:
Hello
The exposure and gain automatic values are set as "On" and "Off", not as true and false, which us used for USB cameras. I am sorry for the not unified values.

 camera.set_tcam_property("Exposure Auto", "Off").
 camera.set_tcam_property("Gain Auto", "Off").

You must set both of them. 

Stefan

ryoji1415, 2019-10-02:
Hello Stefan

Thank you for your quickly response.
The exposure automatic value were set as "Off", I can take images by fixed exposure value.

ryoji1415, 2019-10-03:
Hello Stefan

I tried exposure control was changed to auto from manual set, but I cannot change by setting exposure automatic value as "On".
I guess to change exposure control to auto setting, exposure automatic value should be set "Continuous".

Thank you

TIS-Stefan, 2019-10-07:
Thank you for letting me know.

closed 2019-10-07 by TIS-Stefan