TheImagingSource/tiscamera #323: NexImage Burst C

mssc89, 2019-10-11:
Hi!
I was wondering, does this repo support NexImage Burst C (Color) camera? Seems like it's been manufactured by Imaging Source and then rebranded by Celestron, just like NexImage 5 (#29). And hence, it is correctly recognized by firmware-update utility, and I can burn any .euvc I want. As far as I'm aware, it's based on DFK42 camera, but burning its firmware (no matter which one) renders camera unusable - it correctly shows up and mounts in /dev/videoX but trying to display image with VLC/guvcview/oacapture throws errors - or just empty screen. But what is funny, when i tried burning DMK42 firmware, it works perfectly fine, allows me to grab images and stuff but with one problem - only in monochrome. What may be the problem?

TIS-Stefan, 2019-10-14:
Hi

The camera provides bayer raw data only, so you must debayer the images. I am not sure, whether this is in oaCapture implemented.
I suppose, you use Linux , right?

Stefan

mssc89, 2019-10-14:
Yes, I use Linux (Ubuntu 18.04 to be specific).
oaCapture has such a feature, called "demosaic" but it newer gave me good results. I will try again, but I need to know, which DMK42 (or other) firmware would be best?

TIS-Stefan, 2019-10-14:
What does "never good results" mean? Did you select the correct bayer pattern? If yes and the image is green, the there is no white balance adapted.
Firmware file should be dfk42uc03_3014.euvc or dmk42uc03_3014.euvc. Both of them have been tested in the industrial DFK 42 and DMK 42.

TIS-Stefan, 2019-10-15:
Hi
I forgot to answer your question "I was wondering, does this repo support NexImage Burst C (Color) camera? ". It does. 

Stefan

closed 2019-11-29 by TIS-Stefan