TheImagingSource/tiscamera #325: error: ‘libusb_strerror’ was not declared in this scope

doanxuanthy, 2019-11-06:
Hello guys, 
I install follow instruction. When I do step: make 
I have a problem 
[ 57%] Building CXX object src/libusb/CMakeFiles/tcam-libusb.dir/AFU420Device.cpp.o
In file included from /usr/tiscamera-master/tiscamera/src/libusb/LibusbDevice.h:25:0,
                 from /usr/tiscamera-master/tiscamera/src/libusb/AFU420Device.h:25,
                 from /usr/tiscamera-master/tiscamera/src/libusb/AFU420Device.cpp:17:
/usr/tiscamera-master/tiscamera/src/libusb/AFU420Device.cpp: In member function ‘int tcam::AFU420Device::setup_bit_depth(int)’:
/usr/tiscamera-master/tiscamera/src/libusb/AFU420Device.cpp:1089:56: error: ‘libusb_strerror’ was not declared in this scope
                    hr, libusb_strerror((libusb_error)hr));

Could you help me to solve it ? Thank you very much.

TIS-Edgar, 2019-11-07:
Hi,

What version of libusb are you using? 
What distribution are you using?

TIS-Stefan, 2019-11-29:
We got no answer. Is the issue solved?
Stefan

closed 2021-02-19 by TIS-Stefan