TheImagingSource/tiscamera #326: Image is burned and too dark at the same time on DFK27BUP006

alejandronespereira, 2019-11-11:
I am using three DFK27BUP006 cameras and there does not seem to be a set of parameters that give me a good image. There is a high source of light in the office and the image gets burned at its surrounding area, but at the same time the rest of the image looks extremely dark. 

I need these cameras to work in medium to poorly lit environments, and I am having issues with this, using tcam-capture. I can not see this working on any automated fashion. 

Here's a comparison with the best configuration I could achieve compared to an image done by an out of the shelf Ricoh Theta S
[https://imgur.com/a/IbTydIN](url)

What am I doing wrong?

TIS-Stefan, 2019-11-12:
Hi
You are not doing anything wrong. That is simply physic and limitation of the sensor. You scene has too high dynamic. Therefore the sensor is dazzled/blinded on the part with the extremely bright light source. The sensor has not enough dynamic sensitivity for this kind of scene. 

Only thing you could do is closing the lens' iris and use short exposure time, until the extremely bright light source is not more overexposed and then process the image by histogram spreading, so the dark parts become better visible by the eye. 
In case you have a Windows PC, you can use IC Capture for testing and try the RGB64 format and WDR and see, whether you get better results. If this works better, then you could start implementing the same in Linux. I would try to help.

I do not know, what image processing your Ricoh Theta S performs. 

Stefan

alejandronespereira, 2019-11-12:
I am now using RGB8 for image acquisition, both on the tcam-capture and on my gstreamer pipeline I use for acquisition. I've read through the documentation of the sensor and it has a 12bit range. 

How can I use this range?

TIS-Stefan, 2019-11-12:
Hello

First of all, we have two threads now. Either we can continue here, or we can continue by email. I asked my Linux colleague for the dutils stuff. It is not released yet, but maybe he has a alpha version for you.

The pixelformat for 10 or 12 bits is GRAY16_LE. This is 16bit bayer raw data.
The DFK 27UP006 does not provide color data, but bayer raw data only. The debayering is done in software. Unfortunately, there is no GStreamer module, which can debayer 16 bit (which contains 10/12 bits) per channel, but I think, OpenCV can do.

Therefore, I asked for testing Windows  with IC Capture, because there you can see, what is possible.

Stefan

alejandronespereira, 2019-11-12:
Hi Stefan, 

I think it's better if we continue here so anyone with a similar doubt can use this thread in the future. Sorry for the duplication. I'll check on Windows and let you know. 

Thanks!

TIS-Stefan, 2019-11-12:
I tried my version of the not released tcamdutils, mentioned as optional in the documentation, and got it not installed. 

However, trying on Windows is a good start.
Driver : https://www.theimagingsource.com/support/downloads-for-windows/device-drivers/icwdmuvccamtis/
IC Capture : https://www.theimagingsource.com/support/downloads-for-windows/end-user-software/iccapture/

Look for "WDR" in IC Capture.

Stefan

alejandronespereira, 2019-11-12:
I'm having some issues with IC capture. It hangs about 5 seconds after getting images and I have to plug/unplug the camera and reset it every time. Nevertheless, it seems that there is some improvement using WDR. 

I'll let you know when I have more results.

TIS-Stefan, 2019-11-12:
That is weird. Some computer issues?

alejandronespereira, 2019-11-12:
It's ok, I just wanted to check whether WDR was a viable solution for us and it seems it is. We're in a bit of a hurry, developing-wise, and we could really use the alpha version. Could you send me one over mail?

TIS-Stefan, 2019-11-12:
Yes. However, own debayering can be done with OpenCV. 
The WDR can be done with a lookup table, in order to make it fast. The table needs only 4096 entries, because you have 12 bits only.
Maybe OpenCV has functions for this too.

Stefan

alejandronespereira, 2019-11-13:
Hi Stefan, 

I am having some problems with this. I am using tcam-capture to get the 16bit depth images and test opencv-debayering myself before starting developing anything, but I don't think tcam-capture is saving the images on their full width. I have tried both .png and .pnm formats, both should support 16bits  per pixel.

How can I be sure I am saving the full width using tcam-capture? 

In case that's not possible, can you point me in the direction to use a gstream pipeline to record 16 bit data? Here is  my current pipeline, that I use to write into a socket that will be later read by ros gscam package. It's ok if it saves the raw 16bit images in a folder:

`
gst-launch-1.0 tcamsrc serial="11910072" name=source ! video/x-bayer,width=2592,height=1944,framerate=15/1 ! tcamautoexposure exposure-min=0 exposure-max=50000 left=550 top=350 width=1500 height=1200 brightness-reference=128 auto-gain=true gain-min=0 gain-max=16 ! tcamwhitebalance ! bayer2rgb ! videoconvert ! video/x-raw,format=RGB,width=2592,height=1944,framerate=15/1 ! shmsink socket-path=/tmp/socket
`

I guess I have to change the video/x-bayer but I am not sure about any of the other pieces.

TIS-Stefan, 2019-11-13:
Hi

I made a c++ sample:
[gray16-le-sample.zip](https://github.com/TheImagingSource/tiscamera/files/3841208/gray16-le-sample.zip)
It saves the 10th image as binary file, so the raw data is saved in the image callback:
```
        if( strcmp( gst_structure_get_string (str, "format"),"GRAY16_LE") == 0)  
        {
            // Now query the width and height of the image
            gst_structure_get_int (str, "width", &width);
            gst_structure_get_int (str, "height", &height);
            FILE* pFile;
            pFile = fopen("gray16p.binary", "wb");
            fwrite(info.data, info.size, 1 , pFile);
            fclose(pFile);
            //Create a cv::Mat, copy image data into that and save the image.
            
            /*
            pCustomData->frame.create(height,width,CV_16UC(1));
            memcpy( pCustomData->frame.data, info.data, width*height*2);
            char ImageFileName[256];
            sprintf(ImageFileName,"image%05d.bmp", pCustomData->ImageCounter);
            cv::imwrite(ImageFileName,pCustomData->frame);
            */
            printf("Done\n");
        }
```
The part in the comment shows, how to create an 16bit raw data OpenCV Mat from this.
I am not sure, whether the saved BMP file contains really 16 bit or something different.

The sample contains a cmakelists txt. You build that sample with
```
mkdir build
cd build
cmake ..
make
```
You must adapt the serial number, maybe the resolution and the frame rate.

Stefan

mbcel, 2021-07-15:
I tried your proposed approach, however, I am not able to get a color image from the GRAY_16LE.

I try to debayer the image as follows:
```
cv::Mat frame;
cv::Mat frameColor;
frame.create(height, width, CV_16UC1);
memcpy(frame->data, info.data, width * height * 2);
cv::cvtColor(*frame, frameColor, CV_BayerRG2BGR);
```
However, the resulting image looks grayscale. Although the color channels do have slightly different values, e.g bgr=(48398, 48014, 47944).

Are you sure with GRAY_16LE the image is an bayer encoded image?

TIS-Stefan, 2021-07-16:
You may get not a color images, because CV_BayerRG2BGR may is the wrong type? It is 8 bit only? Please check the OpenCV documentation. 
Are you sure with 

> GRAY_16LE the image is an bayer encoded image?

If you have a color camera, yes.

mbcel, 2021-07-17:
Hm no, Opencv CV_BayerRG2BGR does work correctly on Bayer encoded 16bit images that I get from the IC Capture Windows software.

Furthermore, when I look at the raw output images (e.g. write the image to file as .png or .tiff) I get a grayscale image that does not seem to have the typical bayer grid pattern on it. It really looks like a normal grascale image. In comparison, raw saved images with IC capture clearly show the typical bayer pattern when looking at it.

Could it be possible that in the linux examples the gstreamer pipeline is created with some extra component that decodes the bayer encoding somewhere? I am using tcamcamera.cpp class from the Linux examples for creating the pipeline. The relevant parts are as follows:

```
TcamCamera::TcamCamera(std::string serial = "")
{
    if (!gst_is_initialized())
        throw std::runtime_error("GStreamer is not initialized! gst_init(...) needs to be called before using this function.");
    create_pipeline();
    if (serial != "")
        g_object_set(tcambin_, "serial", serial.c_str(), nullptr);
    ensure_ready_state();
    GstElement *src = gst_bin_get_by_name(GST_BIN(tcambin_), "tcambin-source");
    assert(src);
    g_object_set(src, "do-timestamp", false, nullptr);
    gst_object_unref(src);

    videocaps_ = initialize_format_list();
}

TcamCamera::create_pipeline()
{
    pipeline_ = gst_pipeline_new("pipeline");
    tcambin_ = gst_element_factory_make("tcambin", nullptr);
    if (!tcambin_)
        throw std::runtime_error("'tcambin' could not be initialized! Check tiscamera installation");
    capturecapsfilter_ = gst_element_factory_make("capsfilter", nullptr);
    tee_ = gst_element_factory_make("tee", "tee");
    GstElement *queue = gst_element_factory_make("queue", nullptr);
    capturesink_ = gst_element_factory_make("appsink", nullptr);
    g_object_set(capturesink_, "max-buffers", 4, "drop", true, nullptr);
    assert(pipeline_ && tee_ && capturecapsfilter_ && queue && capturesink_);

    gst_bin_add_many(GST_BIN(pipeline_),
                     tcambin_, capturecapsfilter_, tee_, queue, capturesink_, nullptr);
    assert(gst_element_link_many(tcambin_, capturecapsfilter_, tee_, queue, capturesink_, nullptr));
}

TcamCamera::set_capture_format(std::string format, FrameSize size, FrameRate framerate)
{
    GstCaps *caps = gst_caps_new_simple("video/x-raw",
                                        "width", G_TYPE_INT, size.width,
                                        "height", G_TYPE_INT, size.height,
                                        "framerate", GST_TYPE_FRACTION, framerate.numerator, framerate.denominator,
                                        nullptr);
    assert(caps);
    if(format != "")
        gst_caps_set_simple(caps, "format", G_TYPE_STRING, format.c_str(), nullptr);

    g_object_set(G_OBJECT(capturecapsfilter_), "caps", caps, nullptr);
    gst_caps_unref(caps);
}
```

I do use this class like this:
```
TcamCamera tcamCamera("xxxxx");
tcamCamera.set_capture_format("GRAY16_LE", gsttcam::FrameSize{width, height}, gsttcam::FrameRate{fps, 1});
```

TIS-Stefan, 2021-07-19:
Hello

16 bit handling is always somewhat tricky, because most programs do not support 16 bits. I made some sample pipelines, that show how to get raw data at https://theimagingsource.deskpro.com/de/kb/articles/linux-set-video-formats-in-camera-and-memory
You need to exchange the pipeline in the tis class or you may write your own class with correct numpy conversion. 

Stefan

mbcel, 2021-08-05:
I tried the suggested pipeline for 16bit bayer but I am not able to start the stream:

```
gst-launch-1.0 tcambin ! "video/x-bayer,format=rggb16,width=1440,height=1080,framerate=30/1" ! capssetter join=false replace=true caps="video/x-raw,format=GRAY16_LE,width=1440,height=1080,framerate=30/1" ! videoconvert ! videoscale !  ximagesink 
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...

```

Furthermore I realized that tcam-ctrl -c <serial> just gives me:

```
tcam-ctrl -c xxx
Available gstreamer-1.0 caps:
video/x-bayer,format=rggb,width=1440,height=1080,framerate={2500000/10593,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=640,height=480,framerate={600/1,480/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1}

```

Why are there no rggb16 streams available? I am using Linux. Under Windows IC Capture bayer16 is available for this camera. Am I missing something here?

mbcel, 2021-08-05:
Your provided example only shows how to set a pipeline with GRAY16_LE. However this resulted only in a normal grayscale image without the bayer pattern on it (as asked above earlier). You then sent me [this](https://theimagingsource.deskpro.com/de/kb/articles/linux-set-video-formats-in-camera-and-memory) example pipeline that I just tried where I get no output at all. This pipeline is different to the GRAY16_LE pipeline in the cpp example, which does not use an element like the "capsetter". I would like to implement this pipline with an appsink however I cannot find any resources how to set up the "capsetter" in cpp which seems to be important. Do you have any examples to this?

Furthermore, to me it seems that the camera somehow does not support rggb16 in my Linux install. `tcam-ctrl -c xxx` only shows rggb no rggb16 format. As shown above. Is this relevant?

Doing this:
`gst-launch-1.0 tcambin ! "video/x-bayer,format=rggb16,width=1440,height=1080,framerate=30/1" ! capssetter join=false replace=true caps="video/x-raw,format=GRAY16_LE,width=1440,height=1080,framerate=30/1"` without videoconvert and imagesink does also give the same error. Which to me looks more like it cannot find rggb16 format?

TIS-Stefan, 2021-08-05:
I would like to suggest to adapt the pipeline in the sample to your needs.
I only wanted to demonstrate with this sample, how you could use 16 bit formats in own programs.

TIS-Stefan, 2021-08-05:
I just see, your camera does not offer a 16 bit format at all.

video/x-bayer,format=rggb,width=1440,height=1080,framerate={2500000/10593,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=640,height=480,framerate={600/1,480/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1}

Unfortunately, you pass "xxx" as serial, so I am not able to check, which camera you use. I am sorry for a wrong guess. However, the tcam-ctrl output explains, why you can not use rggb16.

mbcel, 2021-08-05:
It is this camera model: DFM 37UX273-ML
In Windows and IC Capture I can use rgb64 (10bit apparently encoded in 16bit). Only in my Linux install this somehow is not avaliable...

mbcel, 2021-08-05:
I double checked with a different camera model: DFM 37UX178-ML
Same problem: only rggb and no rggb16

TIS-Stefan, 2021-08-06:
Hello

I checked with my Nano. I get 
```
video/x-bayer,format=rggb,width=640,height=480,framerate={280/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=1536,height=1024,framerate={137/1,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=1920,height=1080,framerate={130/1,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=2048,height=2048,framerate={70/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=3072,height=2048,framerate={60/1,30/1,15/1,5/1,1/1};
```
Then I checked on a desktop computer with Ubuntu:
```
video/x-bayer,format=rggb,width=640,height=480,framerate={280/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=1536,height=1024,framerate={137/1,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=1920,height=1080,framerate={130/1,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=2048,height=2048,framerate={70/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=3072,height=2048,framerate={60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=640,height=480,framerate={280/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=1536,height=1024,framerate={120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=1920,height=1080,framerate={90/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=2048,height=2048,framerate={45/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=3072,height=2048,framerate={30/1,15/1,5/1,1/1}
```
I guess, the v4l2 driver of the Nano does not know the 16 bit formats.

Therefore, your suggestion  TCAM_ARAVIS_USB_VISION=ON could solve the issue. I must admit, I did not test that.

Stefan

mbcel, 2021-08-06:
I got the imagesink gstreamer pipline working so far with aravis and rggb16 on Jetson Xavier. However, I could not test yet if the actual appsrc output really is 16bit depth since I did not write the corresponding code yet. I will report if that works later.

Furthermore with aravis the image stream starts to hang for framerates of >50fps Gray16 or 120 fps Gray8. This does not happen for 120fps Gray8 with the non-aravis gstreamer pipeline. Could this be due to aravis somehow using more memory? Do I have to increase usb memory size? Or do you have any other idea how to solve this?

TIS-Stefan, 2021-08-09:
Hello

From the sample at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/OpenCV-Gstreamer-GetImage/main.cpp you can copy the function "GstFlowReturn new_frame_cb(GstAppSink *appsink, gpointer data)" and use a callback for the appsink. It creates OpenCV::Mat according to the pixel format in the GstSample you receive.

> Furthermore with aravis the image stream starts to hang for framerates of >50fps Gray16 or 120 fps Gray8. This does not happen for 120fps Gray8 with the non-aravis gstreamer pipeline. Could this be [...]

This may is an issue of CPU load, so the USB controller can not save all image data packets into memory. With Gray16 you have the double (well, not really, but almost) bandwidth allocation than you have for Gray8. Therefore, you may check for CPU load.
Aravis is not as efficient as v4l2, this may also an issue. I have no real good idea, how to solve it.

> Furthermore according to the manual the camera should support Bayer12bitPacked. When using the aravis viewer however I cannot select this format and instead Bayer12bitPacked and Bayer16 is grayed out. Why is that?

I suppose, it is grayed out, because there are no converters for this format to be displayed in the viewer. That is a question to ask the Aravis author at https://github.com/AravisProject/aravis.
As far as I know, we do not use the latest Aravis version, because the build process became somewhat more complicated.

Stefan

mbcel, 2021-08-09:
I got the code working so far with callback etc., however, I do not know how to implement the part:

```... ! capssetter join=false replace=true caps="video/x-raw,format=GRAY16_LE,width=1440,height=1080,framerate=30/1" ! ...```

from the command based pipeline in code.

I use this
```
GstCaps *caps = gst_caps_new_simple("video/x-bayer",
                                        "format", G_TYPE_STRING, "rggb16",
                                        "width", G_TYPE_INT, size.width,
                                        "height", G_TYPE_INT, size.height,
                                        "framerate", GST_TYPE_FRACTION, framerate.numerator, framerate.denominator,
                                        nullptr);
```
to define the capsfilter to get the raw bayer output of the camera, after that I somehow need to define the "capsetter" but I can not find any function to create this class nor can I find any good documentation about it. How can I create this capsetter?

Using the pipeline with "Gray16_LE" without capsetter does only give a normal grayscale image without bayer pattern.

mbcel, 2021-08-09:
Okay I got it to work. It looks all correct so far, however, I could not thoroughly test it yet. If I find any other problems later I will open a new issue then.

TIS-Stefan, 2021-08-09:
Oh, I just was finished with another task and wanted to start here.

Good to know, you get raw data now.

You may create a new issue or support case at https://www.theimagingsource.com/support/contact/ if you encounter an issue.

Stefan

closed 2021-08-09 by TIS-Stefan

mbcel, 2021-08-09:
Okay one last question that I just find out and gonna post here for completion.

I just realized that the 16bit pixel data comes in step size of 16 which would make sense for a 12bit sensor mapped to 16bit (2^16/2^12). However according to the manual of the used camera (DFM 37UX273-ML) the sensor should only be capable of 10bit pixel values, so it should be a step size of 2^16 / 2^10 = 64. Does Aravis mess up something here or magically turn this sensor from 10bit to 12bit, or am I misunderstanding something?

mbcel, 2021-08-09:
Okay got it. This was a result of the debayering. The raw output has step size of 64.

TIS-Stefan, 2021-08-09:
Hello
The lower 4, resp. 6 bits are not relevant. But you get 16 bits as uint16

Stefan