TheImagingSource/tiscamera #327: DMK 33UP camera issues in raspberry

emejoti, 2019-12-10:
Hello,

I'm writting a Python code to create a streaming view for a DMK33UP5000 camera in a raspberry pi 3B+. I'm using as examples the ones i found in this git, like: https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Snap%20an%20Image%20and%20convert%20to%20OpenCV%20in%20Python and https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/03-live-stream.py . 
I'm also trying the command ` gst-launch-1.0 tcambin serial="X" ! videoconvert ! ximagesink` but I got this: 
`Estableciendo el conducto a PAUSA …
ERROR: El conducto no quiere pausarse.
Estableciendo el conducto a NULL …
Liberando la tubería…
`
It's like the raspberry can't find the camera. Before that command I insert this one `tcam-ctrl -l` and it give me the camera model, type and serial, but after the gst-launch-1.0, if I try again the tcam-ctrl command, no devices are found. 

Any help you can provide would be great

TIS-Stefan, 2019-12-10:
Hello

The PI3 has USB 2.0 controller only, therefore, you can not use the DMK 33UP5000 on it:
- too slow USB speed
- too less USB power

I am sorry, bu it is the wrong camera for your computer.

Stefan

emejoti, 2019-12-10:
Hi Stefan,

I was hoping that It would work even with the 2.0 USB.

Thaks for your quick answer.

closed 2019-12-10 by emejoti