TheImagingSource/tiscamera #328: SEG Fault in image acquisition

sarthak-gupta-sg, 2019-12-26:
I have adapted the Linux-tiscamera-Programming-Samples/simple-snap-image-in-c++ program for image acquistion with qt. However, during some acquisitions, I get a segmentation fault randomly, whereas during other times the program works correctly. Following is my adaptation:

```c++
int acquireImage(std::string camName, Ui::MainWindow *ui)
{
    bool fail = true;
    cv::Mat OpenCVImage;
    /* Initialize our TcamCamera object "cam" with the serial number
     * of the camera, which is to be used in this program.
     */
    //TcamImage cam("42719953");
    //TcamCamera cam("00001234");
    TcamImage cam(camName);

    // Set a color video format, resolution and frame rate
    cam.set_capture_format("GRAY8", FrameSize{960, 600}, FrameRate{30,1});

    // Comment following line, if no live video display is wanted.
    //cam.enable_video_display(gst_element_factory_make("ximagesink", NULL));

    // Start the camera
    cam.start();

    std::this_thread::sleep_for(std::chrono::milliseconds(2000)); //2 second margin for camera to attain optimum exposure

    for(auto i = 0; i < init.num_of_images; ++i)
    {
        QString writeout = "Acquiring Image " + QString::number(i + 1) + " of " + QString::number(init.num_of_images) ;
        ui->textBrowser->append(writeout);
        ui->progressBar->setValue( i * 3 );
        QApplication::processEvents();

        // Snap an Image with 60 ms timeout. (DOUBLE THE FRAME RATE) Should be set accordingly to frame rate.
        if( cam.snapImage(500) )
        {
            // On succes do something with the image data. Here we create
            // a cv::Mat and save the image
            OpenCVImage.create( cam.getHeight(),cam.getWidth(),CV_8UC(cam.getBytesPerPixel()));
            memcpy( OpenCVImage.data, cam.getImageData(), cam.getImageDataSize());

            //Perform median smoothing with 7x7 kernel
            cv::medianBlur(OpenCVImage, OpenCVImage, 7);

            //Save Image
            std::string fname = "Images/" + std::to_string(i + 1) + ".bmp";
            cv::imwrite(fname, OpenCVImage);

            if(i == 4)
            {
                //Final state reached no more motor rotation needed
                fail = false;
                break;
            }

            //Rotate by 45 degrees
            SerialDrive(init.cmd);
            std::this_thread::sleep_for(std::chrono::milliseconds(init.milliseconds)); //9 seconds required by controller to rotate motor by 45 degrees
        }
        else
        {
            printf("Timeout at snapImage()\n");
            break;
        }
    }

    cam.stop();
    OpenCVImage.release();

    if(fail)
    {
        return -1;
    }
    return 0;
}
```
On printing the stack trace I have been able to isolate the issue with cam.close() call. Unfortunately, since I am newbie in C++ programming, I have not saved the executable that produced the core file (only the core file is with me). Therefore the symbol table is lost. 

Also, I tried to keep the camera open during full processing cycle. However, that leads to seg faults every time during processing.
Kindly help me to fix the issue.

TIS-Edgar, 2020-01-06:
Hi,

Thank you for your report.
Did you configure tiscamera with the cmake option `-DCMAKE_BUILD_TYPE=Debug`?
The segfault does not contain useful information due to missing debug symbols.

Could you either send me the entire code of the project or unset the camName?
From what I can gather you have hardcoded the camera serial.
That prevents me from reproducing the segfault.
You can send the code to support@theimagingsource.com if you do not want to publicly disclose it.

Furthermore:
Could you give me the output of your application with the following environment variables set.

```
export TCAM_LOG=DEBUG
export GST_DEBUG=tcamsrc:5
Fizeau
```
This enables gstreamer logging and tiscamera internal logging to the console giving us an insight as to what is happening.
From the looks of it the camera is not opened and thus crashes later on.
Which still should not happen but one step at a time.

-Edgar

sarthak-gupta-sg, 2020-01-07:
Hello @TIS-Edgar,

Thanks for your reply,

> Could you either send me the entire code of the project

I am attaching the full source code. 
[FizeauCPP.zip](https://github.com/TheImagingSource/tiscamera/files/4030857/FizeauCPP.zip)
I have added a README_INSTALL file for creating the correct folder structure for the binary.

> Could you give me the output of your application with the following environment variables set.

I am attaching output from gstreamer and tiscamera internal logging. 
[log.txt](https://github.com/TheImagingSource/tiscamera/files/4030766/log.txt). On examing the log I am finding an error statement which states: 
`1548492    <ERROR> /home/bvtest/tiscamera/src/v4l2/V4l2Device.cpp:104: Error closing udev monitoring pipe. write return '18446744073709551615'`

> Did you configure tiscamera with the cmake option -DCMAKE_BUILD_TYPE=Debug?

I am using the production release version (.deb) of tiscamera (0.11.1) i.e. the one on this link [https://www.theimagingsource.com/support/downloads-for-linux/install/tiscamera/](url). So I have not build the library. However, my Qt project was configured in release mode. I have switched back to debug mode.

> From what I can gather you have hardcoded the camera serial.

The camera serial number is a user supplied value, read from a Qt interface variable.

> From the looks of it the camera is not opened and thus crashes later on.

The camera gets opened and acquires images. The seg fault occurs only after acquiring all images. Also, the seg fault is random. Generally it occurs when the acquisition------>processing cycle has been completed a few times. 

Side Note: There is a bug in tcam-capture utility in version 0.11.1. On trying to invoke it, the error is:
`
Traceback (most recent call last):
File "/usr/bin/tcam-capture", line 28, in <module>
from tcam_capture.TcamDevice import TcamDevice, TcamDeviceIndex
ModuleNotFoundError: No module named 'tcam_capture'`
This error was not present in the previous release. If the older .deb installer (version 0.10) is used, tcam-capture starts correctly.

TIS-Edgar, 2020-01-10:
I have no idea why your program is triggering this problem and not any of our tests/applications.
You seem to be running in a timing issue with the destruction of buffers after the tcamsrc is already destroyed.
Can you install this version for me:
[tiscamera_0.11.1_amd64.deb.zip](https://github.com/TheImagingSource/tiscamera/files/4045023/tiscamera_0.11.1_amd64.deb.zip)
It contains a fix for the segfault I was able to identify.

I have not yet had the time to look into the tcam-capture installation issue you are experiencing. That will be next.

sarthak-gupta-sg, 2020-01-14:
I shall try out the version this week at the optical workshop where the software is currently deployed. 

Regarding the tcam-capture issue, I noticed that the underlying `tcam_capture` is installed at location 
`/usr/lib/python3.5/site-packages/tcam_capture/` . However the default python3 in ubuntu 18.04 is python3.6 and thus python3.5/site-packages is not in the path.

I have Ubuntu 18.04.3 at both my development workstation and deployment site.

sarthak-gupta-sg, 2020-01-17:
@TIS-Edgar 

The new version provided by you is working as expected. I have not seen any seg faults in over 30 continuous runs of my application (Earlier the application used to seg fault within 6 to 10 runs). I have tried on both my development workstation as well as the deployment machine at the site.

Also, by following the examples in the `tiscamera` repository, I have improved the application to automatically list the attached camera devices and find out the serial number for further use. I have also set the "Exposure Auto"  and "Gain Auto" as follows:
```cpp
GValue set_auto = G_VALUE_INIT;
sarthcam.set_property("Exposure Auto", set_auto);
sarthcam.set_property("Gain Auto", set_auto);
sarthcam.start();
std::this_thread::sleep_for(std::chrono::milliseconds(2000));
```
However, I have noticed that on cold boot of camera, all acquired images are dark. From second run onwards the camera exposure is working as expected. Is the above code correct? Do I need to increase the thread sleep time?

Regarding `tcam_capture` install issue, it is also present in the version linked by you. The version 0.10.0 installed in `/usr/lib/python3/dist-packages` and thus was available in the path.

So, I created a soft link as follows:
```bash
sudo ln -s /usr/lib/python3.5/site-packages/tcam_capture /usr/local/lib/python3.6/dist-packages/tcam_capture
```
The application starts and detects the camera but there is no view from camera. Also following errors are generated on the command line:
```bash
sarthak@workstation:~/Documents/SarQT/build-FizeauCPP-Desktop-Debug/FizeauCPP$ tcam-capture 
17-01-2020:16:18:14,914 ERROR    [PropertyWidget.py:94] Property Gain Max reported a range that could not be handled
17-01-2020:16:18:14,959 ERROR    [TcamView.py:460] ERROR: bin : Could not create element 'tcamautoexposure'.
17-01-2020:16:18:16,829 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:18,744 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:20,742 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:22,743 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:24,743 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:26,743 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:28,744 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:30,742 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:32,743 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
17-01-2020:16:18:34,743 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
```

TIS-Edgar, 2020-01-17:
Hi,

It is good to hear that that really was the cause of the segmentation faults you experienced. 
I will try to make it available on the master branch as soon as i can.

Concerning gvalues:
```
    GValue set_auto = G_VALUE_INIT;
    g_value_init(&set_auto, G_TYPE_BOOLEAN);
    g_value_set_boolean(&set_auto, TRUE);
```

`G_VALUE_INIT` is defined as `#define G_VALUE_INIT  { 0, { { 0 } } }`
That means it still requires information about the type you use and the value.

Concerning the installation issues of python modules:

The code to determine the installation path changed because the code in 0.10.0 did not work on __some__ systems like NVidia Jetsons.
I am looking into how that code can be made more generic to simply install into `/usr/lib/python3/`.

Concerning your last error:

`bin : Could not create element 'tcamautoexposure'.` 
Does a pipeline launched with gst-launch-1.0 work?
Your application also uses tcambin and does not report this error right?
Assuming both your application and tcam-capture use the same environment they should both work .....

I will have to think about that one and report back to you.

sarthak-gupta-sg, 2020-01-29:
@TIS-Edgar 
`That means it still requires information about the type you use and the value.`
Understood. I have updated the code as follows:
```cpp
    GValue set_auto = G_VALUE_INIT;
    g_value_init(&set_auto, G_TYPE_BOOLEAN);
    g_value_set_boolean(&set_auto, TRUE);
    sarthcam.set_property("Exposure Auto", set_auto);
    sarthcam.set_property("Gain Auto", set_auto);
```

`Does a pipeline launched with gst-launch-1.0 work?`
Yes. Infact I am using it directly in my application also to provide live view to the users. Here is the function:
```cpp
void MainWindow::on_alignButton_clicked()
{
    system("gst-launch-1.0 tcambin ! video/x-raw, format=GRAY8, width=960, height=600, framerate=15/1 ! videoconvert ! ximagesink");
    return;
}
```
`Your application also uses tcambin and does not report this error right?`
Correct. My application doesn't report this error using both 0.11.0 as well as 0.11.1 provided by you.

Is it possible to have a version of 0.11.1 compiled and packaged as .deb for Raspberry Pi platform?

TIS-Edgar, 2020-01-29:
You can always create a valid package for additional plattforms on your own.
In fact it is recommended. Since you do not use GigE cameras you can disable aravis support and minimize dependencies. Our packages are always genereic and built to include all cameras, etc.
Simply build the project and call `make package`. See https://www.theimagingsource.com/documentation/tiscamera/packaging.html for additional information.

sarthak-gupta-sg, 2020-02-11:
@TIS-Edgar 
I am trying to disable auto exposure and set the value manually. I have tried following code:
```c++
    GValue set_auto = G_VALUE_INIT;
    g_value_init(&set_auto, G_TYPE_BOOLEAN);
    g_value_set_boolean(&set_auto, FALSE);
    sarthcam.set_property("Exposure Auto", set_auto);
    sarthcam.set_property("Gain Auto", set_auto);

    GValue val = G_VALUE_INIT;
    g_value_init(&val, G_TYPE_INT);
    g_value_set_int(&val, 15);
    sarthcam.set_property("Exposure", val);
```
However, it is having no effect at all. Also I tried to set the Exposure to manually set value using tcam-ctrl by issuing following command:
```bash
tcam-ctrl -p -s "Exposure=15" 33810623
```
The command results in following:
`No property with name 'Exposure'`

Our camera is DMK23UX236.

Side Note: I am attaching a preliminary draft of a 2 page document where I have compared results from DMK23UX236 and Basler ace1300-30gm. In the table on first page you can see that DMK23UX236 has some wild swings in the measured value. I request you to go through the document and suggest me any acquisition parameter that I can control so as to improve the result.
[SG-OSTI-v2.pdf](https://github.com/TheImagingSource/tiscamera/files/4185440/SG-OSTI-v2.pdf)

TIS-Stefan, 2020-02-11:
Instead of "Exposure", you may try "Exposure in (us)". You can use tcam-capture for listing property names.

sarthak-gupta-sg, 2020-02-11:
@TIS-Edgar 
Got it. The property is `Exposure Time (us)`. Just tried it in code, it is working and visible in tcam-ctrl also. I shall try it in the shop floor and report back about results.

As discussed above, tcam-capture is currently not working for us.

sarthak-gupta-sg, 2020-07-08:
@TIS-Edgar @TIS-Stefan 
I have deployed version 0.12 on production systems at our site. This version solves all of the issues viz. the seg fault issue as well as the tcam-capture startup issue. The development and production system are running Debian 10 amd64. This issue can be closed.

closed 2020-07-08 by sarthak-gupta-sg