TheImagingSource/tiscamera #329: Image overexposed and too dark at the same time; DFK 37AUX265

omarsomey, 2020-01-14:
Hi, 
I'm in a kind of a hurry to meet a client's deadline.
I'm using one DFK 37AUX265 camera and there does not seem to be a set of parameters that give me a good image under outdoor lightening. Usually there is a high source of light and a high dynamic range for the application and the image tends to get overexposed at some areas, what set of parameters is suitable and how can I implement WDR on linux ?
here is my gstreamer pipeline:
"tcambin serial="" name=source ! video/x-raw,format=RGBx,width=1920,height=1080,framerate=30/1 ! appsink name=sink" 
I'm using python for the application.

TIS-Stefan, 2020-01-14:
Hello

For Linux I have currently solution for using the 10bit range per color channel of your camera. This is currently implemented in Windows with WDR/Tonemapping property only. We still work on a closed source library containing these algorithms.

So, what you can do on your own:
First of all, overexposed areas do not contain any information. Thus, avoid overexposed areas. This, of course, results in a dark image. But the dark areas usually contain enough information, so with a more or less smart histogram spreading you can brighten up the dark areas and lower the bright ones.
You can use 
I did not test on my own, but you may try following modules:
https://gstreamer.freedesktop.org/documentation/frei0r/frei0r-filter-equaliz0r.html?gi-language=c#frei0r-filter-equaliz0r (I do not know, how to install that)

https://gstreamer.freedesktop.org/documentation/videofilter/gamma.html?gi-language=c#gamma
! videoconvert ! gamma gamma=0.2 ! ximagesink
Play with values less 1.0 or above.
You may find more filters at https://gstreamer.freedesktop.org/documentation/plugins_doc.html?gi-language=c

Stefan

omarsomey, 2020-01-14:
Thanks for the quick reply,
I tested the WDR/Tonemaping property on Windows and the image is perfectly fine. 
So to be sure there is no such property available for linux as of right now and if not do you have a specific timeline?
One more question before testing the Histogram spreading,
Is it normal for the Highlight Reduction to be kind of slow ?

TIS-Stefan, 2020-01-14:
The tonemapping is done inside the Windows driver, thus it is done in software.
It is currently not avaialble in Linux, we are in process of writing that as Gstreamer module.

> Is it normal for the Highlight Reduction to be kind of slow ?
Depends on the meaning of "slow".  A particular mimium area of the image must be over exposed, until this property comes into effect.

omarsomey, 2020-01-14:
Thanks a lot. Good luck

TIS-Stefan, 2020-01-14:
> Good luck
Hopefully it is not a matter of luck. But the current version is far too slow.

Did you try the "gamma" module?

omarsomey, 2020-01-15:
I'm actually testing with the highlight reduction and playing with the gamma values to increase the darker areas of the frames, it's definitely better but still we're loosing too much information in these underexposed areas. Getting out of a tunnel for example is dazzling the sensor.
The application is in the context of Autonmous driving , should we consider another camera that performs HDR on the sensor level like the CMOS IMX390 ?

TIS-Stefan, 2020-01-15:
Hello
Getting out of a tunnel, in particular by car with higher speed than a pedestrian is indeed an issue. 

Currently I have no solution. Our Gstreamer module with tonemapping support makes progress, but it is not releasable  or testable yet.

I can not say anything about the IMX390 here.

I am sorry for this.

omarsomey, 2020-01-16:
Well thanks anyways.

TIS-Stefan, 2020-01-16:
You may ask your question directly at https://www.theimagingsource.com/support/contact/, because, If I am right, the IMX390 will be available soon  for MIPI directly on TX Nano and TX2.

omarsomey, 2020-01-21:
Hi Stefan,
If I consider changing to Windows, will the HDR/tonemapping property be available ?

TIS-Stefan, 2020-01-21:
> If I consider changing to Windows, will the HDR/tonemapping property be available ?

Yes. It is implemented in our drivers.

Stefan

omarsomey, 2020-01-21:
I'd definitely consider changing to Windows if I could improve the image exposure.
is there like major changes in the python code ?

TIS-Stefan, 2020-01-22:
> is there like major changes in the python code ?
Unfortunately, yes. The Windows SDK is completely different and the Python support is poor. In Windows we are more in C#, C++ and VB.NET

Stefan

closed 2021-02-19 by TIS-Stefan