TheImagingSource/tiscamera #330: Cannot get Exposure but can set it

guppykang, 2020-01-15:
Hello, 
I am having an issue where the exposure is successfully set in the beginning of streaming : 
`tiscam_right.Set_Property("Gain Auto", False)
        tiscam_right.Set_Property("Gain", 160)
        tiscam_right.Set_Property("Exposure Auto", False)
        tiscam_right.Set_Property("Exposure", 128)
        tiscam_right.Set_Property("Exposure Time (us)", 29000)`
I then attempt to tiscam_right.Get_Property("Exposure").value but get this error : ```
return CameraProperty(*self.source.get_tcam_property(PropertyName))
TypeError: unknown type (null)
``` 
This makes sense since tcam-ctrl -p <serial> has no exposure property. However, I checked that the set is working on tcam-capture and I have confirmed the "set" is working. Why can I not Get?

TIS-Stefan, 2020-01-16:
Which camera model do you use?
Maybe you can use "Exposure Time (us)"" only.

guppykang, 2020-01-17:
DFK 33GX264

TIS-Stefan, 2020-01-20:
Hello
The interface GigE and USB cameras are somewhat different, unfortunately. Now, that I know you use a Gige Camera, I found the error:
Wrong:
tiscam_right.Set_Property("Exposure Auto", False)
Correct
tiscam_right.Set_Property("Exposure Auto", "Off")
Then

TIS-Stefan, 2020-01-20:
You wrote

> This makes sense since tcam-ctrl -p has no exposure property. 
On my comptuer it has:
```
Exposure Auto Upper Limit Auto          (bool) default=true value=true category=Exposure
Exposure Auto Upper Limit               (double) min=60 max=5e+07 step=0.001 default=33333 value=33333 category=Exposure group=429785090
Exposure Auto Lower Limit               (double) min=60 max=5e+07 step=0.001 default=60 value=60 category=Exposure group=429785090
Exposure Auto Reference                 (int) min=0 max=255 step=1 default=128 value=128 category=Exposure
Exposure Auto                           (enum)  default=Continuous category=Exposure
						value=Continuous
							Continuous
							Off
Exposure                                (int) min=60 max=50000000 step=1 default=333 value=333 category=Exposure
```
Can you please show me your tcam-ctrl -p <serial> output?
Can you also please show me your project source code or make a small project showing the issue?

Thank you in advance!

cSDes1gn, 2021-02-10:
+1 to OP issue:
```bash
self.source.get_tcam_property("Exposure Time (us)")
TypeError: unknown type (null)
```

This issue is non-repeatable for whatever reason.

TIS-Stefan, 2021-02-11:
I must admit, you lost me here a little bit. We have to different camera models. I do not the issue posted by cSDes1gn.  Please help me here.
Thank you!

Stefan

closed 2021-02-19 by TIS-Stefan