TheImagingSource/tiscamera #332: GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1

loyclay, 2020-01-23:
Fails to load/open a video stream from cv2.VideoCapture format 1280* 960 using DFK 42BCU03 on an appsink. 
A video of resolution 640 * 480 loads, but the same error is generated.

`[ WARN:0] global /home/pi/opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1`

I checked that the camera supports these video formatting.
`Available gstreamer-1.0 caps:
video/x-raw,format=GRAY8,width=1280,height=960,framerate={30/1,25/1,15/1,10/1};
video/x-raw,format=GRAY8,width=640,height=480,framerate={60/1,30/1,25/1,15/1};
video/x-raw,format=GRAY8,width=320,height=240,framerate={60/1,30/1,25/1,15/1};
video/x-bayer,format=grbg,width=1280,height=960,framerate={30/1,25/1,15/1,10/1};
video/x-bayer,format=grbg,width=640,height=480,framerate={60/1,30/1,25/1,15/1};
video/x-bayer,format=grbg,width=320,height=240,framerate={60/1,30/1,25/1,15/1}`

I checked this error on `gst_element_query_position` (line 935) of [Gstreamer.cpp](https://github.com/opencv/opencv/blob/master/modules/videoio/src/cap_gstreamer.cpp) , and could not understand much.

TIS-Stefan, 2020-01-23:
Which pipeline do you pass to cv2.VideoCapture()?

loyclay, 2020-01-23:
pipeline on cv2.videocapture `("tcambin serial=50910677 ! videoscale ! video/x-raw, format=BGRx,width= 1280,height=960,framerate=20/1 ! videoconvert ! appsink", cv2.CAP_GSTREAMER)`

TIS-Stefan, 2020-01-24:
oh, that pipeline makes no sense, because the videoscale gets a fixed videoformat size. Better do:

("tcambin serial=50910677 !  video/x-raw, format=BGRx,width= 1280,height=960, framerate=20/1 ! videoconvert ! videoscale ! appsink", cv2.CAP_GSTREAMER)

I wonder, whether the videoconvert and videoscale make sense at all, because the appsink accepts all.  They are needed, if you display the live video in an imagesink. With this pipeline you will get RGB32 format in the sink in byte order b g r a. The video scale does nothing, because the appsink does not change the display resolution. Thus your pipeline should look like:

("tcambin serial=50910677 !  video/x-raw, format=BGRx,width= 1280,height=960, framerate=20/1 !  appsink", cv2.CAP_GSTREAMER)

Also you will receive a not "negotiated error", because the camera does not support 20 fps. Please use 30, 25, 14 or 10 frames per second. No values in between, the v4l2 driver does not support his.

Stefan

loyclay, 2020-01-28:
Well, I followed your suggestions for 
`"tcambin serial=50910677 ! video/x-raw, format=BGRx,width= 1280,height=960, framerate=20/1 ! videoconvert ! videoscale ! appsink", cv2.CAP_GSTREAMER`

and i still get the same error as before `[ WARN:0] global /home/pi/opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1`

From the second suggestion; 

> tcambin serial=50910677 ! video/x-raw, format=BGRx,width= 1280,height=960, framerate=25/1 ! appsink", cv2.CAP_GSTREAMER)

I get the following errors;

`[ WARN:0] global /home/pi/opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module tcambin-source reported: Internal data stream error.
[ WARN:0] global /home/pi/opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/pi/opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created`

You are right removing videoscale! makes no change

> "tcambin serial=50910677 ! video/x-raw, format=BGRx,width= 1280,height=960, framerate=20/1 ! videoconvert ! appsink", cv2.CAP_GSTREAMER

same error generated as previously.

TIS-Stefan, 2020-01-28:
Hello

I suggest as first step to get the camera showing a live video on your PI from the terminal

 Get the available video formats and frame rates using tcam-ctrl:

tcam-ctrl -c 50910677 
You should receive
Available gstreamer-1.0 caps:
video/x-raw,format=GRAY8,width=1280,height=960,framerate={30/1,25/1,15/1,10/1};
video/x-raw,format=GRAY8,width=640,height=480,framerate={60/1,30/1,25/1,15/1};
video/x-raw,format=GRAY8,width=320,height=240,framerate={60/1,30/1,25/1,15/1};
video/x-bayer,format=grbg,width=1280,height=960,framerate={30/1,25/1,15/1,10/1};
video/x-bayer,format=grbg,width=640,height=480,framerate={60/1,30/1,25/1,15/1};
video/x-bayer,format=grbg,width=320,height=240,framerate={60/1,30/1,25/1,15/1}
Choose a valid frame rate from this list. 

A pipeline would be
`gst-launch-1.0 tcambin ! video/x-raw, format=BGRx, width=1280,height=960, framerate=10/1 ! videoconvert ! ximagesink
`
You should see a colored live video from the camera with 10 frames per second. 
Does this work? Then we can step over to OpenCV.
If the camera pipeline does not work, you may try instead
videotestsrc ! videoconvert ! appsink
I am not sure, whether videoconvert is necessary, because the appsink accepts all. It is be necessary, if you want a particular pixel format , e.g. BGR in the sink.

Also I searched for your error message in the internet and found: https://github.com/opencv/opencv/issues/10324

Currently I try to rebuild opencv for use with GStreamer, but my Linux hangs doing so.

Stefan

TIS-Stefan, 2020-01-28:
I rebuilt and installed OpenCV successfully. The code
```
const char *pipeline = " tcambin serial=15810833 ! video/x-raw, format=BGRx, width=1280,height=960, framerate=25/1 ! videoconvert ! appsink";
cv::VideoCapture *cap = new cv::VideoCapture(pipeline, cv2.CAP_GSTREAMER);
```
works perfectly on my system.

(Annotation: I use latest OpenCV 4 from OpenCV Github)

Stefan

loyclay, 2020-01-29:
I rebuilt my OpenCV and now its solved, thanks!

loyclay, 2020-02-20:
I would like to know what conditions `gst_element_query_position`  in line 933 of [Gstreamer.cpp](https://github.com/opencv/opencv/blob/master/modules/videoio/src/cap_gstreamer.cpp#L933) 

Based on [opencv/opencv#10324] (https://github.com/opencv/opencv/issues/10324). The issue originates from opencv, and is still unresolved

TIS-Stefan, 2020-02-24:
The stream position is determined by Gstreamer, not by the camera video source. Since we created a live source with the cameras the duration is always undetermined. 
Maybe the programming team finds a workaround.

arnfaldur, 2020-06-18:
I encountered the exact same error and the problem turned out to be conflicting opencv installs. I had 4.3 installed on arch linux through pacman while my project was using opencv 4.2 from pip. Uninstalling 4.3 with pacman solved the issue for me.

Hope this helps.

tkasidakis, 2020-09-05:
`def shootPicture(self):

		print("shootPicture() invoked")
		
		camera = cv2.VideoCapture(0) #  main camera

		time.sleep(0.1)  # If you don't wait, the image will be dark

		return_value, image = camera.read() # shoot a picture

		#cv2.imwrite("opencv.png", image) # save it localy

		del(camera)  # so that others can use the camera as soon as possible
		#print("invoked from a mobile service in order to take a picture. Now i am returning the picture")
		cv2.imwrite("pic.png", image) # write the image locall , so you can send bytes of the network
		f = open("pic.png","rb") # open a file descriptor in order to read the picture
		picbytes = f.read() # read the bytes of the picture
		f.close()
		subprocess.Popen(['rm','pic.png']) # delete the photo
		print("shootPicture() finished")
		return picbytes # return the bytes of the picture`

@TIS-Stefan    

I am taking this warning.

[ WARN:0] global /home/ubuntu/opencv/modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1

How am I suppose to fix it ?

closed 2021-02-19 by TIS-Stefan

Ayshine, 2021-04-19:
Hello, 
I have the same error and I am using Qt where should I add this code you suggested? 
`const char *pipeline = " tcambin serial=15810833 ! video/x-raw, format=BGRx, width=1280,height=960, framerate=25/1 ! videoconvert ! appsink";
cv::VideoCapture *cap = new cv::VideoCapture(pipeline, cv2.CAP_GSTREAMER);` or this one 
`"tcambin serial=50910677 ! video/x-raw, format=BGRx,width= 1280,height=960, framerate=20/1 ! videoconvert ! videoscale ! appsink", cv2.CAP_GSTREAMER`

TIS-Stefan, 2021-04-19:
The warning is an OpenCV issue and not caused by our modules. Please follow the link  [opencv/opencv#10324] (opencv/opencv#10324). 

A live source like a camera does not have a video position at all.

bigwhitebird29, 2021-05-07:
> I encountered the exact same error and the problem turned out to be conflicting opencv installs. I had 4.3 installed on arch linux through pacman while my project was using opencv 4.2 from pip. Uninstalling 4.3 with pacman solved the issue for me.
> 
> Hope this helps.

I'm pretty new to linux. Could you tell me how to uninstall opencv2 without removing opencv4? or do I need to remove all opencv and then reinstall opencv4?

rajhlinux, 2022-09-08:
I have the same error, How can I solve it in c/c++?