TheImagingSource/tiscamera #333: Cannot build tiscam and run tiscam without libusb.

ZiCog, 2020-01-23:
I would like to build and run tiscam on Debian Buster running under the Windows Subsystem for Linux on Windows 10. I would like to not use libusb or v4l2.

I run cmake, make, and install with those disabled:
```
cmake -DBUILD_ARAVIS=ON -DBUILD_TOOLS=ON -DBUILD_LIBUSB=OFF -DBUILD_V4L2=OFF ..
make
sudo make install
```
However it does not run:
```
$ sudo gst-launch-1.0 tcambin ! videoconvert ! fakesink
Setting pipeline to PAUSED ...
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to initialize libusb. Ret value: -99
Aborted (core dumped)
```
I have a DFK 23GM021 connected and the Windows tools, GigECam IP Config and IC Capture work just fine.

TIS-Edgar, 2020-01-24:
Hi, 

As the error indicates it tries to iterate libusb devices.
Have you tried other configurations before settling on this one?
I suspect that cmake had some caching from previous builds that result in libusb still being included (for whatever reason).
Can you uninstall, delete the build directory, rebuild and try again?
May attempts of replicating this show that libusb does indeed fail under WSL.
GigE however works as it should with your configuration...

Does `tcam-ctrl -l` cause the same error?   
Is the gige-daemon running?

ZiCog, 2020-01-24:
Sad to say that did not work:

I removed all trace of libusb from my Debian under Win 10 WSL:
```
$ sudo apt-get remove libusb-0.1-4 libusb-dev libusb-1.0-0 libusb-1.0-0-dev
$ sudo apt-get autoremove
```
I uninstalled, cleaned and rebuilt tiscamera:
```
$ cd build/
$ make uninstall
$ cd 
$ rm -rf tiscamera
$ git clone https://github.com/TheImagingSource/tiscamera.git
$ cd tiscamera
$ mkdir build
$ cd build
$ cmake -DBUILD_ARAVIS=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=OFF -DBUILD_LIBUSB=OFF ..
$ make
$ sudo make install
```
After which the the example pipeline fails badly:
```
$ gst-launch-1.0 tcambin ! videoconvert ! fakesink
Setting pipeline to PAUSED ...

(gst-launch-1.0:15582): GStreamer-CRITICAL **: 21:49:01.286: gst_caps_get_structure: assertion 'GST_IS_CAPS (caps)' failed

(gst-launch-1.0:15582): GStreamer-CRITICAL **: 21:49:01.292: gst_structure_copy: assertion 'structure != NULL' failed

(gst-launch-1.0:15582): GStreamer-CRITICAL **: 21:49:01.296: gst_structure_get_value: assertion 'structure != NULL' failed
...
...
lots more like this...
...
...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0: Could not create element 'bayer2rgb'.
Additional debug info:
Could not create element 'bayer2rgb'.
Setting pipeline to NULL ...
Freeing pipeline ...
```

The good news is that tcam-ctrl does find my camera:
```
$ tcam-ctrl -l
Available devices:
Model           Type    Serial

DFK 23GM021.I   Aravis  27410335
```

Also gige-daemon seems to work 
```
$ gige-daemon start
d$ ps -A | grep gig
15598 ?        00:00:00 gige-daemon
$ gige-daemon list
The Imaging Source Europe GmbH-27410335
```

ZiCog, 2020-01-24:
Further:

In order to stream to my RTMP server I have to install the "bad" gsteamer plugins:
```
$ sudo apt-get install gstreamer1.0-plugins-bad
```
Then I can use rtmp like so:
```
$ gst-launch-1.0 videotestsrc !  videoconvert ! x264enc ! flvmux ! rtmpsink location='rtmp://otaniemi.conveqs.fi/video/mykey3 live=1'
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock 
```
But that pulls in libusb. So now the example pipeline fails as before:
```
$ gst-launch-1.0 tcambin ! videoconvert ! fakesink
Setting pipeline to PAUSED ...
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to initialize libusb. Ret value: -99
Aborted (core dumped)
```

ZiCog, 2020-01-24:
Further:

Having installed the 'bad' plugins and hence libusb I find tcam-ctrl has the same problem, although gige-daemon does not:
```
$ gige-daemon list
The Imaging Source Europe GmbH-27410335
$ tcam-ctrl -l
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to initialize libusb. Ret value: -99
Aborted (core dumped)
```

TIS-Stefan, 2020-01-27:
Hello
I found something about WSL,Ubuntu 18.04 and USB cameras in the internet at https://github.com/MicrosoftDocs/WSL/issues/493

It says you can not access USB cameras from WSL.

Stefan

TIS-Arne, 2020-01-27:
The issue is that the library throws a runtime error exception instead of handling the absence of USB devices gracefully. 

This is an issue with the tiscamera software. We will discuss this internally how we could get it resolved quickly.

TIS-Edgar, 2020-01-27:
The steps you took were all correct.
Can you give me the output of `find /usr -name "*libusb*"`

Can you also give me the content of the file `install_manifest.txt`.
The file is located in the build directory from which you installed.

I want to make sure that there is no error in our build scripts/installation routines.
Are there maybe previous installations that did not get fully removed?

TIS-Stefan, 2020-01-27:
I just checked in Windows Subsystem Linux:
A test of ls /dev/vid* while an v4l2 camera is connected to the computer results in "No such file or directory". Thus I wonder, what the libusb should change here. However, maybe my lack of knowledge,

ZiCog, 2020-01-27:
What I have is this:
```
$ gige-daemon list
The Imaging Source Europe GmbH-27410335
$ tcam-ctrl -l
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to initialize libusb. Ret value: -99
Aborted (core dumped)
$ find /usr -name "*libusb*"
/usr/lib/tcam-0/libtcam-libusb.so
/usr/lib/tcam-0/libtcam-libusb.so.0
/usr/lib/tcam-0/libtcam-libusb.so.0.11.1
/usr/share/doc/libusb-1.0-0
$ find -name "install_manifest.txt"
./install_manifest.txt
$ cat install_manifest.txt
/usr/share/theimagingsource/tiscamera//examples/README.md
/usr/share/theimagingsource/tiscamera//examples/c/00-list-devices.c
/usr/share/theimagingsource/tiscamera//examples/c/01-list-properties.c
/usr/share/theimagingsource/tiscamera//examples/c/02-set-properties.c
/usr/share/theimagingsource/tiscamera//examples/c/03-live-stream.c
/usr/share/theimagingsource/tiscamera//examples/c/04-list-formats.c
/usr/share/theimagingsource/tiscamera//examples/c/05-set-format.c
/usr/share/theimagingsource/tiscamera//examples/c/06-softwaretrigger.c
/usr/share/theimagingsource/tiscamera//examples/c/07-appsink.c
/usr/share/theimagingsource/tiscamera//examples/c/08-save-stream.c
/usr/share/theimagingsource/tiscamera//examples/c/09-device-lost.c
/usr/share/theimagingsource/tiscamera//examples/c/Makefile
/usr/share/theimagingsource/tiscamera//examples/python/00-list-devices.py
/usr/share/theimagingsource/tiscamera//examples/python/01-list-properties.py
/usr/share/theimagingsource/tiscamera//examples/python/02-set-properties.py
/usr/share/theimagingsource/tiscamera//examples/python/03-live-stream.py
/usr/share/theimagingsource/tiscamera//examples/python/04-list-formats.py
/usr/share/theimagingsource/tiscamera//examples/python/05-set-format.py
/usr/share/theimagingsource/tiscamera//examples/python/06-softwaretrigger.py
/usr/share/theimagingsource/tiscamera//examples/python/07-appsink.py
/usr/share/theimagingsource/tiscamera//examples/python/08-save-stream.py
/usr/share/theimagingsource/tiscamera//examples/python/09-device-lost.py
/usr/lib/libtcam.so.0.11.1
/usr/lib/libtcam.so.0
/usr/lib/libtcam.so
/usr/lib/libtcamprop.so.0.1
/usr/lib/libtcamprop.so
/usr/share/gir-1.0/Tcam-0.1.gir
/usr/lib/x86_64-linux-gnu/girepository-1.0/Tcam-0.1.typelib
/usr/include/tcamprop.h
/usr/lib/libtcamgstbase.so
/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamsrc.so
/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamwhitebalance.so
/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautoexposure.so
/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautofocus.so
/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambiteater.so
/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambin.so
/usr/lib/libtcam-algorithms.so.0.11.1
/usr/lib/libtcam-algorithms.so.0
/usr/lib/libtcam-algorithms.so
/usr/lib/libtcam-network.so.0.11.1
/usr/lib/libtcam-network.so.0
/usr/lib/libtcam-network.so
/usr/lib/tcam-0/libtcam-aravis.so.0.11.1
/usr/lib/tcam-0/libtcam-aravis.so.0
/usr/lib/tcam-0/libtcam-aravis.so
/usr/lib/pkgconfig/tcam.pc
/usr/share/theimagingsource/tiscamera/images/23uc256.png
/usr/share/theimagingsource/tiscamera/images/favicon_16x16.png
/usr/share/theimagingsource/tiscamera/images/logo.jpg
/usr/share/theimagingsource/tiscamera/images/logo_small.jpg
/lib/systemd/system/gige-daemon.service
/usr/bin/tcam-ctrl
/usr/bin/tcam-capture
/usr/lib/python3/dist-packages/tcam_capture/Cache.py
/usr/lib/python3/dist-packages/tcam_capture/CapsDesc.py
/usr/lib/python3/dist-packages/tcam_capture/DeviceDialog.py
/usr/lib/python3/dist-packages/tcam_capture/Encoder.py
/usr/lib/python3/dist-packages/tcam_capture/FileNameGenerator.py
/usr/lib/python3/dist-packages/tcam_capture/ImageProvider.py
/usr/lib/python3/dist-packages/tcam_capture/MediaSaver.py
/usr/lib/python3/dist-packages/tcam_capture/OptionsDialog.py
/usr/lib/python3/dist-packages/tcam_capture/PropertyDialog.py
/usr/lib/python3/dist-packages/tcam_capture/PropertyWidget.py
/usr/lib/python3/dist-packages/tcam_capture/ROICollection.py
/usr/lib/python3/dist-packages/tcam_capture/ROIGroup.py
/usr/lib/python3/dist-packages/tcam_capture/ROIRectItem.py
/usr/lib/python3/dist-packages/tcam_capture/ResizeableRectItem.py
/usr/lib/python3/dist-packages/tcam_capture/Settings.py
/usr/lib/python3/dist-packages/tcam_capture/TcamCaptureData.py
/usr/lib/python3/dist-packages/tcam_capture/TcamDevice.py
/usr/lib/python3/dist-packages/tcam_capture/TcamScreen.py
/usr/lib/python3/dist-packages/tcam_capture/TcamSignal.py
/usr/lib/python3/dist-packages/tcam_capture/TcamView.py
/usr/lib/python3/dist-packages/tcam_capture/ViewItem.py
/usr/share/applications/tcam-capture.desktop
/usr/lib/python3/dist-packages/tcam_capture/image_dir.py
/usr/bin/camera-ip-conf
/usr/bin/gige-daemon
/usr/lib/libtcam_gigewrapper.so
/usr/lib/python3/dist-packages/tcamgigetool/__init__.py
/usr/lib/python3/dist-packages/tcamgigetool/__main__.py
/usr/lib/python3/dist-packages/tcamgigetool/controller.py
/usr/lib/python3/dist-packages/tcamgigetool/tcam_gigetool.py
/usr/lib/python3/dist-packages/tcamgigetool/version.py
/usr/bin/tcam-gigetoolmichael@monster:~/tiscamera/build
$
```
Hope that helps.

What I don't understand is why tiscam is even looking for anything on USB after having been built with the options that say not to.

TIS-Edgar, 2020-01-28:
Ok, what happens is

libtcam.so dynamically loads available backends that the system offers.  
These are:
- libtcam-aravis.so  
- libtcam-libusb.so  
- libtcam-v4l2.so  

What now happens on your system is that a previous installation was not correctly uninstalled (at least that is my suspicion) and libtcam.so still finds the libusb backend.

The install_manifest.txt tells us what files habe been installed from your current build. As you can see there is no libtcam-libusb.so in there.

The simplest solution to all this is probably if you manually delete all files that are listed with `find /usr -name "*libusb*"`.

The dynamic loading has already been removed on the development branch so I am certain that these issues will not happen again once 0.12.0 is released.

ZiCog, 2020-01-28:
Yes! That fixed it.
```
$ gst-launch-1.0 tcambin ! \
>                video/x-raw,format=BGRx,width=1280,height=960,framerate=25/1,drop-only=true ! \
>                videoconvert ! \
>                x264enc ! \
>                flvmux ! \
>                rtmpsink location='rtmp://otaniemi.conveqs.fi/video/mykey3 live=1'
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-bayer, framerate=(fraction)25/1, width=(int)1280, height=(int)960, format=(string)grbg
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...```
Now I wonder why it consumes over 200% CPU on my 4 core x86_64 PC, but that is another story...

Thanks.

closed 2020-01-28 by ZiCog