TheImagingSource/tiscamera #335: softerware trigger and copy image from camera doesn't work

wutongyu98, 2020-03-30:
Hi, I am using the usb3.1 board camera from imaging source. I can successfully run the "06-softwaretrigger" and "07-appsink" from the "tiscamera" example .  I know "07-appsink" is used to access data via "sink" from camera, and "06-softwaretrigger" is used to trigger the camera action. But when put them together, I cannot get the data out. Could you provide a short example how to get these two function (software trigger and copy data from camera) work together? Thanks

TIS-Stefan, 2020-03-31:
Hello

you may look at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples for complete samples, that try to do, what you want.

Stefan

wutongyu98, 2020-03-31:
thanks stefan. I figured that out. Just need to set the syn=true, I can get the synchronized camera action.

closed 2020-04-01 by TIS-Stefan