TheImagingSource/tiscamera #336: Cmake compilation error

f2013619, 2020-04-03:
Hi, 
After installing the dependancies I am trying to run the cmake command but I am getting the following error-
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  cmake/modules/FindGObjectIntrospection.cmake:39 (pkg_check_modules)
  src/gobject/CMakeLists.txt:20 (find_package)

-- Configuring incomplete, errors occurred!
Can you please help me with this.

TIS-Stefan, 2020-04-03:
Which Linux Distribution do you use?

f2013619, 2020-04-03:
Hi, 
I am using Jetson Tx2 board with Ubuntu 18.04 LTS.

TIS-Stefan, 2020-04-03:
Then the scripts should have installed all. even the Gobject Introspectio. You cloned the project first and then run the scripts for installing runtime and dependencies?

f2013619, 2020-04-03:
Hi, I installed the dependencies and ran cmake command it was successful then I ran make command and now my system is not going to login mode. A black screen appears that it goes into phs daemon. Can you please help me with this.

TIS-Stefan, 2020-04-06:
Hello

I wonder, what you did, we never had such an issue before.
As long as you can not change with Crtl+Alt+F1 into another terminal and repair that somehow (I must admit I do not know how to) you better reinstall the Ubuntu and try again.

I am curios: You cloned the tiscamera repository from https://github.com/TheImagingSource/tiscamera, run the 
sudo ./scripts/install-dependencies.sh --compilation --runtime
and your computer does not start correctly anymore?

Stefan

f2013619, 2020-04-13:
Hello,
It seems some files got corrupted in my system. So I reinstalled ubuntu and
now there are no issues in using the tcam-capture.
I am currently using DFM 25G445-ML camera with Jetson TX2. When I was
trying to take pictures with the same I am just getting reaction to lights
but not clear images.
I do not have the mount for the lens.
Please advise on the same.
Please find attached the few pictures I took with the camera.
Regards,
Shivani Choudhary

TIS-Stefan, 2020-04-14:
Hello Shivani

For a lens and lens holder, you must contact your camera dealer. He will be able to calculate the best lens and the needed lensholder. Please look at the document https://s1-dl.theimagingsource.com/api/2.5/packages/publications/whitepapers-cameras/wpgigeboard/7b8496e0-5271-5cc1-a10f-5c76ede2f1c6/wpgigeboard_1.0.en_US.pdf
It shows, which parts you need.

Stefan

f2013619, 2020-04-14:
Hi,
I am using the Intellinet 5-port gigabit ethernet POE + switch.(
https://intellinetsolutions.com/intellinet-en-5-port-gigabit-ethernet-poe-switch-561228.html
)
After I use that one light on the ethernet port on the camera is green and
the other one is orange. Also POE light on the switch is orange. Can you
please tell me about this?

TIS-Stefan, 2020-04-14:
Hello

> After I use that one light on the ethernet port on the camera is green and
> the other one is orange. Also POE light on the switch is orange. Can you
> please tell me about this?
This tells us, that you have a GigE connection (green light) and the blinking orange light shows communication. 

Stefan

closed 2021-02-19 by TIS-Stefan