TheImagingSource/tiscamera #337: does not compile with clang 6.00

midjji, 2020-04-03:
Specifically: 

...tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1573: error: expected the class name after '~' to name a destructor     (&self->device_serial)->std::string::~string();

Note gcc erroneously instead calls the wrong destructor, but it does compile. 

Proposed fix, remove the line, and check for other c style errors such as the next line which is causing undefined behaviour later when the destructor is called again when the object goes out of scope later.  Explicitly calling the destructors for c++ class member variables is almost certainly a sign of a bad port from c. 

                                          ^

midjji, 2020-04-03:
(&self->device_serial)->~basic_string() // would possibly be an alternative. But I still think something is very strange when explicitly calling destructors.

TIS-Stefan, 2020-04-03:
I have to ask. I only wonder, why you have this issue, but all other users not. The code did not change.

midjji, 2020-04-03:
export CC=/usr/bin/clang; export CXX=/usr/bin/clang++ in my zshrc
it looks like it works fine with gcc, but bear in mind that it is a known compiler bug that it does.

midjji, 2020-04-03:
Ubuntu 18.04 fully updated, clang 6.0.0 default clang, install dependencies, then compile.
Does it fail to replicate?

TIS-Stefan, 2020-04-06:
Hello

Currently we support and test the default compilers only, not clang. 
Maybe on long term we test clang.

Also: The destructor calls in [gst_tcam_src_finalize](https://github.com/TheImagingSource/tiscamera/blob/master/src/gstreamer-1.0/gsttcamsrc.cpp#L1573) are there due to legacy support for older g++ versions. This is also documented in [gst_tcam_src_init](https://github.com/TheImagingSource/tiscamera/blob/master/src/gstreamer-1.0/gsttcamsrc.cpp#L1537). 

Stefan

midjji, 2020-04-06:
As you like,
Gcc generally is a bit less strict and buggy with the c++11 and onwards
stuff, so combined with clangs more and better warnings I use it by default.

In this case the code is invoking undefined behaviour, and clang gives an
error while gcc just does nothing/removes the line( probably depends on
settings).

TIS-Stefan, 2020-04-06:
Hello

I totally agree with you. However, that is, what my programmers told me, they will do about that currently. I am bound to them.

Stefan

midjji, 2020-04-07:
No worries, easy to fix in fork. Go ahead and close.

I'm curious if there is an example of using the tis camera with a
fullfledged c++11 api i.e. without using the gst interface? The
CaptureDevice seems to be a much nicer interface to work with, but no
examples?  Spotted callbacks for on dropped/lost frame and on partial frame
deeper in the code too, which dont seem to be exposed by the gst interface
or did I miss that somewhere ?

I'd also love if there was a way to access the camera trigger id counter to
uniquely associate frames with time, which something in the gst pipeline
does not support atm if I understand it right. I know it works in windows,
so the functionality is there, but reverse engineering the api seems like a
hassle. I'd be fine with a firmware version which encodes it in a few
pixels too btw.

TIS-Edgar, 2020-04-07:
Hi,

For various reasons the decision back in the day had been made to opt for gstreamer and gobject introspection. The reason for lack of official documentation is that the CaptureDevice is our backend API and not the 'official' API. You are free to use it but do not expect a while lot of support for it.

There may be changes to this in the future but nothing within the next 1-2 years....

Frames lost are only counted and given to the user as meta information.  
Definition [here](https://github.com/TheImagingSource/tiscamera/blob/master/src/base_types.h#L191) and usage in gstreamer [here](https://github.com/TheImagingSource/tiscamera/blob/master/src/gstreamer-1.0/gsttcamsrc.cpp#L1467).
The meta information contain frames lost, is damaged, frame count frame time (if available).
Please be aware that metainfo support is relatively new in gstreamer which leads to the fact that things like the bayer2rgb element simply drop the information.

If you want to save device property states I suggest you look into our development branch.
It contains a json state unit that allows reading/writing json descriptions.
You can find the backend implementation [here](https://github.com/TheImagingSource/tiscamera/blob/development/src/json.cpp). "Frontend" usage would be tcambin/src and tcam-ctrl.
The release of tiscamera 0.12.0 will hopefully be in the next couple of weeks. We are currently in the test/fixing phase of the development branch.

Does that answer your questions?

midjji, 2020-04-07:
yes,
thank you.

I'll check it out-

closed 2021-02-19 by TIS-Stefan