TheImagingSource/tiscamera #340: Linux, DMM 72BUC02-MLPB missing "Strobe Enable" Property

junsujang, 2020-04-14:
Hello,
I wanted utilize the strobe functionality of DMM 72BUC02-MLPB upon trigger. However, it seems like the property "Strobe Enable" is unavailable using tiscamera library. I saw another post #309 , which faced the same problem. I was wondering if the same firmware update script is applicable for this model. I didn't want to mess up with the firmware. 

Thank you!

TIS-Stefan, 2020-04-14:
Hello

I am very sorry, but the strobe enable is not in the Linux firmware of the camera.
In case you trigger the camera externally, you can use the external trigger source for your strobe light.
If the camera is free running, then there is no pause between to images, so the strobe wont change and makes no sense.

Stefan

futureoceanlab, 2020-06-01:
Thanks, external trigger would work.

closed 2020-06-01 by junsujang